MarlinFirmware/Marlin #8850: Delta Calibration + UBL Y_tower to infinity

mybrainsshit, 2017-12-20:
### Bug Report
- Description: trying to use UBL and Autocalibration with Delta-Kossel Mini/Rostock Anycubic. 
- Expected behaviour: Move to position and have 0mm gap to glassbed.
- Actual behaviour: After Autocalibration and UBL, and UBL is active: Problem is that as soon as i go to the Y axis stand to check the calibration with paper, the nozzle drives up to infinity. This can only be stopped by turning of the machine. It completly ignores the endstops and rams into the top of the printer. Same issue on  1.1.5 and 1.1.7. 
- Steps to reproduce:
With my Config and firmware freshly flashed (Config_adv wasn't touched):
I do a 
M502
M500
G33 with rather high standard deviation of around 0.08;
M500 to save
G28
G29 P1
G29 T
G29 S1
G29 F10.0
G29 A
M500

Then i remove my probe manually, 
G28 again and check weather the leveling has worked:
G0Z0: approx 0.4mm distance to plate in center. not good
G0Z0X-52Y-30 at x tower: scratching glass
G0Z0X0Y60  at z tower: approx 0.4mm distance again
G0Z0X52Y-30 at y tower: printer goes crazy. (moves up until power cut)

So to summarize, it seems to be 2 problems:
1. Bed is not leveled after UBL even tho G29 T seems valid
2. Y-Tower manual checking goes nuts. 

Config:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1576242/config.zip)

Thank you so much in advance! I've been fiddling around with this for quite a while and i can't get it to work. Marlin is awesome(if bed is leveled manually ;) )

Roxy-3D, 2017-12-20:
Can you please post your G29 T and a G29 W 

It kind of sounds like you are moving the nozzle off of the mesh.  Or maybe the mesh cell is not defined where you are moving the nozzle.   You are not doing a G29 P3 of any flavor, so that is a distinct possibility.

ErshoF, 2017-12-23:
UBL includes a third phase, G29 P3, which fills in points on the mesh that were not probed automatically or manually. Note that unlike in bilinear leveling, UBL does not automatically extrapolate correction beyond the bounds of the mesh. If a mesh point is not defined no correction will be applied, and a missing point can affect up to 4 mesh cells.

Issue G29 P3 (no other parameters) to do a ‘smart fill’ of missing mesh points. This uses an extrapolation algorithm - which varies between delta and Cartesian systems - to give the unfilled mesh points reasonable initial values. You may need to run this more than once – each instance of G29 P3 will fill in one missing line of the grid. This allows fine tuning between P3 steps when filling the remainder of larger grids. From this point, G26 and G29 P4 can be used to iteratively refine the mesh.

G29 P3 Cx.xx can be used to manually fill a value into a mesh point(s), like M421, if for some reason that is necessary. G29 P3 Cx.xx Rn will fill the nearest n points with the value x.xx; ‘nearest’ is referenced to the nozzle position, unless X and Y arguments are provided to override the search start point.

Again, G29 S[n] will save the mesh to EEPROM.

Roxy-3D, 2017-12-23:
Thank You @ErshoF  !   That is exactly correct.

mybrainsshit, 2017-12-23:
Okay thank you for this input, will try this as soon as i can :)

mybrainsshit, 2018-01-10:
Sorry for the long wait, i tore of my nozzle while tightening after a clog -.- and diddn't have spares. 
I enabled the debugging as requested.
Issuing G29 P3 just lets the printer sit there doing nothing. i don't get any return values nor does it move.
What seems a bit curious to me is the probe offset: i entered -19.38 in arduino, but it has alot of digits in the output, is that a float problem due to delta kinematics?

I didn't change anything on the bed leveling grid so here it is:

[cmds_output](https://github.com/MarlinFirmware/Marlin/files/1618738/cmds.txt)

SENDING:G29 T
Bed Topography Report:
(0,9)                                                                   (9,9)
(-40,40)                                                                (40,40)
  0.057    0.033    0.028    0.024    0.001   -0.023   -0.051   -0.090   -0.095   -0.095

  0.018    0.028    0.011    0.017    0.002   -0.023   -0.036   -0.050   -0.102   -0.112

 -0.008    0.001    0.027    0.026    0.009   -0.000   -0.008   -0.037   -0.063   -0.092

  0.004    0.014    0.027    0.037    0.040    0.031    0.001   -0.009   -0.011   -0.011

  0.000    0.023    0.038    0.051    0.053    0.039    0.033    0.016   -0.000    0.017

  0.030    0.033    0.038    0.049  [ 0.018]   0.049    0.045    0.029    0.025    0.031

  0.029    0.026    0.027    0.034    0.039    0.036    0.030    0.033    0.013    0.005

  0.026    0.011    0.021    0.031    0.028    0.030    0.020    0.007    0.010   -0.002

  0.015    0.002    0.006   -0.002    0.030    0.027    0.013    0.017   -0.002   -0.022

  0.007    0.001   -0.015    0.006   -0.011    0.004    0.018    0.001    0.000   -0.023
(-40,-40)                                                                    (40,-40)
(0,0)                                                                     (9,0)

SENDING:G29 W
Unified Bed Leveling System v1.01 active.
Mesh 1 Loaded.
UBL object count: 1
planner.z_fade_height : 10.0000
'#' of samples: 100
Mean Mesh Height: 0.006677
Standard Deviation: 0.034605
zprobe_zoffset: -19.3799991
MESH_MIN_X  ((0 - 90.0) + 50)=-40.00
MESH_MIN_Y  ((0 - 90.0) + 50)=-40.00
MESH_MAX_X  ((0 + 90.0) - (50))=40.00
MESH_MAX_Y  ((0 + 90.0) - (50))=40.00
GRID_MAX_POINTS_X  10
GRID_MAX_POINTS_Y  10
MESH_X_DIST  8.89
MESH_Y_DIST  8.89
X-Axis Mesh Points at: -40.000  -31.111  -22.222  -13.333  -4.444  4.444  13.333  22.222  31.111  40.000
Y-Axis Mesh Points at: -40.000  -31.111  -22.222  -13.333  -4.444  4.444  13.333  22.222  31.111  40.000
Kill pin on :41  state:1
Unified Bed Leveling sanity checks passed.

Roxy-3D, 2018-01-10:
You have a very flat bed.  Especially so for a Delta.    You can move the nozzle slowly in X & Y and verify the nozzle height is correcting properly.    If the nozzle is getting too close to the bed, you can edit those mesh points and cause the nozzle to track higher.   

G26 is very helpful to make sure all the mesh points are set to correct values.

Very soon, I'll be adding capability to edit mesh points outside the DELTA_PRINTABLE_RADIUS.   It is possible that will help you too for the bed area very close to the DELTA_PRINTABLE_RADIUS.

closed 2018-02-11 by thinkyhead