MarlinFirmware/Marlin #8854: [FR] Set (and switch between) two or three heating PIDs presets to take care of fan airflow at different speeds

MasterPIC, 2017-12-21:
Hi,
what about adding (and dynamically switching between) two or three heating PID presets to take care of fan airflow varying with fan speed (and inevitably affecting the nozzle temperature)?
Maybe also with one PID preset but "adapting" it to fan speed the airflow affecting the nozzle temperature could be reduced but there would be the need for a proper algorithm and a different calibration too.

Thanks
Armando

kcheeeung, 2017-12-22:
A PID controller is a feedback mechanism that corrects for errors in the heating for your nozzle. There should be no need to change it multiple times, unless you change to a different heating element.

You can calibrate the PID for your system with `M303 E0 S(your target temp) C(# cycles)`. You would also need to have an auto-fan enabled on your heating element.

LVD-AC, 2017-12-22:
IMO 2 settings should do: PID with fans on at 100% and PID with fans off; every time fan speed is changed it can interpolate PID settings between the two. On my printer PID autotune with fans on at 50% gives about halfway results between 0% and 100%.

MasterPIC, 2017-12-22:
I know that PID, thanks to K_i and K_d takes care of the "history" of temperature error. But PID is still calibrated at one fan speed state. As LVD-AC noted, calibrating at different fan speeds would cause different PID K_p, K_i and K_d values; what I was speaking about...

@LVD-AC yep, I thought a similar approach...

Another option could be to add a **Reduced**_PID_FUNCTIONAL_RANGE in case of an (huge?) fan speed change to reduce the risk of a false positive runaway error...

thinkyhead, 2018-02-11:
You can change the PID at runtime with `M301`, so just have your slicer insert `M301` with the PID values you like at whatever layer your fan turns on / off / etc.

closed 2018-02-11 by thinkyhead