MarlinFirmware/Marlin #8855: [2.0.x] Heat sensor spikes and thermal runaway error

teddybee, 2017-12-21:
In a print I got this error in MKS Sbase:
Recv: Error:Thermal Runaway, system stopped! Heater_ID: 0
Changing monitoring state from 'Printing' to 'Error: Thermal Runaway, system stopped! Heater_ID: 0

Also I discovered very strange spikes in hotend temperatures. I attached a screenshot.
[image]

The firmware was built one hour ago.

b3nn0, 2017-12-21:
Exactly the same problem here with latest Git version on MKS Sbase.

tcm0116, 2017-12-27:
I'm having the same issue on my Re-ARM board. I now have an army of Pikachu lower halves because of this. Haha,

@Bob-the-Kuhn any ideas?

p3p, 2017-12-27:
With the amount of software filtering I added to the the LPC1768 platforms `HAL_adc_get_result` when this issue with the ADCs first came up I have no idea how a graph like that is even possible, is this something that has appeared again recently?

b3nn0, 2017-12-27:
I guess so, at least it seems to be present in my recent builds straight from git.
It is quite surprising. Initially I used the thermistor PID values that worked fine in older versions and saw these issues. Then I re-calibrated the PID and received very similar results. The PID calibration temperature curve looks absolutely normal.

I'm not too familiar with the Marlin code, but intuitively, it doesn't even look like a hardware-dependent problem and more like a bug in the PID loop. But then again, many more boards would have issues then, so it is quite confusing...

AnHardt, 2017-12-27:
That can't have anything to do with PID. 
Temperatures can't rise faster than when full heating and can't fall faster than when not heating.
These spices come from a the measurement system - not the heating system.

b3nn0, 2017-12-27:
Then I see two issues here. The two big spikes on the one hand and the generally bad heating performance on the other hand.
You are right about the spikes yes.
The temperature curve in general, however, more looks like the PID loop kinda expects a lot of overshoot when heating, so it stops early and only very slowly wiggles towards the target temperature.

Reading the OP again, the spikes were the thing that were reported. I just saw a temp graph very similar to mine and decided to comment. Probably not the smartest move.

tcm0116, 2017-12-27:
I agree with @AnHardt as the temperature of the nozzle just can't change that fast. 

I took a look at the history for the LPC1768 HAL, and couldn't find any obvious changed that would have caused this. There only thing remotely close is #8872, but I wouldn't think we're calling `SET_OUTPUT` on the ADC inputs. 

Looks like we're just going to have to check old commit points to see if we can isolate where the issue starts.

tcm0116, 2017-12-27:
@b3nn0 the thing is that the spikes are probably messing with the PID loop. The strange thing is that I would think the PID loop would increase the nozzle temperature in response to the spikes, but I was getting the thermal runaway due to the temperature being too low. It might be useful to monitor the nozzle temp using a different measuring system to see what's actually happening.

teddybee, 2017-12-27:
@tcm0116 there was no spikes when I done the PID tuning, and I saw very uniform sin curves. I will attach a picture about that too.

makz27, 2017-12-29:
Same problem, with mks sbase, my temperature curve from my last print
[image]

p3p, 2017-12-29:
I can't replicate this on my Re-ARM my temperature control looks the same as it did when I implemented the filters to compensate for the notoriously bad (and badly affected by noise) ADCs on the LPC1768, The bed sensor looks pretty clean.. the best I can offer you at the moment is to turn the lowpass filter up to max (8), [HAL/HAL_LPC1768/HAL.cpp#L135](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_LPC1768/HAL.cpp#L135) it may help... may not

teddybee, 2017-12-30:
@p3p Could you check the ADC pin initialisation? There are lots of parameters that makes the reads rock solid on Cortex M(at least with native code). I don't think it is an electrical noise.
[http://kishorepunniyamurthy.blogspot.hu/2014/05/arm-tutorial-2-gpio-and-adc.html](http://kishorepunniyamurthy.blogspot.hu/2014/05/arm-tutorial-2-gpio-and-adc.html)

tcm0116, 2017-12-30:
**ADC_TwoSamplingDelay** looks interesting. I wonder if that a feature of the chip or the SDK.

p3p, 2017-12-30:
> Could you check the ADC pin initialisation? There are lots of parameters that could make the reads rock solid on Cortex M(at least with native c code).

The errata for the LPC1768 says otherwise, very susceptible to toggling pins next to the ADC and general noise, hopefully there is a better configuration though, it's currently the simplest it could be.

teddybee, 2017-12-30:
I have made a PID tune, there was no glitch during this(as I remembered).
[image]
#define  DEFAULT_Kp 20.779268
#define  DEFAULT_Ki 1.718431
#define  DEFAULT_Kd 62.815727

p3p, 2017-12-30:
@teddybee without activity on the steppers there should be no noise, so clean results.

makz27, 2017-12-30:
I have to add, i used the same hardware with smoothieware firmware for months without any issues, if now I switch back to Smoothie, I'll have a straight curve (even for the bed, on my screenshot the bed is not good as it was on smoothieware). So it's really software related

(You can see the difference here https://github.com/MarlinFirmware/Marlin/issues/8743 )

teddybee, 2017-12-30:
I tried to print again with the latest git compile.
[image]
When the printer make an auto leveling it switches off the heating, the nose temp falls back to around 225C. But after it takes more than 5 minutes to climb back to 240, its a lots of time...
Until the 3rd sharp spike the motors didn't even move, after that the printing was started.

p3p, 2017-12-30:
> So it's really software related

@makz27 I wasn't saying this was working as intended, just explaining the root cause is the hardware, we obviously are not handling the ADCs well enough and filtering the signal as much as needed yet, I only have my printer to test on, and I assume because of my setup (Re-ARM board) and shielded wiring (CAT 7) I am not effected this badly, It's hard to fix problems you don't have. Saying that although I have no visible noise my pid control still varies erratically by a few decrease +/-.

teddybee, 2017-12-30:
@p3p Are you using diode hack on the DRV8225s? May it helps with this issue?

p3p, 2017-12-30:
@teddybee I don't use DRV8825s the Re-ARM doesn't include drivers it plugs into a RAMPS shield. I cant help but think your PID values are wrong, in general it should overshoot slightly on the initial ramp, and then converge .. I re-tuned mine today and apart from the small oscillation that really shouldn't be there it works reasonably well, I'm not sure why the PID control oscillates so badly, but that's a different issue to the bad ADC readings (although perhaps related).

[image]
(this graph is before the filter modifications, It made no difference to me)

I've played around a bit with this today, I can't reproduce the noise levels shown in this issue but I added an 13 sample median filter in this [branch](https://github.com/p3p/Marlin/tree/pr_bf2_more_adc_filter) that should reject a lot of bad readings, there is already a mean filter because of the lowpass .. I may have more time tomorrow. I can't see anything fundamentally wrong with how the ADCs are configured although I will look through again.

tcm0116, 2018-01-01:
I took a look at what Smoothie is doing, and determined that they're using a different filtering technique. They're taking an average of the median 16 values from the previous 32 values sampled, which is basically a moving average, but with some outlier clipping included. I've created a version for the LPC1768 HAL that seems to be working well for me:

https://github.com/tcm0116/Marlin/commit/8aae442ccbe3f5cd679cddacfc21947891c51de2

I'm still seeing oscillations on the temperature, but at least no spikes. Please give it a try and let me know if if works for you.

p3p, 2018-01-01:
@tcm0116 My algorithm acted (almost) the same way, although I only used a 3 value median originally, (and the lowpass filter doesn't need to store previous samples), that's why I upped the median to 13 was hoping to get some feedback on that.

tcm0116, 2018-01-01:
@p3p I'll give yours a try tomorrow. Any thoughts on the oscillations? It's
extremely consistent in mine with an amplitude of around 5 degrees at 3-4
second interval. I've run the PID autotune several times, with no
improvement. Is the PID loop being run at the same rate on AVR as the
LPC1768?

teddybee, 2018-01-01:
@p3p I compiled your branch, did a PID tune and looks like the spikes was eliminated(at least covered). But still has a massive "undershoot" with the pid tune. I will rise the P and reduce the D part manually.
[image]

AnHardt, 2018-01-01:
Play with `PID_FUNCTIONAL_RANGE`.

tcm0116, 2018-01-01:
@p3p I tested your branch this morning, and your expanded Median filter looks like it's working.

I also did some digging around and found the cause of the oscillations. The `PID_dT` macro value was being computed using `(F_CPU / 64.0 / 256.0)`, which was resulting in a value of over 6000 Hz for the LPC1768, when it should be 1000 Hz, which is the rate that the temperature ISR runs at. I've created a commit with the change: https://github.com/tcm0116/Marlin/commit/2c3aa825e49d880be6b4f253758908506204dd43

Do you want to create a PR with both changes?

p3p, 2018-01-01:
@tcm0116 Ah I'm glad it was something so simple, I'm sure we will find other things I missed when trying to finish and constantly rebase bobcs HAL implementation (I'm surprised it works as well as it does ;) ), I'l add that fix to my adc filter pr and submit it.

I've been working on updating all the file naming convention through the HAL implementations and make the HAL redirection headers less problematic but without the ability to specify include / source folders (Arduino IDE compatibility) I think its always going to a mess.

p3p, 2018-01-01:
@teddybee  @makz27 could you test this [branch](https://github.com/p3p/Marlin/tree/pr_bf2_more_adc_filter) (pr #8996) to make sure this fixes things reasonably for you, I now get stable hotend temperature control with the occasional 0.5 degree variance while printing, heated bed never varies.

teddybee, 2018-01-02:
@p3p @tcm0116 Thank you guys for solving the problem, PID looks good now:
[image]

556duckvader, 2018-01-02:
Issue appears to be fixed now. Thanks @p3p @tcm0116.  I just uploaded this to my ReArm/Ramps ran the PID autotune and now its pretty rock solid.  It does a little blip once it gets around the set temp then levels out.
[image]

makz27, 2018-01-02:
Hmmm better at the beginning and fail a bit after
[image]

tcm0116, 2018-01-02:
@makz27 are you running the branch from #8996? I did a 10 hour print with it yesterday, and it was very stable. Also, did you apply the PID values with by passing the U1 parameter to M303 or with M301 after the PID auto tune completed?

makz27, 2018-01-02:
@tcm0116 Yep, now the temperature is very stable like it was on smoothieware but i start to have spikes when i'm printing (it's stable when i'm printing with smoothieware).

There's an evolution but still need improvements

[image]

tcm0116, 2018-01-02:
@makz27 can you try increasing `ADC_MEDIAN_FILTER_SIZE` to a larger number, perhaps something like 23?

https://github.com/p3p/Marlin/blob/52110427653fb70f1fc8b0043e3da629f9f3c12d/Marlin/src/HAL/HAL_LPC1768/HAL.cpp#L134

p3p, 2018-01-02:
To get the amount of noise through my original filter that is shown in @makz27 original comment I'm surprised there is any signal there at all. You can try increasing  ADC_MEDIAN_FILTER_SIZE and ADC_LOWPASS_K_VALUE in "Marlin/src/HAL/HAL_LPC1768/HAL.cpp", The more we filter this data the longer the step delay gets but as temperature change is relatively slow acting I guess we can just go mad as long as it doesn't confuse the PID controller.

tcm0116, 2018-01-02:
@p3p Smoothieware uses a 32 value median filter, so I would think we can go
a bit higher than 13 without too much issue.

makz27, 2018-01-02:
@tcm0116 Way better with ADC_MEDIAN_FILTER_SIZE at 23 (i didn't ran a autotune after the modification)
[image]

556duckvader, 2018-01-04:
Ok didn't have a chance to try any extruding the other day when I posted the image of the temp being stable.  Today I was testing my extruder and it seem like as long as I am not extruding the temp will remain stable.  As soon as I start extruding it starts to fluctuate again.

[image]

tcm0116, 2018-01-04:
@556duckvader it sounds like the extruder step pluses are inducing noise on your thermistor signal. Can you try increasing `ADC_MEDIAN_FILTER_SIZE` to 23 and see if that cleans it up?

556duckvader, 2018-01-05:
@tcm0116 I replaced all my stepper drivers with better ones and its not doing it now.  I am doing a test cube print now and the time is pretty rock solid.

PR #8996 (2018-01-15): [2.0.x] PID control fix and more LPC1768 ADC filtering
This PR introduces even more sample rejection and filtering for the LPC1768 ADCs (increased median filter from 3 to 23 samples) it also fixes the PID control for all non-avr platforms as the frequency was wrong in the PID_dT calculation. 

fix #8855, fix #8743
closed 2018-02-11 by thinkyhead