MarlinFirmware/Marlin #8857: [2.0.x] DUE+RADDS1.5+RADDS LCD

subdancer, 2017-12-21:
Hello, i'm trying to compile Marlin 2.0.x  with PlatformIO
for an Arduino DUE with a RADDS 1.5 Shield and the original RADDS Display:
http://doku.radds.org/dokumentation/lcd-display/

When i only configure DUE and RADDS the compiler lists tons of warnings see:

```
...
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp: In member function 'void TMC26XStepper::send262(long unsigned int)':
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp:956:30: error: 'SPCR' was not declared in this scope
unsigned char oldMode =  SPCR & SPI_MODE_MASK;
^
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp:956:37: error: 'SPI_MODE_MASK' was not declared in this scope
unsigned char oldMode =  SPCR & SPI_MODE_MASK;
^

*** [.pioenvs\DUE\lib\TMC26XStepper\TMC26XStepper.o] Error 1
 [ERROR] Took 35.68 seconds
 
 [SUMMARY]
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment anet10                      [SKIP]
Environment LPC1768                     [SKIP]
Environment LPC1768_debug_and_upload    [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment STM32F1                     [SKIP]
Environment teensy20                    [SKIP]
Environment teensy35                    [SKIP]
Environment DUE                         [ERROR]
 [ERROR] Took 35.68 seconds
```

when i set 
lib_ignore  = TMC26XStepper, Adafruit NeoPixel
it compiles through but there are still lots of warnings like:

```
...
Compiling .pioenvs\DUE\lib\U8glib-HAL_ID1932\lib\u8g_com_arduino_port_d_wr.o
.piolibdeps\U8glib-HAL_ID1932\src\lib\u8g_com_arduino_no_en_parallel.c:93:20: warning: assignment from incompatible pointer type [enabled by default]
u8g_data_port[3] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D3]));
^
.piolibdeps\U8glib-HAL_ID1932\src\lib\u8g_com_arduino_no_en_parallel.c:96:20: warning: assignment from incompatible pointer type [enabled by default]
u8g_data_port[4] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D4]));
^
.piolibdeps\U8glib-HAL_ID1932\src\lib\u8g_com_arduino_no_en_parallel.c:98:20: warning: assignment from incompatible pointer type [enabled by default]

u8g_data_port[5] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D5]));
^
.piolibdeps\U8glib-HAL_ID1932\src\lib\u8g_com_arduino_no_en_parallel.c:100:20: warning: assignment from incompatible pointer type [enabled by default]
u8g_data_port[6] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D6]));
^
.piolibdeps\U8glib-HAL_ID1932\src\lib\u8g_com_arduino_no_en_parallel.c:102:20: warning: assignment from incompatible pointer type [enabled by default]
u8g_data_port[7] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D7]));
^
...
```

```
...
Compiling .pioenvs\DUE\lib\U8glib-HAL_ID1932\lib\u8g_com_arduino_st7920_hw_spi.o
.piolibdeps\U8glib-HAL_ID1932\src\lib\u8g_com_arduino_no_en_parallel.c: In function 'u8g_com_arduino_no_en_parallel_init':
.piolibdeps\U8glib-HAL_ID1932\src\lib\u8g_com_arduino_no_en_parallel.c:87:20: warning: assignment from incompatible pointer type [enabled by default]
u8g_data_port[0] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D0]));
^
.piolibdeps\U8glib-HAL_ID1932\src\lib\u8g_com_arduino_no_en_parallel.c:89:20: warning: assignment from incompatible pointer type [enabled by default]
u8g_data_port[1] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D1]));
^
[... 145 lines cut ...]

defining:
`#define SDSUPPORT`

results in the following compiler output:
```
sketch\src\lcd\ultralcd.cpp: In function 'void lcd_init()':

sketch\src\lcd\ultralcd.cpp:4534:5: error: 'lcd_sd_status' was not declared in this scope

     lcd_sd_status = 2; // UNKNOWN

     ^

sketch\src\lcd\ultralcd.cpp: In function 'void lcd_update()':

sketch\src\lcd\ultralcd.cpp:4644:22: error: 'lcd_sd_status' was not declared in this scope

     if (sd_status != lcd_sd_status && lcd_detected()) {

                      ^

exit status 1

```

Bob-the-Kuhn, 2017-12-21:
Yes, there are LOTS of warnings when compiling for Due.

Thanks for the **lib_ignore = TMC26XStepper** - much better than my solution.

@ejtagle is doing lots of work on the Due code.  Let's give him some time to address these annoyances.

subdancer, 2017-12-23:
Ok, i've got it working. Display is fine now, encoder and sd card reader work perfect as well.
small vid here: https://www.youtube.com/watch?v=ctDZYEvaJ20
can we get the following changes to make it available for all?
in
/src/inc/Conditionals_LCD.h add:
```
#elif ENABLED(RADDS_DISPLAY)
  #define ULTRA_LCD 
  #define ULTIPANEL
  #define ENCODER_PULSES_PER_STEP 2
  #define SDSUPPORT
```

in /Configuration.h add:
```
/**
 * RADDS Display
 *
 * Enables the original RADDS Display + Encoder + Buzzer+ SDCardReader. 
 * you must uncomment the following option or it won't work.
 */
//#define RADDS_DISPLAY
```

Bob-the-Kuhn, 2017-12-23:
You can do a pull request (PR) for this.  There's a write up in the Marlin docs on how to do that.

closed 2017-12-25 by thinkyhead