MarlinFirmware/Marlin #8858: Need Marlin to Ignore X homing and set X=0 when told to home.

znelson710, 2017-12-21:
I am working a vinyl plotter made of 3d printed parts and printer parts.
It's mased on a melzi set up from a Maker Select. The design of the plotter uses a rotating drum for the x axis to cut down machine size. As such there is no hom position for x. 

What I am doing right now during testing is manually pushing a limit switch so it thinks x has homed.
What I would like to is to just tell the printer when homing x that it is now at x=0.

G92 only works after homing or when it knows what location it is at. When turning on the printer or after turning the steppers off, the firmware sets the coordinates to unknown.

KevMags, 2017-12-22:
Turn off machine, manually move X axis to where you want X=0 to be, turn on machine.  Marlin will assume that X is now at zero.  Do not home X or Marlin will move X and expect a limit switch.   You can then use  "G28 Y Z" to home the other axes if needed.

thinkyhead, 2017-12-25:
Find the lines in Marlin that read `HOMEAXIS(X)`. Replace them with the following two lines:
```cpp
    set_axis_is_at_home(X_AXIS);
    SYNC_PLAN_POSITION_KINEMATIC();
```
That should do the trick.

znelson710, 2017-12-31:
That worked!
Thanks!

closed 2017-12-31 by znelson710