MarlinFirmware/Marlin #8859: [1.1.7]Strange spikes on bed temperature sensor

betocabus, 2017-12-21:
Hi huys! I made one corexy 24v, 2 12v fonts in series.

in the tests, I used the Marlin RC2 from my prusa with the changes to corexy, worked very well, tried to reproduce the errors that I'll show below, but is perfect when it's printing. but RC2 has some bugs.

I decided to download some days ago the 1.1.6, but in every fast movments of the motors, this spikes appear (temperatures like -2000 Celsius) . So, I decide to do some firmware tests. First, update to 1.1.7, same problem, tried bugfixes, nothing. Tried 1.1.5, 1.1.4 nothing too. Then went back to te 1.1.0 RC3, worked as well as RC2. Tried 1.1.0 RC4 and the problem started again, but in this case, the spikes appears only on the first movment and the first retract, on the rest of print, everything is fine!
Is there some modification from RC3 to RC4 that can happen this?
[image]

every trial I rewrite all the configuration, reseted eeprom, saved, but I can't find the fix.
Tried with another thermistor outside the bed, changed thermistor code, nothing, with hotend and bed turned off, happens too.

[1.1.7 configurations.zip](https://github.com/MarlinFirmware/Marlin/files/1580358/1.1.7.configurations.zip)
[1.1.0 RC3 configurations.zip](https://github.com/MarlinFirmware/Marlin/files/1580360/1.1.0.RC3.configurations.zip)

Grogyan, 2017-12-21:
I believe the problem is the fact that you are using power supplies in series.
The typical power supplies used for 3D printers aren't really designed to be driven in series.

I would suggest just trying a single 24V power supply

betocabus, 2017-12-21:
But as I said, in Marlin 1.1.0 RC3 everything is perfect, no issue, so I believe it's a software and not hardware error

Grogyan, 2017-12-21:
You need to look at the surge current draw, as Marlin has evolved significantly, with some features, in some areas drawing peak loads.
These loads are easily compensated inside individual power supplies with big capacitors.
When used in series, like with 2 power supplies in series, these capacitors have half the capacitance.  So that capacitance may not be enough to absorb the surge currents.

In any case, try a single 24V power supply and see if the problem continues.

Tannoo, 2017-12-22:
Also can try `bugfix-1.1.x`.

betocabus, 2017-12-22:
Already tried the bugfix as i said. Will try one 24v font to see if the problem is this.

betocabus, 2017-12-23:
Ok, tried with a single 24v PSU, the problem happens again, so I really think is a software problem. New Log: I can see this variation only in simplyfy 3d, in temperature plot. in LCD, everything is fine. Will try another computer, pronterface, etc.

betocabus, 2017-12-23:
changed to pronterface, the problem is gone. Maybe is a simplify3d bug. The last thing I'll do before ignore this problem is check on oscilloscope and put capacitors if needed. So, please don't close this problem before I try for the last time

betocabus, 2017-12-28:
OK, SOLVED. I was looking in ramps and the green connector was black!, the 25A car fuses melted, maybe had bad contact and was heating or power arcs in the bad contacts. changed the green connector, jumped the car fuse and problem SOLVED!

Tannoo, 2017-12-28:
Ooooh. Car fuses! That's nice. I have those big, thin yellow fuses soldered to the board.

Tannoo, 2017-12-28:
Jumping the fuse is never a good thing, though.

betocabus, 2017-12-28:
Off course! Bought today a fuseholder to put In the wire :) thanks!!

closed 2018-02-11 by thinkyhead