MarlinFirmware/Marlin #8861: [2.0.x] UBL MULTIPLE_PROBING

Tannoo, 2017-12-21:
### Bug Report
- Description: UBL not following `MULTIPLE_PROBING` intent
- Expected behaviour: Probe 1st time fast and subsequent probes slow.
- Actual behaviour: All multiple probing is done fast.
- Steps to reproduce:
  - Set `MULTIPLE_PROBING` to 2 or greater
  - Build UBL mesh
  - Watch probing actions.

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1580613/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1580614/Configuration_adv.txt)

Tannoo, 2017-12-21:
I'm finding that `#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)` is not enough to be "slow".

Tannoo, 2017-12-21:
I set `Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)` to `Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 4)` and it's great except when I disable `MULTIPLE_PROBING`, then all probings are "slow".

AnHardt, 2017-12-21:
@Tannoo 
In theory the exact value for `Z_PROBE_SPEED_SLOW` is <= `DEFAULT_ZJERK`.
`DEFAULT_ZJERK` is the highest speed you can start (from 0) the stepper with, without losing steps. (And also the highest speed you can immediately stop to 0)

Why are you complaining about a configuration setting?
Just set it to the value you feel comfortable with.
That's what they are for.
All values are only suggestions, even when calculated from another. Some work pretty well for a lot of machines, some do not. (Is there a 'correct' value for the thermistor type?)

AnHardt, 2017-12-21:
There is a simple test.
If the measured value for the height in `M48` shifts away, you are too fast.

Tannoo, 2017-12-22:
Hmm...well, I'm not sure if you are confused, or trying to confuse me?

Double probing for Z homing works just fine. It probes the first time "fast" and then "slow" a second time.

This is what I thought that the "slow" was supposed to be. But it did not... I was probing twice at the exact same speed.

Tannoo, 2017-12-22:
I corrected that on my end.

In `probe.cpp`:

```cpp
-      // Move down slowly to find bed, not too far
-      if (do_probe_move(-10, Z_PROBE_SPEED_SLOW)) return NAN;

+  // Move down to find bed, not too far
+  if (do_probe_move(-10,
+    #ifdef MULTIPLE_PROBING
+      Z_PROBE_SPEED_SLOW
+    #else
+      Z_PROBE_SPEED_FAST
+    #endif
+    )) return NAN;
```

Sorry if I rose such a stink.

closed 2017-12-22 by Tannoo

thinkyhead, 2017-12-25:
Is that a proposed fix? As near as I can tell, that probe move should always be done at slow speed.