MarlinFirmware/Marlin #8862: Thermistor table #8

ABH10, 2017-12-21:
In Configuration.h and in thermistortables.h it is listed that thermistor table # 8 can be used for a
thermistor like Vishay NTCS0603E3104FXT with a 4.7k pullup resistor.
NTCS0603E3104FXT is a 100 kOhm thermistor with beta_25/85 = 4100K according to this components datasheet.
This seems to be wrong: Table 8 cannot not linearize this combination of thermistor and pullup.
Table 8 seems to be for a 100 kOhm thermistor with beta_25/85 = 3950K using 10 kOhm pullup resistor.

thinkyhead, 2017-12-25:
@DanieleSdei — Would you care to comment on this? This particular table seems to have come from the RepRapPro Huxley fork of Marlin.

boelle, 2019-02-19:
@ABH10 @DanieleSdei is this still an issue?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

Phr3d13, 2019-06-10:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

closed 2019-07-06 by shitcreek