MarlinFirmware/Marlin #8868: RepRap Full Graphic Smart Controller Not Turning On with RUMBA

skc5741, 2017-12-22:
### Bug Report
- Description: ---

I recently purchased a RepRap full discount smart controller to support my RUMBA powered 3D printer. I downloaded U8glib and uncommented the statement `#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`, and yet when I run the program nothing shows up on my screen. I've tried turning the potentiometer to brighten the contrast with no luck. In all the tutorials I've watched it felt like a fairly straightforward process to set up the smart controller, and I can't explain why it is acting this way.

Here is how the controller is mounted as of now to my printer:

[image]

It now had EXT1 on the controller connected to EXT1 on the RUMBA and EXT2 on the controller connected to EXT2 on the RUMBA. I've tried to connect them the other way since it was not working, but I get a clicking signal from the buzzer.

Is there some special setup that the RUMBA requires to get the smart controller working?

Thanks!

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1582437/Configuration.zip)

Bergerac56, 2017-12-22:
Which version of marlin do you use? (looking at your configuration file, it seems that it is 1.1.x, but which one?)

When printer is started, is the screen illuminated or just dark?

If it is illuminated (You have then a good chance that the cabling is correct), do you have a potentiometer on the LCD board? If yes, this pot is modifying the contrast. When too low, you will not see the caracters. Try to modify it.

If none of that, I suspect your cables or the adaptator to the RAMP.

gbyczyns, 2017-12-22:
I also have RUMBA board and a RepRapDiscount LCD. As far as I remember, the labels on the PCB are wrong and I had to swap the cables and then rotate the connectors by 180 degrees. This is a well known problem, just google it.

Bergerac56, 2017-12-22:
Yes indeed. I found this:
The following confusion was encountered with Rumba (board blue LKDR20150602, also see #RepRapDiscount_Smart_Controller) and Smart Controller from kickstarter "Prusa i3 with Diamond Hot end": Rumba-boards small white triangle (pin 1 indication - visible if no plug) is inconsistent with the cable red-line indicating pin-1. And to further "confuse you", the "Smart Controller" plug exp1 must NOT go to exp1 - it should go to exp2 on Rumba. If you orient the plug on Rumba pin1-to-cable-pin1 - and plug exp1-to-exp1 and exp2-to-exp2 you will get the following error meassage when surveying the serial port (via USB on Rumba e.g. with Arduino IDE 1.0.6) (the exact message might differ): 

And:
 Symptoms

The smart controller is beeping and flashing, rumba isn't responding.
Correction

On some rumba boards (e.g. board blue LKDR20150602) the silk screen labeling on EXP1 and EXP2 is wrong (- or the Smart Controller silk screen labeling is wrong?) - and turn each connector by 180°.

skc5741, 2017-12-22:
> turn each connector by 180° 

That was it! Now it lights up and follows my commands!

Thanks so much for your help!

Bergerac56, 2017-12-23:
@skc5741 Hello. If evething is OK for you, you can close this issue.

Thanks

closed 2018-02-11 by thinkyhead