MarlinFirmware/Marlin #8873: Chimera Z offset for fine tuning

RajaaKahel, 2017-12-23:
Hi, What's the best option to use for fine tuning Z offset for Chimera (dual nozzles in one carriage), Is there an option in Marlin 2.0 or maybe using G-code in Tool Change Script?
Thank you.

revilor, 2017-12-24:
If I get your question right you are looking for a way to have different z offsets for both of the Chimera nozzles. Actually there should be no need for this. You have to setup the Chimera with both nozzles level at exactly the same height. Otherwise you will run into problems sooner or later. So if the nozzles of your Chimera are not level that's a mechanical issue which can not be fixed in firmware or by using G-code.

RajaaKahel, 2017-12-24:
revilor, Thanks. Yes, that's my question. It's really hard to level both nozzles at the exact level (I'm talking about less than 0.05mm). I think it can be easily fixed in firmware or G-code, but I'm looking for the best solution.

tcm0116, 2017-12-27:
@RajaaKahel you really need to get the two nozzles aligned as close as possible. If they're not aligned, then the lower nozzle may drag across the part when printing with the higher nozzle.

closed 2018-02-11 by thinkyhead