MarlinFirmware/Marlin #8874: [bugfix-1.1.x] Official Linear Advance test pattern file is printed strange

ikarisan, 2017-12-23:
### Bug Report
- Description: I generated some GCODE files via this website: [http://marlinfw.org/tools/lin_advance/k-factor.html](url) to calibrate linear advance. But neither with Octoprint nor with other software like "S3D" or pronterface it is possible to print the code with Marlin 1.1.7.

- Expected behaviour: Calibration pattern printed as expected. 

- Actual behaviour: The printer print the prime line from X50 Y67.5 to X50 Y167.5 and then starts acting strange.  It moves the nozzle to the first start point at X70 Y67.5 and prints from X70 to X150. But instead of just moving the X axis the printer also moves to Y axis to Y=0. Once reached Y=0 every following line is printed at Y=0! See the following link to [my YouTube](https://youtu.be/zoTm1md4AD8) video that will visualize the problem.

- Steps to reproduce: Flash Marlin 1.1.x(bugfix) [1.1.7] (on an Anet A8 board) and print the attached GCODE file. 

[Marlin_Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1584304/Marlin_Configuration.zip)
[GCODE kfactor_strange.zip](https://github.com/MarlinFirmware/Marlin/files/1584302/kfactor_strange.zip)

Seems to me like a G90 problem. O.o

Sineos, 2017-12-23:
Actually the pattern itself is printed as G91 using relative coordinates. You can try the attached version (same settings as yours) using purely absolute coordinates.
[kfactor.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/1584411/kfactor.gcode.txt)

ikarisan, 2017-12-23:
@Sineos Thanks! Printing with absolute coordinates is working as expected.

Sineos, 2017-12-23:
Well the gcode file looks fine and it is working for 1.1.6 with relative coordinates. Looks like G91 then. Maybe @thinkyhead or @Bob-the-Kuhn can have a look.

ikarisan, 2017-12-23:
Yes, it worked for me on 1.1.6 either.

thinkyhead, 2017-12-25:
The 1.1.7 planner is a bit different from 1.1.6, but the simple behavior of Relative Mode is precisely the same. When coordinates are fetched from the G-code, the current position is added to them before use. That's the only difference, and it should yield good results with the given G-code.

I've tweaked the script at http://marlinfw.org/tools/lin_advance/k-factor.html to produce slightly slimmer output, but it still uses relative mode.

If there's a general issue with `G91`, it should be easy to spot by doing `G1` moves from the host.

chambm, 2017-12-26:
I've used that generator code with 1.1.7 without a problem. Well, I'm not too happy with the linear advance behavior, but the calibration pattern itself prints as expected.

ikarisan, 2017-12-26:
Did you use the tweaked (by thinkyhead two days ago) code or the old one?

chambm, 2017-12-27:
No, I was using the original 1.1.7 (2f2564350d5)

Sineos, 2017-12-27:
The tweak by thinkyhead should not change anything. It just made the gcode a bit leaner but did not change any logic.

ikarisan, 2017-12-27:
Okay, the "tweaked" version prints as expected!
I tried a "dry run" and this test code doesn't work:

`G28; Using "safe homing", sensor is at X:110 and Y:110
G21; set units to mm
G90; use absolute coordinates
G1 X110 Y110; nozzle to center of the bed
G91; realative coordinates
G1 X20 Y0; moves 20mm to the right and Y back a few mm
G1 X20 Y0; moves 20mm to the right and Y back a few mm
G1 X-20 Y0; moves 20mm to the left and Y back a few mm
G1 X-20 Y10; moves 20mm to the left and NOT THE Y AXIS!
G1 X-20 Y0; moves 20mm to the left and Y back a few mm`

This one works:

`G28; Using "safe homing", sensor is at X:110 and Y:110
G21; set units to mm
G90; use absolute coordinates
G1 X110 Y110; nozzle to center of the bed
G91; realative coordinates
G1 X20; moves 20mm to the right and Y IS NOT MOVING
G1 X20; moves 20mm to the right and Y IS NOT MOVING
G1 X-20; moves 20mm to the left and Y IS NOT MOVING
G1 X-20 Y10; moves 20mm to the left and Y IS NOT MOVING
G1 X-20; moves 20mm to the left and Y IS NOT MOVING`

Sineos, 2017-12-27:
> The tweaked version does not contain the "Y0" coordinates"!

For me that should not make any difference. Y0 is correct from the coordinates (talking relative) and as such should not produce any strange movement. It is true, that it is surplus but still it does in no way change the logic.

thinkyhead, 2018-02-11:
Presuming this is solved.

closed 2018-02-11 by thinkyhead