MarlinFirmware/Marlin #8875: stop print won't disable the steppers

amedsoprano, 2017-12-23:
I have a tevo tornado and in the stock firmware whenever i press stop print under the lcd menu the steppers are immediately released so i can pull the print away from the hot nozzle. i recently updated to the new v1.1.7 and that isn't working anymore. when i press stop, it just stops right there and i cannot pull the bed because all the steppers are still enabled.

[Tornado_Marlin-1.1.7.zip](https://github.com/MarlinFirmware/Marlin/files/1584348/Tornado_Marlin-1.1.7.zip)
[Tornado Firmware.zip](https://github.com/MarlinFirmware/Marlin/files/1584349/Tornado.Firmware.zip)

 it would also be nice to have an option to move the head out of the way when stopping (like PARK_HEAD_ON_PAUSE, but on stop)

revilor, 2017-12-24:
As far as I can tell the described  behavior is related to activating ADVANCED_PAUSE_FEATURE. In case ADVANCED_PAUSE_FEATURE is enabled the steppers will not be disabled when the printer is idle. Someone more involved in the Marlin development will have to tell if this a bug or a feature.

amedsoprano, 2017-12-24:
i checked on the stock firmware (the one that does disable the steppers on stop) and the advanced pause settings are the same.

revilor, 2017-12-24:
I'm referring to the files you attached. In Configuration_adv.h from Tornado-Firmware.zip ADVANCED_PAUSE_FEATURE is not enabled (Configuration_adv.h Line 787), but it is in the one from Tornado_Marlin-1.1.7.zip (Configuration_adv.h Line 878)

amedsoprano, 2017-12-24:
I just read that again and it is commented, I'll test that as soon as I finish this print. I also found a setting called SD_FINISHED_STEPPERRELEASE.

thinkyhead, 2017-12-25:
When `SD_FINISHED_STEPPERRELEASE` is set to `true` it should cause the G-code in `SD_FINISHED_RELEASECOMMAND` to be executed 1/2 second after any quick-stop of the steppers, as with Stop Print.

amedsoprano, 2017-12-28:
I disabled advanced pause and it still doesn't disable the steppers on stop, so that isn't the cause.
SD_FINISHED_STEPPERRELEASE is set to true and it doesn't work either.

thinkyhead, 2017-12-31:
@amedsoprano Does `M18` work to turn them off? And if `M18 S2` used to set the stepper off timer (for example) does that cause the steppers to turn off after 2 seconds?

amedsoprano, 2017-12-31:
both `M18` and `M84` work with or without the `S` attribute

msutas, 2018-03-21:
I have the same issue with Marlin 1.1.8

amedsoprano, 2018-04-08:
it hasn't sorry for the long wait, i've been busy upgrading to tmcs

thinkyhead, 2018-04-08:
It appears that this is by design, to prevent things like a delta falling on the print.

dgrat, 2018-07-06:
I can believe that this behavior has an intention, but it is just annoying. Is there a fix to restore the old behavior?

thinkyhead, 2018-07-07:
```cpp
void CardReader::printingHasFinished() {
  planner.synchronize();
  file.close();
  if (file_subcall_ctr > 0) { // Heading up to a parent file that called current as a procedure.
    . . .
  }
  else {
    . . .
    #if ENABLED(SD_FINISHED_STEPPERRELEASE) && defined(SD_FINISHED_RELEASECOMMAND)
      planner.finish_and_disable();
    #endif
    . . .
  }
}

void Planner::finish_and_disable() {
  while (has_blocks_queued() || cleaning_buffer_counter) idle();
  disable_all_steppers();
}
```

boelle, 2019-02-19:
@amedsoprano do you still have an issue?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

Phr3d13, 2019-06-10:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

closed 2019-07-06 by shitcreek