MarlinFirmware/Marlin #8877: [2.0.x] Can't move anymore stepper

GMagician, 2017-12-23:
Description:
  once installed latest 2.x bugfix it seems step motors can't be moved.
 once you move am axes it seems that everything is slowed down (trying to move a servo if moves after several seconds) but stepper never moves.
Expected behaviour:
 once commanded motors should move
Actual behaviour:
  step motors don't move (nor X, Y or Z)
Steps to reproduce:
  go to lcd menu and move a stepper (or even by repetier host)

N.B. bug is caused by #8852 or #8863 because I reverted to version before these merge and now everything work as expected.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1584443/Configuration.zip)

DaveAhrendt, 2017-12-23:
Was just coming here to report this issue. Same problem here with a Folger I3 2020.  Z stepper just vibrates.
Commit 5BC2..... works OK
Commit 97D5..... fails.
I replaced the planner.cpp from 97D5 with the one in 5Bc2 and the steppers work OK

tcm0116, 2017-12-24:
Can you try the fix from #8878 to ensure that is fixes the issue?

DaveAhrendt, 2017-12-24:
Went back to 97d5... an added the patch from #8878.  Steppers are moving fine now.

tcm0116, 2017-12-24:
@DaveAhrendt can you verify that #8878 added to the current bugfix-2.0.x works for you?

DaveAhrendt, 2017-12-24:
Yes, just downloaded the current bugfix and applied the two line patch to planner.cpp.
Stepper motors seem OK now. I can home now and move XYZ.  Didn't try to print anything.

closed 2017-12-25 by thinkyhead