MarlinFirmware/Marlin #8881: Printer keeps pausing on longer prints

jrubinstein, 2017-12-24:
Referencing https://github.com/MarlinFirmware/Marlin/issues/8775#issuecomment-351404151

I have an anet a8 with ramps using the latest bug fix branch. 
During longer prints, the printer will pause, staying in place with the heat still on. Going into the menu and hitting "resume print" works fine, but for longer prints this is untenable. I have increased the buffer to 32, but that has not solved the problem. This happens when I use cura or use slic3r pe. I'm using the SD card for printing exclusively.

Would love any suggestions to solve this issue, I've got some big prints to do! 

Thanks,
James

jrubinstein, 2017-12-24:
attached config files
[configurations-jrubinstein.zip](https://github.com/MarlinFirmware/Marlin/files/1585049/configurations-jrubinstein.zip)

thinkyhead, 2017-12-25:
> the printer will pause, staying in place with the heat still on. Going into the menu and hitting "resume print" works fine, but for longer prints this is untenable.

What version of Marlin are you using? Try `bugfix-1.1.x` and see if the problem still exists.

Marlin doesn't have any feature that spuriously pauses SD printing. How is the wiring to your LCD/SD controller? You may want to enable `SD_CHECK_AND_RETRY` for extra error-checking with the SD card.

jrubinstein, 2017-12-25:
I'm on the latest bugfix now. LCD wiring is solid. I'll try the SD Check setting and see if that helps. 
Merry Christmas :)

jrubinstein, 2017-12-27:
Fingers crossed, I've done a couple of long-ish prints and it's worked okay. Those have been vase mode, so I'll try something I've previously had difficulty with and see how that goes later today.

jrubinstein, 2018-01-01:
Things seem to be working, so I'm closing this issue. Thanks for the help!

closed 2018-01-01 by jrubinstein