MarlinFirmware/Marlin #8884: Z endstop-bed leveling

Thesecky, 2017-12-24:
Hi. I have a question regarding ramps 1.4 and manual bed leveling using LCD. My z endstop is set to trigger obout 0.2mm away from the first left corner of my heated bed. So after i go in my lcd on ramps and begin bed leveling, the first corner doesnt need any adjustments. But when the nozzle moves to the second location, the middle of the bed, i need to move the nozzle down obout 0.3mm, so much, that it triggers my z endstop. Does triggering that z endstop prevents the printer from going lower on the first layer, or does the bed leveling override the first layer z endstop height. Thanks!

Tannoo, 2017-12-25:
>Does triggering that z endstop prevents the printer from going lower on the first layer?

Nope. Depending on how your endstop is setup, be careful on how much lower Z has to go.

Thesecky, 2017-12-25:
So if i understand you correctly. When i level my bed, and i hit the z switch, i can go further down with my encoder wheel to drop the nozzle even more down, if i need to, and the first layer will override that z endstop according to my bed leveling. The switch itself allows me to go down obout 1,5mm after it has been triggered. Btw, i have marlin 1.1.5. Somebody from the facebook group told me that leveling past the z  switch doesnt get recognised when you start printing.

thinkyhead, 2018-02-11:
Correct.

closed 2018-02-11 by thinkyhead