MarlinFirmware/Marlin #8886: Auto leveling reversed on corexy

TheLukey21, 2017-12-24:
so i have finally got my corexy printer working properly in everything except for the auto leveling which seems to work reversed as when i print a grid its squashed at one side and you can see the nozzle is to far away at the opposite end so deposits a round bead which just about sticks to the glass, after doing a little research and making sure each axis moves in the proper direction and having looked for similar issues i found how to enable the debugging so here it is for my auto leveling, if only i could rotate the bed in the firmware just for the auto leveling correction.

Thanks Luke

>>>m111 s255
SENDING:M111 S255
echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
>>>g28
SENDING:G28
echo:G28
>>> gcode_G28
Machine Type: Core
Probe: SERVO PROBE
Probe Offset X:51 Y:-1 Z:-18.50 (Right-Front & Below Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
  current_position=(96.00, 106.00, 33.50) : setup_for_endstop_or_probe_move
> endstops.enable(true)
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, 442.50, 0.00)
  current_position=(0.00, 106.00, 33.50) : sync_plan_position
echo:busy: processing
echo:busy: processing
<<< do_homing_move(X)
Move Away:
>>> do_homing_move(X, -5.00, 0.00)
  current_position=(0.00, 106.00, 33.50) : sync_plan_position
<<< do_homing_move(X)
Home 2 Slow:
>>> do_homing_move(X, 10.00, 25.00)
  current_position=(0.00, 106.00, 33.50) : sync_plan_position
<<< do_homing_move(X)
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 295.00
> home_offset[X] = 0.00
  current_position=(295.00, 106.00, 33.50) :
<<< set_axis_is_at_home(X)
  current_position=(295.00, 106.00, 33.50) : sync_plan_position
  current_position=(295.00, 106.00, 33.50) : > AFTER set_axis_is_at_home
<<< homeaxis(X)
  current_position=(295.00, 106.00, 33.50) : > homeX
>>> homeaxis(Y)
Home 1 Fast:
>>> do_homing_move(Y, -315.00, 0.00)
  current_position=(295.00, 0.00, 33.50) : sync_plan_position
echo:busy: processing
<<< do_homing_move(Y)
Move Away:
>>> do_homing_move(Y, 5.00, 0.00)
  current_position=(295.00, 0.00, 33.50) : sync_plan_position
<<< do_homing_move(Y)
Home 2 Slow:
>>> do_homing_move(Y, -10.00, 25.00)
  current_position=(295.00, 0.00, 33.50) : sync_plan_position
<<< do_homing_move(Y)
>>> set_axis_is_at_home(Y)
For Y axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 210.00
> home_offset[Y] = 0.00
  current_position=(295.00, 0.00, 33.50) :
<<< set_axis_is_at_home(Y)
  current_position=(295.00, 0.00, 33.50) : sync_plan_position
  current_position=(295.00, 0.00, 33.50) : > AFTER set_axis_is_at_home
[... 622 lines cut ...]
Extrapolate [ 2-  2-] (done)
Extrapolate [ 1+  0+] (done)
Extrapolate [ 2-  0+] (done)
Extrapolate [ 1+  3-] (done)
Extrapolate [ 2-  3-] (done)
Extrapolate [ 0+  1+] (done)
Extrapolate [ 3-  1+] (done)
Extrapolate [ 0+  2-] (done)
Extrapolate [ 3-  2-] (done)
Extrapolate [ 0+  0+] (done)
Extrapolate [ 3-  0+] (done)
Extrapolate [ 0+  3-] (done)
Extrapolate [ 3-  3-] (done)
Bilinear Leveling Grid:
      0      1      2      3
 0 +0.098 -0.344 -0.773 -1.152
 1 +0.414 -0.015 -0.410 -0.714
 2 +0.680 +0.273 -0.087 -0.377
 3 +0.810 +0.416 +0.063 -0.196
G29 uncorrected Z:33
 corrected Z:33.51
  current_position=(219.00, 199.00, 33.51) : clean_up_after_endstop_or_probe_move
<<< gcode_G29
X:219.00 Y:199.00 Z:33.51 E:0.00 Count A:83600 B:4000 Z:80400
  current_position=(219.00, 199.00, 33.51) : sync_plan_position

TheLukey21, 2018-01-25:
Any ideas?

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

closed 2018-03-07 by thinkyhead