MarlinFirmware/Marlin #8898: [2.0.x] Marlin bugfix 2.0 - Special Christmas regression edition...

fiveangle, 2017-12-26:
New big block of commits, new big block of regressions :)

anet10 is first failure caught by CircleCI compile of example configs:

https://circleci.com/gh/fiveangle/Marlin/72?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link

[My config](https://github.com/fiveangle/Marlin/tree/bf2-xmas_regression_edition/Marlin) fails with the following <details><summary>PIO console output:</summary>
```
from /Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/Snooze.cpp:30:
/Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h:79:3: error: expected unqualified-id before numeric constant
} SPEED;
^
In file included from /Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/SnoozeBlock.h:34:0,
from /Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/Snooze.h:35,
from /Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/Snooze.cpp:30:
/Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h: In function 'void start_lptmr_systick()':
/Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h:96:5: error: 'SIM_SCGC5' was not declared in this scope
SIM_SCGC5 |= SIM_SCGC5_LPTIMER;
^
/Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h:96:18: error: 'SIM_SCGC5_LPTIMER' was not declared in this scope
SIM_SCGC5 |= SIM_SCGC5_LPTIMER;
^
/Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h:98:5: error: 'LPTMR0_CSR' was not declared in this scope
LPTMR0_CSR = 0;
^
/Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h:100:5: error: 'LPTMR0_PSR' was not declared in this scope
LPTMR0_PSR = LPTMR_PSR_PCS( LPTMR_OSCERCLK ) | LPTMR_PSR_PRESCALE(3);
^
compilation terminated due to -fmax-errors=5.
avr-ar rcs ".pioenvs/teensy20/lib/libAdafruit NeoPixel.a" ".pioenvs/teensy20/lib/Adafruit NeoPixel/Adafruit_NeoPixel.o" ".pioenvs/teensy20/lib/Adafruit NeoPixel/esp8266.o"
avr-ranlib .pioenvs/teensy20/lib/libLiquidTWI2.a
In file included from /Users/speedster/.platformio/packages/toolchain-atmelavr/avr/include/avr/iousb1286.h:38:0,
from /Users/speedster/.platformio/packages/toolchain-atmelavr/avr/include/avr/io.h:192,
from /Users/speedster/.platformio/packages/toolchain-atmelavr/avr/include/avr/interrupt.h:38,
from /Users/speedster/.platformio/packages/framework-arduinoteensy/cores/teensy/WProgram.h:18,
from /Users/speedster/.platformio/packages/framework-arduinoteensy/cores/teensy/Arduino.h:1,
from /Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/SnoozeBlock.h:33,
from /Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/SnoozeBlock.cpp:27:
/Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h:79:3: error: expected unqualified-id before numeric constant
} SPEED;
^
In file included from /Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/SnoozeBlock.h:34:0,
from /Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/SnoozeBlock.cpp:27:
/Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h: In function 'void start_lptmr_systick()':
/Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h:96:5: error: 'SIM_SCGC5' was not declared in this scope
SIM_SCGC5 |= SIM_SCGC5_LPTIMER;
^
/Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h:96:18: error: 'SIM_SCGC5_LPTIMER' was not declared in this scope
SIM_SCGC5 |= SIM_SCGC5_LPTIMER;
^
/Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h:98:5: error: 'LPTMR0_CSR' was not declared in this scope
LPTMR0_CSR = 0;
^
/Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h:100:5: error: 'LPTMR0_PSR' was not declared in this scope
LPTMR0_PSR = LPTMR_PSR_PCS( LPTMR_OSCERCLK ) | LPTMR_PSR_PRESCALE(3);
^
compilation terminated due to -fmax-errors=5.
avr-ranlib ".pioenvs/teensy20/lib/libAdafruit NeoPixel.a"
In file included from /Users/speedster/.platformio/packages/toolchain-atmelavr/avr/include/avr/iousb1286.h:38:0,
from /Users/speedster/.platformio/packages/toolchain-atmelavr/avr/include/avr/io.h:192,
from /Users/speedster/.platformio/packages/toolchain-atmelavr/avr/include/avr/interrupt.h:38,
from /Users/speedster/.platformio/packages/framework-arduinoteensy/cores/teensy/WProgram.h:18,
from /Users/speedster/.platformio/packages/framework-arduinoteensy/cores/teensy/Arduino.h:1,
from /Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/SnoozeAlarm.cpp:8:
/Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h:79:3: error: expected unqualified-id before numeric constant
} SPEED;
^
In file included from /Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/SnoozeBlock.h:34:0,
from /Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/SnoozeAlarm.h:33,
from /Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/SnoozeAlarm.cpp:9:
[... 39 lines cut ...]
^
/Users/speedster/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h:100:5: error: 'LPTMR0_PSR' was not declared in this scope
LPTMR0_PSR = LPTMR_PSR_PCS( LPTMR_OSCERCLK ) | LPTMR_PSR_PRESCALE(3);
^
compilation terminated due to -fmax-errors=5.
*** [.pioenvs/teensy20/lib/Snooze/utility/SnoozeAudio.o] Error 1
========================== [ERROR] Took 18.10 seconds ==========================
================================== [SUMMARY] ==================================
Environment megaatmega2560          	[SKIP]
Environment megaatmega1280          	[SKIP]
Environment anet10                  	[SKIP]
Environment DUE                     	[SKIP]
Environment LPC1768                 	[SKIP]
Environment LPC1768_debug_and_upload	[SKIP]
Environment rambo                   	[SKIP]
Environment sanguino_atmega644p     	[SKIP]
Environment STM32F1                 	[SKIP]
Environment teensy20                	[ERROR]
Environment teensy35                	[SKIP]
========================== [ERROR] Took 18.10 seconds ==========================
```
</details>
.

[These are the commit diffs](https://github.com/fiveangle/Marlin/compare/bf-pb_mbl_lcd...fiveangle:bf2-xmas_regression_edition) between last known good (12/15) and today.

fiveangle, 2018-01-02:
@ejtagle - I finally got time to bisect this out and your changes for https://github.com/MarlinFirmware/Marlin/commit/4a8fd8ae928492d622b6b0d67fd051da54065a41 resulted in this regression.

Here is bf2.0 at your commit with my configs added.  Reverting your #8853 changes for USB mass storage support for Due/etc allows compilation.

It's a bit beyond my capabilities to figure out how to resolve as I don't see how the Due/Re-ARM changes you made are being picked up by a config using teensypp20.

-=dave

fiveangle, 2018-01-02:
Looks like it breaks all teensy20pp targets:

From CircleCI test here: https://circleci.com/gh/fiveangle/Marlin/75?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link
```
testing teensy20 targets...
use_example_configs makibox
```
[...]
```
compilation terminated due to -fmax-errors=5.
*** [.pioenvs/teensy20/lib02a/Snooze/utility/Snoozelc5vBuffer.o] Error 1
========================== [ERROR] Took 14.72 seconds ==========================

================================== [SUMMARY] ==================================
Environment megaatmega2560          	[SKIP]
Environment megaatmega1280          	[SKIP]
Environment anet10                  	[SKIP]
Environment DUE                     	[SKIP]
Environment LPC1768                 	[SKIP]
Environment LPC1768_debug_and_upload	[SKIP]
Environment rambo                   	[SKIP]
Environment sanguino_atmega644p     	[SKIP]
Environment STM32F1                 	[SKIP]
Environment teensy20                	[ERROR]
Environment teensy35                	[SKIP]
========================== [ERROR] Took 14.72 seconds ==========================
Exited with code 1
```

<details><summary>Full log of above makibox build...</summary>

```
testing teensy20 targets...
use_example_configs makibox
.piolibdeps/U8glib-HAL_ID1932/src/lib/u8g_com_arduino_fast_parallel.c:73:0: warning: "PIN_D0" redefined
#define PIN_D0 8
^
In file included from /home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/wiring.h:33:0,
from /home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/WProgram.h:19,
from /home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/Arduino.h:1,
from .piolibdeps/U8glib-HAL_ID1932/src/lib/u8g_com_arduino_fast_parallel.c:69:
/home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/core_pins.h:460:0: note: this is the location of the previous definition
#define PIN_D0  0
^
.piolibdeps/U8glib-HAL_ID1932/src/lib/u8g_com_arduino_fast_parallel.c:74:0: warning: "PIN_D1" redefined
#define PIN_D1 9
^
In file included from /home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/wiring.h:33:0,
from /home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/WProgram.h:19,
from /home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/Arduino.h:1,
from .piolibdeps/U8glib-HAL_ID1932/src/lib/u8g_com_arduino_fast_parallel.c:69:
/home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/core_pins.h:461:0: note: this is the location of the previous definition
#define PIN_D1  1
^
.piolibdeps/U8glib-HAL_ID1932/src/lib/u8g_com_arduino_fast_parallel.c:75:0: warning: "PIN_D2" redefined
#define PIN_D2 10
^
In file included from /home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/wiring.h:33:0,
from /home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/WProgram.h:19,
from /home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/Arduino.h:1,
from .piolibdeps/U8glib-HAL_ID1932/src/lib/u8g_com_arduino_fast_parallel.c:69:
/home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/core_pins.h:462:0: note: this is the location of the previous definition
#define PIN_D2  2
^
.piolibdeps/U8glib-HAL_ID1932/src/lib/u8g_com_arduino_fast_parallel.c:76:0: warning: "PIN_D3" redefined
#define PIN_D3 11
^
In file included from /home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/wiring.h:33:0,
from /home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/WProgram.h:19,
from /home/circleci/.platformio/packages/framework-arduinoteensy/cores/teensy/Arduino.h:1,
[... 367 lines cut ...]
^
/home/circleci/.platformio/packages/framework-arduinoteensy/libraries/Snooze/utility/clocks.h:102:5: error: 'LPTMR0_PSR' was not declared in this scope
LPTMR0_PSR = LPTMR_PSR_PCS( LPTMR_OSCERCLK ) | LPTMR_PSR_PRESCALE(3);
^
compilation terminated due to -fmax-errors=5.
*** [.pioenvs/teensy20/lib02a/Snooze/utility/Snoozelc5vBuffer.o] Error 1
========================== [ERROR] Took 14.72 seconds ==========================

================================== [SUMMARY] ==================================
Environment megaatmega2560          	[SKIP]
Environment megaatmega1280          	[SKIP]
Environment anet10                  	[SKIP]
Environment DUE                     	[SKIP]
Environment LPC1768                 	[SKIP]
Environment LPC1768_debug_and_upload	[SKIP]
Environment rambo                   	[SKIP]
Environment sanguino_atmega644p     	[SKIP]
Environment STM32F1                 	[SKIP]
Environment teensy20                	[ERROR]
Environment teensy35                	[SKIP]
========================== [ERROR] Took 14.72 seconds ==========================
Exited with code 1
```

</summary>

Bob-the-Kuhn, 2018-01-06:
Try deleting the file **uotghs_device_due.c**

I don't know why deleting this file allows it to compile.  Renaming it doesn't fix it.  I don't see any problems with the file itself.  The code itself is properly guarded with  a `#ifdef ARDUINO_ARCH_SAM`

fiveangle, 2018-01-07:
Confirmed resolved with #9069

closed 2018-01-07 by fiveangle