MarlinFirmware/Marlin #8900: Marlin 1.8 Anet A8 example_configurations

PhilippeGalid, 2017-12-26:
### Bug Report
Hi today i want to install Marlin 1.8 with Anet A8 example_configurations and i get in 
Arduino: 1.8.5 (Windows 10) this error.
-------------------------------------------------------------------------------------
Arduino: 1.8.5 (Windows 10), Board: "Anet V1.0"

In file included from sketch\MarlinConfig.h:30:0,

                 from C:\Users\phili\Desktop\Marlin-1.1.8 Meine\Marlin-1.1.8 Meine.x\Marlin\Marlin.ino:31:

Configuration.h:1778: error: #endif without #if

 #endif // CONFIGURATION_H

  ^

exit status 1
#endif without #if
---------------------------------------------------------------------------------------

What can i do ?
Marlin v1.7 works perfect.
And thank you for Marlin and your hard work.

kind regards from Vienna/Austria
Philippe

olivCode, 2017-12-26:
Hello Philippe,

PhilippeGalid, 2017-12-26:
I use the A8 example configuration files from 1.8 and there i have the problem with Arduino ?

olivCode, 2017-12-26:
I use them too and just updated my tutorial this afternoon for my facebook group.
Everything is ok on A8 for several people. 
I suggest you restart from the original, then copy example configuration files then customize for your needs then reflash

The group like the tutorial are in french in case you need.

PhilippeGalid, 2017-12-26:
Thanks do you have a Facebook Link ?

olivCode, 2017-12-27:
https://www.facebook.com/groups/Anet3DFrancophone/

PhilippeGalid, 2017-12-27:
Merci :-)

closed 2018-02-11 by thinkyhead