MarlinFirmware/Marlin #8902: [FR] Update Ultimaker Pins for TMC2130 Compatability

AcrimoniousMirth, 2017-12-26:
Once again a great thank you to the wonderful devs who keep this awesome firmware going!
And a merry Christmas to you all.

I’ve got my hands on some TM2130 silent stepsticks and while I could put them all in my RAMPS based machine I’d like to put two in my GT2560.
Now I’m not experienced in programming enough to sort this myself. 
The GT2560 motherboard has very few spare pins and the TMC2130 needs a few to run its SPI interface (the EMF endstop signals can run directly into the original endstop connectors). However I think it should be possible to reuse the max endstops and possibly spare motor driver pins (?) to communicate with the TMCs.
So would it be possible to add a few definitions to the end for TMC2130 control that override these pins?

Many thanks!

teemuatlut, 2017-12-27:
To me it sounds like you're asking for software SPI interface for the drivers.
Does the board not have HW SPI pins broken out?

thinkyhead, 2018-02-11:
`TMC_USE_SW_SPI`

closed 2018-02-11 by thinkyhead

AcrimoniousMirth, 2018-02-11:
@teemuatlut no, the board is fairly minimal, hence my question. 

@thinkyhead okay, but will that allow me to set which pins are used? No point enabling that if it doesn’t correspond to the righ pins on the board.

thinkyhead, 2018-02-14:
> …will that allow me…?

```cpp
/**
 * Use software SPI for TMC2130.
 * The default SW SPI pins are defined the respective pins files,
 * but you can override or define them here.
 */
//#define TMC_USE_SW_SPI
//#define TMC_SW_MOSI       -1
//#define TMC_SW_MISO       -1
//#define TMC_SW_SCK        -1
```

roosevelt155, 2018-02-18:
I am wondering the same. I am using a Cohesion3D Remix board and am not sure what the default pins are set to. I am also not sure what these "MOSI", "MISO", and "SCK" pins correspond to what pins on the TMC2130. I am only finding the SCK pin on my TMC2130.

thinkyhead, 2018-02-19:
A Google Image search shows us
[image]

- SDO -> MISO
- SDI -> MOSI
- CS -> digital pins
- SCK -> SCK

A Google search for SCK serial leads to https://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus where we find
- MISO = Master In Slave Out
- MOSI = Master Out Slave In
- SCK = Serial Clock
- CS = Chip Select