MarlinFirmware/Marlin #8903: about Marlin 1.1X “Thermal Runaway”

lukasaoka, 2017-12-26:
### Feature Request
I download the Marlin's 1.1.x FW ( I try 1.1.6~1.1.8 ) to my 3D print (The print is made by myself, the motherboard is RAMPS_14_EFB ). but when the print is processing ,  the print suddenly stop and show "Thermal Runaway" on the LCD, press any key not respond except the reset button. I carefully read the comment in the configuration.h file, and modify the  "Thermal Settings" segment in the file of configuration_adv.h. but the trouble is still not resolved, the following is modify segment in the file of configuration_adv.h:

//=================================================================
//=============================Thermal Settings  =========================
//==================================================================
...........................
/**
 * Thermal Protection provides additional protection to your printer from damage
 * and fire. Marlin always includes safe min and max temperature ranges which
 * protect against a broken or disconnected thermistor wire.
 *
 * The issue: If a thermistor falls out, it will report the much lower
 * temperature of the air in the room, and the the firmware will keep
 * the heater on.
 *
 * The solution: Once the temperature reaches the target, start observing.
 * If the temperature stays too far below the target (hysteresis) for too
 * long (period), the firmware will halt the machine as a safety precaution.
 *
 * If you get false positives for "Thermal Runaway", increase
 * THERMAL_PROTECTION_HYSTERESIS and/or THERMAL_PROTECTION_PERIOD
 */
#if ENABLED(THERMAL_PROTECTION_HOTENDS)
  #define THERMAL_PROTECTION_PERIOD 120        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 4     // Degrees Celsius

  /**
   * Whenever an M104, M109, or M303 increases the target temperature, the
   * firmware will wait for the WATCH_TEMP_PERIOD to expire. If the temperature
   * hasn't increased by WATCH_TEMP_INCREASE degrees, the machine is halted and
   * requires a hard reset. This test restarts with any M104/M109/M303, but only
   * if the current temperature is far enough below the target for a reliable
   * test.
   *
   * If you get false positives for "Heating failed", increase WATCH_TEMP_PERIOD
   * and/or decrease WATCH_TEMP_INCREASE. WATCH_TEMP_INCREASE should not be set
   * below 2.
   */
  #define WATCH_TEMP_PERIOD 120                // Seconds
  #define WATCH_TEMP_INCREASE 2               // Degrees Celsius
#endif

/**
 * Thermal Protection parameters for the bed are just as above for hotends.
 */
#if ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD 20    // Seconds
  #define THERMAL_PROTECTION_BED_HYSTERESIS 2 // Degrees Celsius

  /**
   * As described above, except for the bed (M140/M190/M303).
   */
  #define WATCH_BED_TEMP_PERIOD 100                // Seconds
  #define WATCH_BED_TEMP_INCREASE 2               // Degrees Celsius
#endif

..............

------------------------------------------------------------
Pls give me hand! Thank u very much!

DrJuJu, 2017-12-29:
I've also experienced the same issue, with 1.1.8. I've tried adjusting the HotEnd settings in configuration_adv.h to what I thought were reasonable values i.e.

#define THERMAL_PROTECTION_PERIOD 90 // Seconds
#define THERMAL_PROTECTION_HYSTERESIS 10  // Degrees Celsius
&
#define WATCH_TEMP_PERIOD 90 // Seconds
#define WATCH_TEMP_INCREASE 10 // Degrees Celsius

The only way to continue was to disable thermal protection for the hotend in configuration.h

When this was done, I noticed that the hotend temperature started dropping 10~15 degrees after the print had started, which then stabalized at that level. 

Obviously this was causing the thermal runaway, due to the hotend dropping below the required settings, but I don't know why ?

revilor, 2017-12-29:
Can you please attach your Configuration.h file. And did you run PID autotune for the hotend?

DrJuJu, 2017-12-30:
Configs attached, As you will see this is for a DBot printer running on a Rumba board, and yes I ran PID for both hotend and bed.

Julian

Sineos, 2017-12-30:
Some thoughts:

- The default settings are very conservative. Even opening the window might kick it
- Watch temperature with a print host. How big is the drop? Does it settle / recover? ...
- Insulate the hotend to avoid the fan killing your temps, i.e. check the fan if it is blowing against the hotend
- Check your temp sensor. Vibration and movement might cause a temp sensor to fail if it is already pre-damaged, e.g. broken cable, bad connectors
- Check cable routing. Running the temp sensor cable along wires with heavier load (heating, stepper) might cause problems

DrJuJu, 2017-12-30:
This occurs before the part fan kicks in, (for PLA) and no fan for ABS. 

The E3D aero fan kicks in @ 50 degrees & the temp reading stays solid, its only when it starts printing does the temp drop ~10\15 degrees, where it stabilizes, There are no cabling\connector and I can manually run the positioning with no variance in temperature. I've even added some 10nF caps to the sensor ip's....

lukasaoka, 2017-12-31:
I try it many times, when i use the following value, the trouble has resolved:
-----------------------------------------------------------------------------------
#if ENABLED(THERMAL_PROTECTION_HOTENDS)
  #define THERMAL_PROTECTION_PERIOD 180        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 10     // Degrees Celsius

  /**
   * Whenever an M104, M109, or M303 increases the target temperature, the
   * firmware will wait for the WATCH_TEMP_PERIOD to expire. If the temperature
   * hasn't increased by WATCH_TEMP_INCREASE degrees, the machine is halted and
   * requires a hard reset. This test restarts with any M104/M109/M303, but only
   * if the current temperature is far enough below the target for a reliable
   * test.
   *
   * If you get false positives for "Heating failed", increase WATCH_TEMP_PERIOD
   * and/or decrease WATCH_TEMP_INCREASE. WATCH_TEMP_INCREASE should not be set
   * below 2.
   */
  #define WATCH_TEMP_PERIOD 120               // Seconds
  #define WATCH_TEMP_INCREASE 4               // Degrees Celsius
#endif

/**
 * Thermal Protection parameters for the bed are just as above for hotends.
 */
#if ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD 240   // Seconds
  #define THERMAL_PROTECTION_BED_HYSTERESIS 10 // Degrees Celsius

  /**
   * As described above, except for the bed (M140/M190/M303).
   */
  #define WATCH_BED_TEMP_PERIOD 120                // Seconds
  #define WATCH_BED_TEMP_INCREASE 4               // Degrees Celsius
#endif

lukasaoka, 2018-01-01:
But when I use the printer today,  another trouble is came back: When the preheat of printer is completed, the hot-end can reach the target temperature.But while the printing is beginning, the actual temperature will drop, so that, today, the target temperature of hot-end is 200 degree, and the actual temperature has been reached and  stabilized;  but when the print job is beginning, the hot-end temperature will be dropping to 180~190 degree and can't be stabilized. I think the failure my be caused by the power supply because I adjust the current of stepper motor to 1.2A(max) before. I will  make time to proved it.
-------------------------------------------------------------------------------------
If you meet this failure, please give me hand! thank u very much!

lukasaoka, 2018-01-01:
But when I use the printer today, another trouble is came back: When the preheat of printer is completed, the hot-end can reach the target temperature.But while the printing is beginning, the actual temperature will drop, so that, today, the target temperature of hot-end is 200 degree, and the actual temperature has been reached and stabilized; but when the print job is beginning, the hot-end temperature will be dropping to 180~190 degree and can't be stabilized. I think the failure my be caused by the power supply because I adjust the current of stepper motor to 1.2 A(max) before. I will make time to proved it.
If any one meet this failure, please give me hand. Thank u very much!

DrJuJu, 2018-01-01:
You are seeing what I was seeing, but I don't have power supply issues, as I'm using a 20A supply for the electronics & the bed is mains power driven.

lukasaoka, 2018-01-03:
I use a 40A /12V supply for the print system, but the failure still happened.！

trgtylcnky, 2018-01-03:
I am having a similar issue. What I noticed is PWM value stays @62 when the temp is near the target (when it is far from target, it is 127)

I looked through the firmware but didn't see anything limits the pwm value. This might be a bug?

trgtylcnky, 2018-01-03:
I just realized, when I increase PID_MAX value to 255 (from 125), that 62 also increased to 127 . I don't know why it is the half of the value but It seems someone missed something while writing those complicated calculations...

lukasaoka, 2018-01-05:
I also increased PID_MAX value to 255 in the marlin 1.1.7, but the temperature increased very fast. The target was 210℃，but the pre-heat reached 250℃, so I decrease the PID_MAX to 120, the trouble resolved.

trgtylcnky, 2018-01-05:
@lukasaoka This problem could be solved by tuning PID parameters instead of limiting the max output. Try auto tuning: http://reprap.org/wiki/G_code#M303:_Run_PID_tuning

closed 2018-02-11 by thinkyhead

homasholz, 2018-09-20:
I also have this problem that the temperature decreases about 10 degrees and settles there (when the second fan starts). I already have PID_MAX at 255 as default. 
PID autotune does not solve this issue. 
When setting manually 200 degree and then enabling the second fan, the temerature settles back to 200 in some seconds, but while printing it does not.

Any advice?

thx

david38002, 2019-01-17:
BACK UP A MINUTE:
Try removing the blower extension and see if the temperature doesn't stay a lot more stable, thus eliminating the thermal runaway.

micahobaca, 2019-03-15:
i figured out the problem, get rid of the mosfets, they make the heat bed have thermal runway errors

Yondertag, 2020-01-13:
Is there any other way to do this because I don't want my hotbed wire contacts to melt the plastic around the screw-in spots on the motherboard, again.

micahobaca, 2020-01-14:
things that removed my thermal runaway errors, 
1. got rid of mosfet
2. checked my power supply voltage and found out it was only giving off 10v even with the dashpot turned all the way up, i spent 20$ on amazon and bought a better power supply with a fan

Yondertag, 2020-01-14:
Do I HAVE to remove the mosfet? And are you sure there is nothing else I can do to fix it? I really don't want to put my printer at risk again by removing the mosfet.

micahobaca, 2020-01-15:
i would, a mosfet let to my circuit board burning out and and my heat bed not heating up all athe way and thermal errors, also my power supply was faulty. leave the mosfet on and check your power supply voltage, if its above 12vthen get rid of the mosfet, if its below 12v try to adjust the dashpot for 12v, i broke my dashpot trying to adjust it because it was cheap chineese made crap,