MarlinFirmware/Marlin #8904: Marlin 1.1.8 Auto heating over 24000 degrees

anraf1001, 2017-12-26:
After updating my printer's firmware to 1.1.8 it want to heat the bed over 24k degrees every time when my OctoPrint is connecting to it. I have reuploaded an older version with the same config and there were no problems.

Erfquake, 2017-12-28:
I had the same issue with Marlin 1.1.8 on my Tevo Tornado connected to octopi (latest version).  Whenever I used the custom menu to move the axis to the corners or when I issue a G28 command via octoprint to home the printer. Look at the video I have recorded, glad to see I was not the only one.  

I also noticed my octopi was displaying target temperature of 5 digits, at one point my bed temperature was reaching 150 C I was trying to level it before I noticed the temperature spike.

https://www.youtube.com/watch?v=4iKnjEG4ihE
https://www.youtube.com/watch?v=0hTs6TXOVOA

thinkyhead, 2018-02-11:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

boelle, 2019-02-19:
@anraf1001 Please test with the latest  `bugfix-2.0.x`

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead