MarlinFirmware/Marlin #8905: Does not build out-of-the-box for CR-10

RudolphRiedel, 2017-12-26:
### Compile Error
I only copied _Bootscreen.h, Configuration.h and Configuration_adv.h from example_configurations/Creality/CR-10 over and the result is this:

Compiling 'Marlin' for 'Sanguino w/ ATmega1284 or ATmega1284P (16 MHz)'
 
MarlinConfig.h:40: In file included from
Marlin.ino:30: from
 
SanityCheck.h: 418:6: error: #error "ADVANCED_PAUSE_FEATURE requires NOZZLE_PARK_FEATURE"
   #error "ADVANCED_PAUSE_FEATURE requires NOZZLE_PARK_FEATURE"
 
MarlinConfig.h:40: In file included from
blinkm.cpp:27: from
 
SanityCheck.h: 418:6: error: #error "ADVANCED_PAUSE_FEATURE requires NOZZLE_PARK_FEATURE"
Error compiling project sources
Build failed for project 'Marlin'
   #error "ADVANCED_PAUSE_FEATURE requires NOZZLE_PARK_FEATURE"

v1ru3, 2017-12-27:
they moved the pause for M600 under configuration. You need to uncomment NOZZLE_PARK_FEATURE and define x,y,z.

RudolphRiedel, 2017-12-27:
Fixing that individually is not the issue. :-)

v1ru3, 2017-12-28:
not sure i understand.

fiveangle, 2017-12-28:
The error IS your answer:

`#error "ADVANCED_PAUSE_FEATURE requires NOZZLE_PARK_FEATURE"`

RudolphRiedel, 2017-12-28:
Yes, the issue is that it does not build with the provided example configuration and the fix would be to provide a valid example configuration for the next release.
As of why it fails, almost no problem at all.

tcm0116, 2017-12-28:
I missed that in #8864. I'm working on cleaning up #7107, which I've included the change to the CR-10 example config. It should hopefully be merged in the next few days  Sorry about that.

thinkyhead, 2017-12-28:
> the fix would be to provide a valid example configuration for the next release. ;-)

D'oh!

> #7107 … should hopefully be merged in the next few days

I'm testing it now and will merge it today.

tcm0116, 2017-12-28:
>I'm testing it now and will merge it today.

Don't merge it quite yet. I'm working on adding the configurable
load/unload lengths and I found some issues with the LCD not being
displayed at the start of the action.

thinkyhead, 2017-12-28:
Does the LCD issue only manifest on the 2nd and later invocations?

tcm0116, 2017-12-28:
No. It manifests at the start of the process. For change filament, if you
have the home before filament change option enabled and while it's homing,
and for unload filament while it does the z lift. In both cases, the LCD
(full graphic in my case), will be in an ambiguous state until the next
step of the process.

I was working on it this morning, but I had to leave for the day. I'll get
back to it this evening.

thinkyhead, 2017-12-28:
> I was working on it this morning, but I had to leave for the day. I'll get back to it this evening.

Some thoughts…

- If the requested nozzle isn't heated it might be better to just start heating it and jump to the wait for heating screen. This will probably simplify some code also.
- When the nozzle is done heating, it shouldn't start loading right away, but should beep and go to the "Click to Load" or "Click to Unload" screen. These screens should beep according to the number of beeps setting. We should give the beep a distinctive sound, audible but not annoying. Something sort of like this…
```gcode
M300 S880 P50
G4 S1
M300 S660 P50
G4 S1
M300 S880 P50
G4 S1
M300 S660 P50
G4 S1
M300 S880 P50
G4 S1
M300 S660 P50
G4 S1
M300 S880 P50
G4 S1
M300 S660 P50
G4 S1
```

boelle, 2019-02-19:
@tcm0116 is this one "fixed"?

tcm0116, 2019-02-20:
@boelle - I would think so, but it's been almost a year.

boelle, 2019-02-20:
@tcm0116 just stiring the pot so to speak as to try and figure what issues have been solved and could be closed

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-03-16:
@RudolphRiedel status?

RudolphRiedel, 2019-03-17:
I have no idea right now since it has been a while since I built a Marlin for one of my printers.
Not this long but maybe half a year.

I was about to check out a new archive and I am not sure which one.
The 1.1.x one seems to be getting updates despite the statement that it won't after 1.1.9.
The bugfix-1.1.x might be  the one to go then but "This branch is 44 commits ahead, 48 commits behind 1.1.x. "
Then the bugfix-2.0.x maybe? "This branch is 3534 commits ahead, 1816 commits behind 1.1.x."

closed 2019-03-17 by RudolphRiedel

RudolphRiedel, 2019-03-17:
Okay, I went for 2.0, changed env_default in plattformio.ini to melzi.optiboot, copied over the files for the CR-10 and apart from two warnings it just built.
Then I updated my vscode and platformio, issued a Clean and then a Build again.
And it just built, data is 91,5% (14984 / 16384) and program is 93,1% (121064 / 130048).

boelle, 2019-03-17:
> Then the bugfix-2.0.x maybe?

yes, its updated most and it has config files for CR-10 https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/config/examples/Creality/CR-10

boelle, 2019-03-17:
as long you get a success in vscode after build you are good :-)