MarlinFirmware/Marlin #8906: m420 command causes rough surface oncircles

Alexanderio99, 2017-12-26:
- Description: 
on my anycubic kossel plus and marlin 1.1.4 I use command
`M420 S1 Z10` before printing and get result as on the picture (some extra extrusion on the end of segments of circles)
[image]
[image]

- Expected behaviour: 
should work fine with smooth wall

- Actual behaviour: 
uneven surface of the walls

- Steps to reproduce:
use `M420 S1 Z10` after bed leveling and print cilinder

- Workaround
if I use `M420 S1 Z5000` instead `M420 S1 Z10` everything works fine

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1587820/Marlin.zip)

thinkyhead, 2018-02-11:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

Alexanderio99, 2018-02-11:
tested on bugfix-1.1.x - works fine. Thanks)

closed 2018-02-19 by thinkyhead