MarlinFirmware/Marlin #8909: Linear Advance K-Factor Calibration G-code Generator Not Working

Custom3DProducts, 2017-12-26:
Pressing Generate G-code then waiting or immediately following it by pressing download as file does nothing.
Tried on Mac with Chrome and Safari 
Tried on Windows with Chrome and Microsoft Edge
With no difference in result

Sineos, 2017-12-27:
There is a typo in the script, thats why it is not working:
https://github.com/MarlinFirmware/MarlinDocumentation/pull/120

@thinkyhead or @Roxy-3D or @Bob-the-Kuhn: Please merge PR

ArtemKuchin, 2017-12-27:
http://marlinfw.org/tools/lin_advance/K-factor.js

Cebtenzzre, 2017-12-28:
@ArtemKuchin In the Chrome developer console, I ran `var _loadScript = function(path){ var script= document.createElement('script'); script.type= 'text/javascript'; script.src= path; document.head.appendChild(script); }` and then `_loadScript('k-factor.js')` as a workaround.

Sineos, 2017-12-28:
@thinkyhead 
Please...PLEASE merge https://github.com/MarlinFirmware/MarlinDocumentation/pull/120 or just swap the K for a k.
Another thread or pull request about a lower case 'k' and I will mutilate my printer with a fire axe and kill my dog. 

[image]

plunderfest, 2017-12-28:
Not sure I understand the adding of a lowercase k, and how that resolves this?

Cebtenzzre, 2017-12-29:
The pull request was merged and as far as I can tell the issue has been resolved.  Someone should close this.

closed 2017-12-29 by Custom3DProducts