MarlinFirmware/Marlin #8910: [BR] NO_VOLUMETRICS Compile Error ULTRALCD.CPP

houseofbugs, 2017-12-26:
I had to comment out the filament menu and 

[image]
[image]

I fixed it and will merge changes into the codebase.

v1ru3, 2017-12-27:
I did the same this morning, however when I uploaded to my CR-10 the E motor makes a continuous whine noise and it doesn't move when commands are issued for it (gcode or via the screen).

Did you have any issues like that on yours?

Custom3DProducts, 2017-12-28:
@v1ru3 Same issue with mine when I defined NO_VOLUMETRICS hopefully its fully patched soon

Custom3DProducts, 2017-12-29:
@v1ru3 Just uploaded the latest bugfix and #define NO_VOLUMETRICS doesnt appear to be effecting the extruder anymore

v1ru3, 2017-12-29:
Thanks for the follow up. I agree, they actually fixed it the day after.
However, the M851 issue still seems to linger with bilinear bed leveling though.
On mine it varies with M851 Z (if diferent z is specified, the M851 value is different)... good times!

Custom3DProducts, 2017-12-29:
@v1ru3 is it a rounded value? I set my Z on the controller and a M851 shows it rounded (3.875 to 3.88) but it is always correct on the controller

closed 2017-12-30 by houseofbugs

v1ru3, 2017-12-30:
my issue with the offset is more like this
M420 Z5
G28
G29
the offset is 1.250

restart/reset
M420 Z10
G28
G29
the offset is 0.999