MarlinFirmware/Marlin #8914: Build failure for MKS SBASE using bugfix-2.0.x

Jamin82, 2017-12-27:
### Compile Error  - I am very new to github  and platformio so feel a bit lost.. i tried searching online and finding solutions with little luck so am reaching out now for some assistance..
When I compile with `standard configuration files other than selecting board to be  MKS_SBASE` I get an error:

```

[12/27/17 17:10:29] Processing LPC1768 (build_unflags: -Wall; build_flags: !python Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py, -fmax-errors=5, -DU8G_HAL_LINKS; src_build_flags: -Wall; lib_deps: C
MSIS-LPC1768, U8glib-HAL; extra_scripts: Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py; board_f_cpu: 100000000L; platform: nxplpc; lib_extra_dirs: frameworks; lib_ldf_mode: off; src_filter: +<src/*>
 -<src/config>)
 
Warning! Library `CMSIS-LPC1768` has not been found in PlatformIO Registry.
You can ignore this message, if `CMSIS-LPC1768` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.

Verbose mode can be enabled via `-v, --verbose` option
Collected 9 compatible libraries
Looking for dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <CMSIS-LPC1768> v0.0.0
|-- <U8glib-HAL> v0.4
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_spi.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_timers.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HardwareSerial.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:329:0,
from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:29,
from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:23:
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins_CHITU3D.h:24:6: error: #error "Oops!  Make sure you have an STM32F1/4 board selected from the 'Tools -> Boards' menu."
#error "Oops!  Make sure you have an STM32F1/4 board selected from the 'Tools -> Boards' menu."
^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:329:0,
from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:29,
from Marlin\src\HAL\HAL_LPC1768\spi_pins.h:26,
from Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp:57:
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins_CHITU3D.h:24:6: error: #error "Oops!  Make sure you have an STM32F1/4 board selected from the 'Tools -> Boards' menu."
#error "Oops!  Make sure you have an STM32F1/4 board selected from the 'Tools -> Boards' menu."
^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:329:0,
from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:29,
from Marlin\src\HAL\HAL_LPC1768\HAL_timers.cpp:31:
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins_CHITU3D.h:24:6: error: #error "Oops!  Make sure you have an STM32F1/4 board selected from the 'Tools -> Boards' menu."
#error "Oops!  Make sure you have an STM32F1/4 board selected from the 'Tools -> Boards' menu."
^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:329:0,
from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:29,
from Marlin\src\HAL\HAL_LPC1768\HardwareSerial.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins_CHITU3D.h:24:6: error: #error "Oops!  Make sure you have an STM32F1/4 board selected from the 'Tools -> Boards' menu."
#error "Oops!  Make sure you have an STM32F1/4 board selected from the 'Tools -> Boards' menu."
^~~~~
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL.o] Error 1
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_timers.o] Error 1
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_spi.o] Error 1
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HardwareSerial.o] Error 1
 [ERROR] Took 3.40 seconds
Environment LPC1768                     [ERROR]
 [ERROR] Took 3.40 seconds
 
 [SUMMARY]
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment anet10                      [SKIP]
Environment DUE                         [SKIP]
Environment LPC1768_debug_and_upload    [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment STM32F1                     [SKIP]
Environment teensy20                    [SKIP]
Environment teensy35                    [SKIP]

fiveangle, 2017-12-27:
You need to resolve this error in your config by selecting the correct PIO Environment

> #error "Oops!  Make sure you have an STM32F1/4 board selected from the 'Tools -> Boards' menu."

Spawn32, 2017-12-27:
Use the config files form the MKS SBASE examples directory an build with:  

"platformio run -e LPC1768"

fiveangle, 2017-12-27:
@Spawn32 - is there not a correct PIO environment for the MKS_BASE target ?  If not, shouldn't it be added ?

fiveangle, 2017-12-27:
@Spawn32 - doesn't this mean that LPC1768 was the environment they chose ? :

```
 [ERROR] Took 3.40 seconds
Environment LPC1768                     [ERROR]
 [ERROR] Took 3.40 seconds
```

Wouldn't `STM32F1` be the correct environment ?

-=dave

Jamin82, 2017-12-27:
Thanks for the reply.. I did build with LPC1768. 
 That is part of my confusion as to why one of the errors are about a board that is completely different..

I will get the configuration files from the example directory and start again to try.. 

Thank you for the reply

thinkyhead, 2018-02-11:
Presuming this is solved.

closed 2018-02-11 by thinkyhead

Jamin82, 2018-02-11:
I was never able to find assistance with this issue either on github or 
indirectly from forums etc.. I gave up at the end of the day and am running 
smoothieware.. I just needed the printer working... I'll come back to this 
issue in a month or so.. Hopefully there could be some sort of guide to 
help people not familiar with this platform

thinkyhead, 2018-02-14:
> Hopefully there could be some sort of guide to help people not familiar with this platform

Expect it to be written around the time 2.0 is actually released.