MarlinFirmware/Marlin #8918: SanityCheck.h error with autoleveling on prusa i2

lorenjz, 2017-12-27:
Hello,

I recently installed the BLTouch sensor on my prusa i2 and  downloaded the latest version of Marlin to enable the auto bed leveling feature.  I've tried various settings based on another post that seemed to have the same issues without any success.  Below are the settings I've been working with as well as the error that I've received.   

Any help with this would be greatly appreciated.

Thank you!

Loren

```
#define X_PROBE_OFFSET_FROM_EXTRUDER 45
#define Y_PROBE_OFFSET_FROM_EXTRUDER 10
#define Z_PROBE_OFFSET_FROM_EXTRUDER 2  
``` 

```
#define LEFT_PROBE_BED_POSITION 50
#define RIGHT_PROBE_BED_POSITION 155
#define FRONT_PROBE_BED_POSITION 180
#define BACK_PROBE_BED_POSITION 15
```

```
// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 50
#define Y_MIN_POS 15
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 200
```

```
SanityCheck.h:296: error: static assertion failed: Movement bounds ([XY]_MIN_POS, [XY]_MAX_POS) are too narrow to contain [XY]_BED_SIZE.
 static_assert(X_MAX_LENGTH >= X_BED_SIZE && Y_MAX_LENGTH >= Y_BED_SIZE,
 ^
SanityCheck.h:826: error: static assertion failed: FRONT_PROBE_BED_POSITION must be less than BACK_PROBE_BED_POSITION.
       static_assert(FRONT_PROBE_BED_POSITION < BACK_PROBE_BED_POSITION, "FRONT_PROBE_BED_POSITION must be less than BACK_PROBE_BED_POSITION.");
       ^
SanityCheck.h:827: error: static assertion failed: LEFT_PROBE_BED_POSITION can't be reached by the Z probe.
       static_assert(LEFT_PROBE_BED_POSITION >= MIN_PROBE_X, "LEFT_PROBE_BED_POSITION can't be reached by the Z probe.");
       ^
exit status 1
static assertion failed: Movement bounds ([XY]_MIN_POS, [XY]_MAX_POS) are too narrow to contain [XY]_BED_SIZE.

```

lorenjz, 2017-12-27:
I think I have it figured out.  The code compiles.  I had the cartesian plane backwards.

closed 2017-12-27 by lorenjz

AffeP, 2018-01-30:
How did you fix it?