MarlinFirmware/Marlin #8919: Is it possible to use 2 arduinos on a single printer?

manotroll, 2017-12-27:
I wonder if it is possible to use 2 arduinos together to add multi-color support on an A8?
using serial / spi or pi to manage this

tcm0116, 2017-12-28:
Not currently. Does your control board have two Z Motor plugs? If so, you can just wire up the Z motors in parallel using something like this:

LERDGE 3D Printer Parts Motor Parallel Module for Double Z Axis Dual Z Motors for Lerdge 3D Printer Board
http://s.aliexpress.com/AreYNf2Y

tcm0116, 2017-12-28:
Just took a closer look at the control board, and it looks like it only has 4 stepper drivers, so I guess the two Z Motor plugs are already wired to a single driver. 

Your best solution is probably to swap out the control board for one that has 5 drivers on it. 

Sorry for the confusion.

manotroll, 2017-12-28:
yes maybe this is the best option

closed 2017-12-29 by manotroll

thinkyhead, 2018-02-11:
@manotroll See also https://github.com/MarlinFirmware/Marlin/issues/4776#issuecomment-246262879