MarlinFirmware/Marlin #8922: TMC2130's stutter if parts fan is using PWM

SpeedGP, 2017-12-27:
### Bug Report
- Description: When parts fan (D11 with Geetech fan extender) is off, or at 100%, X and Y axis's work fine, but if parts fan is at 1% - 99% X and Y stutter very badly and move at approx 1mm/s. Z axis (without TMC2130) is unaffected either way.
- Expected behaviour: Printer acts the same regardless of  parts fan speed
- Actual behaviour: Axises move slowly/stutter

[configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/1589973/configuration.h.zip)

teemuatlut, 2017-12-27:
Does it happen if you use another driver type on X/Y?
Does it happen if you use another pin than D11?
Does it happen with the parts fan and extender disconnected?
Does it happen if you disable `SENSORLESS_HOMING`?
Does it happen if you disable `STEALTHCHOP`?
Are the drivers made by Watterott or Fysetc?

AnHardt, 2017-12-27:
Hardware PWM on pin D12 usses an timer what is needed elsewhere. Try fan soft pwm.

SpeedGP, 2017-12-28:
fan_soft_pwm worked!

Thank you so much!

Sorry for the burden teemuatlut. 3D printing really wouldn't be what it is
today without you guys!

-James--

psavva, 2017-12-28:
Perhaps a sanity check is needed to avoid this conflict?

thinkyhead, 2018-02-11:
I'm not sure what the sanity would be, exactly. Checking every fan and heater pin to make sure it's not using a timer that's needed elsewhere… tricky to do for all targets. For AVR maybe something like this would work…?

```cpp
#define _IS_TIMER(P,T) (_CAT(DIO_, P) == T)
#if _IS_TIMER(FAN_PIN, OCR1A) && DISABLED(FAN_SOFT_PWM)
  #error "FAN_PIN conflicts with Timer 1. Choose a different pin or enable FAN_SOFT_PWM."
#endif
```

boelle, 2019-02-19:
@SpeedGP so your problem is solved?

maybe close this one?

SpeedGP, 2019-02-20:
It's been a long time. I think part of the issue was heat causing the
drivers to cut back, but also that pwm thing helped a lot. Changing the
frequency or something like that, I don't remember.

Thanks!

boelle, 2019-03-06:
@thinkyhead i think we can close this one

Phr3d13, 2019-06-10:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

closed 2019-07-06 by shitcreek