MarlinFirmware/Marlin #8923: [2.0.x_bugfix] Wont compile on Due/Radds

Haffwa, 2017-12-28:
Hi, I am currently building a 3D printer and I want to use a Arduino Due with Radds and TMC2130 drivers with spi.
### Compile Error
I use Arduino 1.9.0 beta and installed the latest libraries/boards.
I want to try Marlin 2.0 bugfix on a Due with Radds, but I get this error.
Paste_the_error_text_here
Linking everything together...
...
"C:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/libsam_sam3x8e_gcc_rel.a" "C:\Users\admin\AppData\Local\Temp\arduino_build_278910/core\core.a" -Wl,--end-group -lm -gcc
sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_test_unit_ready':

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:24: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_read_capacity':

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:30: undefined reference to `Sd2Card::cardSize()'

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:32: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_removal':

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:49: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_usb_read_10':

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:76: undefined reference to `Sd2Card::readStart(unsigned long)'

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:83: undefined reference to `Sd2Card::readData(unsigned char*)'

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:87: undefined reference to `Sd2Card::readStop()'

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:93: undefined reference to `Sd2Card::readStop()'

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:97: undefined reference to `card'

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:97: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_usb_write_10':

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:110: undefined reference to `Sd2Card::writeStart(unsigned long, unsigned long)'

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:118: undefined reference to `Sd2Card::writeStop()'

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:123: undefined reference to `Sd2Card::writeData(unsigned char const*)'

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:127: undefined reference to `Sd2Card::writeStop()'

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:131: undefined reference to `card'

C:\Users\admin\AppData\Local\Temp\arduino_build_278910\sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:131: undefined reference to `card'

collect2.exe: error: ld returned 1 exit status

Using library Wire at version 1.0 in folder: C:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\libraries\Wire 
Using library U8glib at version 1.19.1 in folder: C:\Users\admin\Documents\Arduino\libraries\U8glib 
exit status 1
Error compiling for board Arduino Due (Programming Port).

------------------------------------------------------------
Any help is highly appreciated!

psavva, 2017-12-28:
I suggest you post your configuration files too.

cjsoong, 2017-12-28:
Clear temp file and then recompile it

Haffwa, 2017-12-30:
Sorry for the late reply, here are my config files.
Which temp file? I cleared the temp folder, but that didnt solve it.

Best regards
[Marlin_2.0.x_Duetmc2130_conf.zip](https://github.com/MarlinFirmware/Marlin/files/1594129/Marlin_2.0.x_Duetmc2130_conf.zip)

Haffwa, 2017-12-30:
Enabling #SDSUPPORT fixed my problem and I am now able to compile. Read about this in the #8949 
Thanks to all.

closed 2017-12-30 by Haffwa