MarlinFirmware/Marlin #8925: Bed Temperatur Control

lphie, 2017-12-28:
### Bug Report
- Description: Heating failed on Bed PID tuning
- Expected behaviour: Bed PID tuning
- Actual behaviour: immediate error when using WATCH_BED_TEMP_PERIOD with values >= 127
- Steps to reproduce:
  - Enter G-code; M303 E-1 C8 S65
  with WATCH_BED_TEMP_PERIOD set to 180 in Configuration_adv.h

- Solution
 change in file temperature.cpp:

      const int8_t watch_temp_period = ...
 to:
      const **uint8_t** watch_temp_period = ...

 This will solve the problem for 127 <= WATCH_BED_TEMP_PERIOD <= 255 seconds.

 Happy new year

 lphie

thinkyhead, 2018-02-11:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

lphie, 2018-02-16:
I have tested Bed PID tuning with  G-code; M303 E-1 C8 S65 and no error occured.

Thank you

closed 2018-02-16 by lphie