MarlinFirmware/Marlin #8926: STOP called because of BLTouch error - restart with M999

lorenjz, 2017-12-28:
Hello,

I've been working on getting a BLTouch sensor to my prusa i2.  I'm running ramps 1.4 and octoprint.  When my octopi starts this is what the terminal outputs:

```
Recv: start
Recv: echo:Marlin 1.1.6
Recv: 
Send: N0 M110 N0*125
Recv: echo: Last Updated: 2017-10-10 12:00 | Author: (none, default config)
Recv: eN0 M110 iled: Dec 28 2017
Recv: echo: Free Memory: 5215  PlannerBufferBytes: 1232
Recv: echo:V41 stored settings retrieved (498 bytes; crc 23991)
Recv: echo:  G21    ; Units in mm
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D3.00
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z4000.00 E97.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X250.00 Y250.00 Z2.00 E22.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1000 Y1000 Z5 E1000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P500.00 R500.00 T500.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P22.20 I1.08 D114.00
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z0.00
Recv: Error:STOP called because of BLTouch error - restart with M999
Changing monitoring state from 'Connecting' to 'Error: STOP called because of BLTouch error - restart with M999
'
Recv: Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
Recv: ok
Changing monitoring state from 'Error: STOP called because of BLTouch error - restart with M999
' to 'Offline'
```
I'm able to restart the serial connection and run Gcode commands.  I found some troubleshooting suggestions [here](https://github.com/MarlinFirmware/Marlin/issues/6756) and this are the results of my testing:

> 1. Blue LED will be ON if a servo signal is present. Blue LED is faint. 
 Result:  Blue LED is on
> 2. M119 ; should return OPEN. If not check that the BLTouch is enabled in the configuration. If enabled then check wiring and check that the correct pin is being used.  
Result:  Open
> 3. M280 P0 S10 ; should deploy the probe and the orange LED will be OFF. 
Result:  probe acted as it should
> 4. M119 ; should return OPEN
Result:  Open
> 5. M280 P0 S90 ; should stow the probe and the orange LED will be ON
Result: probe acted as it should
6. M280 P0 S60 ; puts it into the M119 test mode.
7. The probe should remain stowed and the blue LED should be OFF.
Result: probe acted as it should
8. M119 ; should return TRIGGERED
Result: probe acted as it should
9. M280 P0 S160 ; returns it to normal operation
10. M119 ; should return OPEN.
Result: Triggered

This is the output of terminal when I run G29:

> Send: G28
> [...]
> Recv: echo:busy: processing
> Recv: X:100.00 Y:100.00 Z:0.00 E:0.00 Count X:8000 Y:8000 Z:0
> Recv: ok
> [...]
> Send: G29
> [...]
> Recv: echo:busy: processing
> Recv: Error:STOP called because of BLTouch error - restart with M999
> Changing monitoring state from 'Operational' to 'Error: STOP called because of BLTouch error - restart with M999
> '
> Recv: Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
> Recv: echo:busy: processing
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: Error:Probing failed
> Recv: X:5.00 Y:0.00 Z:25.00 E:0.00 Count X:402 Y:2 Z:100000
> Recv: ok

I've attached is my configuration.h file.  If anyone has any suggestions about what my next step to get the sensor and printer running correctly I would certainly appreciate it.

Thanks so much!

[20171228_Marlin1.1.6config_h.txt](https://github.com/MarlinFirmware/Marlin/files/1591501/20171228_Marlin1.1.6config_h.txt)

knoopx, 2018-01-03:
I have the same problem with a gt2560 rev a and anet 1.0 board. I'm using a 3d touch and properly works until the print is finished. Once I start another print I get a STOP error with the same message. I have to manually pull the probe down and reset the printer in order to get it working again.

jmusser1, 2018-01-12:
I have the same problem with a gt2560 rev A+. I have to cycle the power for it to reset to print again.

trickius, 2018-01-28:
Same problem here. Rumba board with 3dtouch and marlin 1.1.6. First attempt always ends up with bltouch error. After cycling power runs like a champ.

lorenjz, 2018-01-29:
So I’m pretty sure that I had a clone that did’t work properly.  I was able to purchase an authentic part from amazon and once I had it installed, it works as advertised.

closed 2018-02-14 by thinkyhead

lpavaloi, 2018-03-29:
Hi guys... I've tried everything I could get my hands on and I cannot manage to get this sensor to work. To be more precise, I cannot give it any M280 commands.
When I start the printer it does the 2 moves and that is all. It doesn't react when it touches anything and if I let it it would crash my heat bed.
Somebody please have a look on my configuration file, perhaps I did something wrong ... anything is possible.
The sensor itself is properly wired. the two wires to min z and the other three to servos - I double triple checked them to be ok and I am pretty sure they are so my only guess is now connectivity issues.

Link here: https://drive.google.com/file/d/19hiydsplpDVQzGrLapPFqVJmDfg9u9i4/view?usp=sharing

thinkyhead, 2018-03-29:
It should "just work" if everything is properly connected. Unfortunately we cannot offer support for third-party forks of Marlin, and we don't currently have a Tevo Tarantula configuration that can be used with the current `bugfix-x.x.x` branches.

sorgens, 2018-04-06:
Hi, I had similar issue with 3dtouch. I found quick solution that might be help for the others. Try to manually pull prove ( very gentle ) and connect to a printer. Seems that sometimes probe is stucked ant that simple action helps. 

Best regsrds

FullMetalRico, 2018-11-12:
I know it's maybe late but i think there are still people having this issue so here is the **solution** in my case:

First my situation/problem:

1. The Printer works and u can move in X/Y Direction without any problems
2. When u press Auto Home (G28) the Printer moves to your home point, in my case the middle of the print bed.
3. The BlTouch lights red and the blue led is on
4. The BlTouch lights go off and it deploys -> stows the pin once.
5. Then u get the error + u cant move the Z Axis down, but in my case u can move it up

**Solution:**
In my case the Problem was due to no constant or no connection at all from the White/Black cable coming out from the BlTouch to the Printers Z-Endstop Pins. After making sure the connection there is perfect, i didnt get any errors and everything works fine.

Hope i could help anybody with this

Sidans, 2019-01-27:
as FullMetalRico commented, my problem was the Z-min connection as well

FullMetalRico, 2019-01-27:
Glad i could help someone :)

3DGoat, 2019-05-19:
That was the problem for me too, the endstop connection was wrong. Thanks

tinawiman, 2019-10-30:
Thanks, this was really helpful in fixing my problem as well!

kecorbin, 2020-01-17:
Nice find! +1 on @FullMetalRico's fix