MarlinFirmware/Marlin #8927: TMC2130 temporary stalling issue

ss89, 2017-12-28:
### Bug Report
- Description: TMC2130 Stalling Issue on specific speeds
- Expected behaviour: TMC2130 always works properly (given mechanics are good)
- Actual behaviour: TMC2130 temporarily stalls when using a driving speed of between and including 5889mm/m to 6000mm/m.
- Steps to reproduce:
  - Enable TMC2130 for Axis x or y
  - set current to somewhat reasonable, 1000 in my case
  - leave r_sense, hold_multiplier, interpolate, microstepping and hybrid threshold as-is
  - enable stealthchop
  - leave TMC_ADV empty
  - use the host tool of your choice and send a "G0 X100 F5889" and then a "G0 X0 F6000"
  - pray that no components got damaged from the vibration the earlier step made

I also tried this with stealthchop enabled and with the threshold set to 100mm/s (which was my original configuration and was what i expected first to be the issue) - same issue there.
BUT: Disabling stealthchop completely fixes the issue (but due to a bug disables my sensorless homing).
Enabling stealthchop and setting the threshold to 98 fixes it as well since 98*60 = 5880 < 5889.
What bothers me is that it sounds like crazy driving at 6000mm/m, but 6001mm/m is totally fine (sorry for mixing mm/s and mm/m in this issue, but its easiest to explain it like that for me).
This happens on two machines with different drivers on the Y-Axis for me.

ss89, 2018-01-03:
I have to add something that i just noticed: It always stalls around the threshold, could it be that the library for some reason switches between modes (stealthchop and spreadcycle) really fast and the driver acts weird because of it?
Originally i had this issue with a 12V Power supply, just now i changed one of my drivers VM to 24V which makes the driver/stepper motor whine less (but still not as silent as a TMC2100@12V) - The problem persists though.

@teemuatlut do you have an idea about this?

teemuatlut, 2018-01-03:
The switching is done by the driver itself. What the library does is communicate the register values. Before that Marlin figures out the conversion from mm/s to step interval (normalized to 256).
You can read about the switching process in the datasheet.

ss89, 2018-01-03:
Ahh thanks, then there must be something else at fault (in Marlin?).

thinkyhead, 2018-02-11:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

ss89, 2018-10-28:
i have used the newest 1.1.9 now and i can see issues with acceleration.
I am not sure if they relate to tmc2130 or are general issues.
reproducable as follows:
* enable stealthchop
* have hybrid mode enabled and x axis threshold at ~100
* set acceleration for x axis very low, like 200
* put your hand on the x motor
* let your printing host software move the motor to a position +100 from your origin

While accelerating and decelerating (i believe while switching from stealthchop to spreadcycle and back) you should feel a short anormal rattling.
I have now switched from a ramps board to a einsy rumba (from aliexpress), so no wiring issues at any point should occur.
For not having issues i for now disabled stealthchop since i am not in the room while printing usually anyway.

ss89, 2018-11-03:
@thinkyhead

thinkyhead, 2018-11-05:
@ss89

ss89, 2018-11-10:
@thinkyhead have you seen what i posted regarding the acceleration rattling? can you please try to verfiy with your own machine? possibly with tmc2130 AND other drivers?

boelle, 2019-02-19:
@ss89  Please test with the latest `bugfix-2.0.x` 
1.1.9 is outdated, so use bugfix first to see if issue is gone

boelle, 2019-03-06:
@thinkyhead i think we can close this one

Phr3d13, 2019-06-10:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

ss89, 2019-06-23:
i sold the printer that had that issue, but i might test that with another one soon.

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-07-24:
@teemuatlut can you verify this?

boelle, 2019-08-19:
this one is most likely resolved by the square wave stepping mode added awhile ago
or the changes to min stepper pulse

will close but we can always reopen if it turns out that issue is still there

closed 2019-08-19 by boelle

InsanityAutomation, 2019-08-19:
Given whats been changed around this area with the min stepper pulse timings as well as square wave stepping to prevent timing issues to these drivers are high rates, it is very likely this bug is long since gone. Id recommend closing this for now since it was reported with such a drastically different environment and reopen this or a new issue if it still occurs.

Patag, 2019-08-19:
Issue killer is back again !!! ;-) @boelle