MarlinFirmware/Marlin #8931: working marlin as klipper

manotroll, 2017-12-28:
it is possible to configure marlin to function as klipper
the speed is incredible

https://www.youtube.com/watch?v=X2bYn47-snU
https://www.youtube.com/watch?v=clixTeZ3fJA

fiveangle, 2017-12-28:
There appears a similar project with PR already submitted.  See #7047

fiveangle, 2017-12-28:
It does appear it supports AVR + Due targets, which Marlin 2.0 will also support once released:

https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md
>Klipper currently supports Atmel ATmega based micro-controllers, Arduino Due (Atmel SAM3x8e ARM micro-controller), and Beaglebone PRU based printers.

@ejtagle and @colinrgodsey - have you seen this ?  https://github.com/KevinOConnor/klipper

manotroll, 2017-12-28:
will the speed of marlin 2.0 be equal in 32 bits?
even after marlin 2.0 this feature could work the same as klipper?
For that it has several printers it would be cheaper since it would suffice to have a PC descktop running linux to execute the commands

fiveangle, 2017-12-28:
To be clear, there are no current plans as of yet to implement the _klipper_ client code API within Marlin (or if it is even possible).

I'm not sure of your other questions, but I suspect the potential speed improvement in the _klipper_ client/server over Marlin 2.0 running natively on 32-bit targets such as Due may not gain much.  Marlin can already run 100mm/s on most cartesian configs with 8-bit MCUs.  32-bit MCUs such as Due should extend that capability well into the speeds claimed by the `klipper `project (without the added complexity of a server-side pre-processor component).

-=dave

Sebastianv650, 2017-12-28:
@fiveangle I'm aware of klipper, a very intresting approach. Given the documentation, his pressure advance is working exactly like Marlins one, except the look-ahead extruder advance steps limiter. I see no easy way to implement this feature, but I know we need something similar. That's on my list for a LIN_ADVANCE rework in the next days/weeks.

manotroll, 2017-12-29:
I will do some tests with a8 as soon as the customs finish reviewing the taxes
the marlin would have to work in a similar way that would be in having the commands on the host and use the 8 bits just to do the commands and nothing else

klipper may have made all 32bit cards obsolete since it allows multiple 8-bit connections to meet the needs

closed 2018-02-11 by thinkyhead