MarlinFirmware/Marlin #8932: Bugfix 2.x.x ### Compile Error on RAMPS_FD_V1 with Due

dembach, 2017-12-28:
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
n file included from /Users/dembach/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/system/CMSIS/Device/ATMEL/sam3xa/include/sam3xa.h:44:0,
                 from /Users/dembach/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/system/CMSIS/Device/ATMEL/sam3.h:59,
                 from /Users/dembach/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/system/CMSIS/Device/ATMEL/sam.h:198,
                 from /Users/dembach/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/system/libsam/chip.h:25,
                 from /Users/dembach/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/cores/arduino/Arduino.h:42,
                 from sketch/src/gcode/config/../../inc/../HAL/HAL_DUE/HAL_Due.h:34,
                 from sketch/src/gcode/config/../../inc/../HAL/HAL.h:40,
                 from sketch/src/gcode/config/../../inc/MarlinConfig.h:28,
                 from sketch/src/gcode/config/M43.cpp:23:
/Users/dembach/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/system/CMSIS/Device/ATMEL/sam3xa/include/sam3x8e.h:460:30: error: operator '*' has no right operand
 #define TC1        ((Tc     *)0x40084000U) /**< \brief (TC1       ) Base Address */
                              ^
sketch/src/gcode/config/../../pins/pinsDebug_list.h:22:22: note: in expansion of macro 'TC1'
 
                      ^
/Users/dembach/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/system/CMSIS/Device/ATMEL/sam3xa/include/sam3x8e.h:461:30: error: operator '*' has no right operand
 #define TC2        ((Tc     *)0x40088000U) /**< \brief (TC2       ) Base Address */
                              ^
sketch/src/gcode/config/../../pins/pinsDebug_list.h:25:22: note: in expansion of macro 'TC2'
 // Following this pattern is a must.
                      ^
/Users/dembach/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/system/CMSIS/Device/ATMEL/sam3xa/include/sam3x8e.h:460:30: error: operator '*' has no right operand
 #define TC1        ((Tc     *)0x40084000U) /**< \brief (TC1       ) Base Address */
                              ^
sketch/src/gcode/config/../../pins/pinsDebug_list.h:22:22: note: in expansion of macro 'TC1'
 
                      ^
/Users/dembach/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/system/CMSIS/Device/ATMEL/sam3xa/include/sam3x8e.h:461:30: error: operator '*' has no right operand
 #define TC2        ((Tc     *)0x40088000U) /**< \brief (TC2       ) Base Address */
                              ^
sketch/src/gcode/config/../../pins/pinsDebug_list.h:25:22: note: in expansion of macro 'TC2'
 // Following this pattern is a must.
                      ^
In file included from sketch/src/gcode/config/../../pins/pinsDebug.h:103:0,
                 from sketch/src/gcode/config/M43.cpp:29:
sketch/src/gcode/config/../../pins/../HAL/HAL_pinsDebug.h:29:41: fatal error: HAL_DUE/HAL_pinsDebug_Due.h: No such file or directory
   #include "HAL_DUE/HAL_pinsDebug_Due.h"
                                         ^
compilation terminated.
exit status 1
```

[Configuration_1st_try.zip](https://github.com/MarlinFirmware/Marlin/files/1591881/Configuration_1st_try.zip)

Bob-the-Kuhn, 2017-12-29:
I don't think PINS_DEBUGGING is implemented in 2.0.x

dembach, 2017-12-29:
@Bob-the-Kuhn : tooo bad..... it is M43, a basic command to test things out before going live to prevent wrong behavior and the possiblity of defects.
MK4Duo has got support for that... as i mentioned, way ahead Marlin in some cases. you really think over , fusioning with MK4Duo, so we have got best of all.

fiveangle, 2017-12-29:
> I don't think PINS_DEBUGGING is implemented in 2.0.x

Bob, I think you mean not implemented for "Due" target, right ?  It absolutely should be supported for all boards that 1.1.x supports.

@dembach that's great info.  Please find what changes are required and submit a PR so others can benefit.

-=dave

dembach, 2017-12-29:
@fiveangle , it is implemented in MK4Duo by @MagoKimbra in file
[https://github.com/MKFirmware/MK4duo/blob/master/MK4duo/src/core/commands/gcode/debug/m43.h](url)

Bob-the-Kuhn, 2017-12-30:
Correct - PINS_DEBUGGING is not implemented in 2.0.x for Due..

thinkyhead, 2017-12-31:
> …implemented in MK4Duo by @MagoKimbra in file…

And also in some other places, like…
https://github.com/MKFirmware/MK4duo/blob/master/MK4duo/src/HAL/HAL_DUE/HAL_Due.cpp#L670

Reading through the code of MK4Duo is instructive. I highly recommend it.

dembach, 2017-12-31:
For the community i highly recommend to do a fusion with MK4Duo.

the only reason for this fork was unsolved feature requests by this version of Marlin was made to accommodate some requests made by the community RepRap Italy ...

For now, it is the only complete working 32-bit implementation for such a widespread variant of boards and Hardware.

Both Versions are good, but the best for the community would be , the best of both. 

best regards

Markus

MagoKimbra, 2017-12-31:
No need, trust me! In a while, when Marlin 2.0 will come out you will see that it will be at the top. Only time to work. When it is ready, it will be many times better than the current FW. For now if you have a 32bit card, use MK, but as I have also explained to @thinkyhead I interrupt the development of MK, I will only fix if they will serve.

Bob-the-Kuhn, 2017-12-31:
PR #8990 adds Due support for PINS_DEBUGGING.  Hopefully that PR will be merged later today.

dembach, 2017-12-31:
@MagoKimbra so we need to show up, what is missing in Marlin and get it working as good as MK...
and push your special features towards Marlin..... ;-)

dembach, 2017-12-31:
@Bob-the-Kuhn : thanks..... getting closer ;-)

closed 2018-02-04 by dembach