MarlinFirmware/Marlin #8935: [bugfix-1.1.x] Marlin (Linear Advance) produces blobs on 90° corners

ikarisan, 2017-12-28:
### Bug Report
- Description: When using Linear Advance, since version 1.1.7 Marlin produces blobs at almost every 90° corner. 
- Expected behaviour: No blobs are generated, as on 1.1.6

- Actual behaviour: 
When printing objects like the 20x20 XYZ calibration cube or the Linear Advance test pattern, Marlin produces large blobs at avery 90° corner. This is independent of the chosen K factor (see the attached photo). I tried to reduce the extrusion factor from 1.0 to 0.80 but Marlin is printing blobs. Then I increased the retraction distance from 0.6mm (worked in 1.1.6) to 1.2mm but Marlin is still printing blobs... :( But not every movement is affected! Only when switching from short line to long lines the blob occurs. When switching from a long line to a short line, the corner is printed well (green marks in photo)!

[image]

When printed with an extrusion multiplicator of 0.80 and a retraction of 1.0mm the blobs a larger than when printed with E0.9 and a retraction of 0.5!

- Steps to reproduce:
  - Flash 1.1.7-bugfix edition on Anet A8 board
  - Activate LA (K between 40 and 100)
  - Print the Linear Advance alternative pattern (see photo)
    - Use values for retraction and E factor that worked on 1.1.6

The pattern on the photo has been printed with these setting:

; Print Size X = 100 mm
; Print Size Y = 145 mm
; Print Rotation = 0 degree
; Print Pattern = Alternate
; Print Frame = false
; Prime Nozzle = true
; Prime Extrusion Multiplier = 1
; Dwell Time = 2 s
; Filament Diameter = 1.75 mm
; Nozzle Diameter = 0.4 mm
; Nozzle Temperature = 210 °C
; Nozzle / Line Ratio = 1.05
; Bed Temperature = 60 °C
; Slow Printing Speed = 1200 mm/min
; Fast Printing Speed = 5400 mm/min
; Movement Speed = 7200 mm/min
; Use BL = true
; Retraction Distance = 1 mm
; Bed Size X = 220 mm
; Bed Size Y = 220 mm
; Layer Height = 0.2 mm
; Extrusion Multiplier = 0.8
; Starting Value K-Factor = 40
; Ending value K-Factor = 100
; K-Factor Stepping = 5
; Test Line Spacing = 5 mm
; Test Line Length Slow = 20 mm
; Test Line Length Fast = 50 mm

fiveangle, 2017-12-28:
Don't think anyone will be able to help if you don't provide the gcode that produced those results.

-=dave

MasterPIC, 2017-12-29:
Exactly the same issue on my Anet A8 with firmware bugfix-2.0.x.

I used the following gcode:

; ### Marlin K-Factor Calibration Pattern ###
; -------------------------------------------
;
; Created: Thu Dec 28 2017 10:20:44 GMT+0100 (W. Europe Standard Time)
; Settings:
; Print Size X = 90 mm
; Print Size Y = 125 mm
; Print Rotation = 0 degree
; Print Pattern = Alternate
; Print Frame = false
; Prime Nozzle = true
; Prime Extrusion Multiplier = 2.5
; Dwell Time = 2 s
; Filament Diameter = 1.75 mm
; Nozzle Diameter = 0.3 mm
; Nozzle Temperature = 250 °C
; Nozzle / Line Ratio = 1.2
; Bed Temperature = 85 °C
; Slow Printing Speed = 1200 mm/min
; Fast Printing Speed = 4200 mm/min
; Movement Speed = 7200 mm/min
; Use BL = false
; Retraction Distance = 1 mm
; Bed Size X = 200 mm
; Bed Size Y = 200 mm
; Layer Height = 0.1 mm
; Extrusion Multiplier = 1
; Starting Value K-Factor = 0
; Ending value K-Factor = 200
; K-Factor Stepping = 10
; Test Line Spacing = 10 mm
; Test Line Length Slow = 20 mm
; Test Line Length Fast = 40 mm
;
; prepare printing
;
;G28 ; home all axes
M190 S85 ; set and wait for bed temp
M104 S250 ; set nozzle temp and continue
M109 S250 ; block waiting for nozzle temp
G21 ; set units to millimeters
M204 P500 ; lower acceleration to 500mm/s2
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
;
; reset extruder and go to layer height
;
G92 E0 ; reset extruder distance
G1 Z0.1 F1200
;
; prime nozzle
;
G1 X55 Y37.5 F7200
G1 X55 Y162.5 E4.6772 F1200 ; prime nozzle
G1 X55.54 Y162.5 E0.0202 F1200 ; prime nozzle
G1 X55.54 Y37.5 E4.6772 F1200 ; prime nozzle
G1 E-1
;
; start the test pattern
;
G4 P2000 ; Pause (dwell) for 2 seconds
G1 X65 Y37.5 F7200 ; move to pattern start
M900 K0 ; set K-factor
G1 E1
G1 X85 Y37.5 E0.2993 F1200
G1 X125 Y37.5 E0.5987 F4200
G1 X145 Y37.5 E0.2993 F1200
G1 X145 Y42.5 E0.0748 F4200
M900 K10 ; set K-factor
G1 E1
G1 X125 Y42.5 E0.2993 F1200
G1 X85 Y42.5 E0.5987 F4200
G1 X65 Y42.5 E0.2993 F1200
G1 X65 Y47.5 E0.0748 F4200
M900 K20 ; set K-factor
G1 E1
G1 X85 Y47.5 E0.2993 F1200
G1 X125 Y47.5 E0.5987 F4200
G1 X145 Y47.5 E0.2993 F1200
G1 X145 Y52.5 E0.0748 F4200
M900 K30 ; set K-factor
G1 E1
G1 X125 Y52.5 E0.2993 F1200
G1 X85 Y52.5 E0.5987 F4200
G1 X65 Y52.5 E0.2993 F1200
G1 X65 Y57.5 E0.0748 F4200
M900 K40 ; set K-factor
G1 E1
G1 X85 Y57.5 E0.2993 F1200
G1 X125 Y57.5 E0.5987 F4200
G1 X145 Y57.5 E0.2993 F1200
G1 X145 Y62.5 E0.0748 F4200
M900 K50 ; set K-factor
G1 E1
G1 X125 Y62.5 E0.2993 F1200
G1 X85 Y62.5 E0.5987 F4200
G1 X65 Y62.5 E0.2993 F1200
G1 X65 Y67.5 E0.0748 F4200
M900 K60 ; set K-factor
G1 E1
G1 X85 Y67.5 E0.2993 F1200
G1 X125 Y67.5 E0.5987 F4200
G1 X145 Y67.5 E0.2993 F1200
G1 X145 Y72.5 E0.0748 F4200
M900 K70 ; set K-factor
G1 E1
G1 X125 Y72.5 E0.2993 F1200
G1 X85 Y72.5 E0.5987 F4200
G1 X65 Y72.5 E0.2993 F1200
G1 X65 Y77.5 E0.0748 F4200
M900 K80 ; set K-factor
G1 E1
G1 X85 Y77.5 E0.2993 F1200
G1 X125 Y77.5 E0.5987 F4200
G1 X145 Y77.5 E0.2993 F1200
G1 X145 Y82.5 E0.0748 F4200
M900 K90 ; set K-factor
G1 E1
G1 X125 Y82.5 E0.2993 F1200
G1 X85 Y82.5 E0.5987 F4200
G1 X65 Y82.5 E0.2993 F1200
G1 X65 Y87.5 E0.0748 F4200
M900 K100 ; set K-factor
G1 E1
G1 X85 Y87.5 E0.2993 F1200
G1 X125 Y87.5 E0.5987 F4200
G1 X145 Y87.5 E0.2993 F1200
G1 X145 Y92.5 E0.0748 F4200
M900 K110 ; set K-factor
G1 E1
G1 X125 Y92.5 E0.2993 F1200
G1 X85 Y92.5 E0.5987 F4200
G1 X65 Y92.5 E0.2993 F1200
G1 X65 Y97.5 E0.0748 F4200
M900 K120 ; set K-factor
G1 E1
G1 X85 Y97.5 E0.2993 F1200
G1 X125 Y97.5 E0.5987 F4200
G1 X145 Y97.5 E0.2993 F1200
G1 X145 Y102.5 E0.0748 F4200
M900 K130 ; set K-factor
G1 E1
G1 X125 Y102.5 E0.2993 F1200
G1 X85 Y102.5 E0.5987 F4200
G1 X65 Y102.5 E0.2993 F1200
G1 X65 Y107.5 E0.0748 F4200
M900 K140 ; set K-factor
G1 E1
G1 X85 Y107.5 E0.2993 F1200
G1 X125 Y107.5 E0.5987 F4200
G1 X145 Y107.5 E0.2993 F1200
G1 X145 Y112.5 E0.0748 F4200
M900 K150 ; set K-factor
G1 E1
G1 X125 Y112.5 E0.2993 F1200
G1 X85 Y112.5 E0.5987 F4200
G1 X65 Y112.5 E0.2993 F1200
G1 X65 Y117.5 E0.0748 F4200
M900 K160 ; set K-factor
G1 E1
G1 X85 Y117.5 E0.2993 F1200
G1 X125 Y117.5 E0.5987 F4200
G1 X145 Y117.5 E0.2993 F1200
G1 X145 Y122.5 E0.0748 F4200
M900 K170 ; set K-factor
G1 E1
G1 X125 Y122.5 E0.2993 F1200
G1 X85 Y122.5 E0.5987 F4200
G1 X65 Y122.5 E0.2993 F1200
G1 X65 Y127.5 E0.0748 F4200
M900 K180 ; set K-factor
G1 E1
G1 X85 Y127.5 E0.2993 F1200
G1 X125 Y127.5 E0.5987 F4200
G1 X145 Y127.5 E0.2993 F1200
G1 X145 Y132.5 E0.0748 F4200
M900 K190 ; set K-factor
G1 E1
G1 X125 Y132.5 E0.2993 F1200
G1 X85 Y132.5 E0.5987 F4200
G1 X65 Y132.5 E0.2993 F1200
G1 X65 Y137.5 E0.0748 F4200
M900 K200 ; set K-factor
G1 E1
G1 X85 Y137.5 E0.2993 F1200
G1 X125 Y137.5 E0.5987 F4200
G1 X145 Y137.5 E0.2993 F1200
G1 X145 Y142.5 E0.0748 F4200
;
; mark the test area for reference
;
G1 E-1
G1 X85 Y142.5 F7200
G1 E1
G1 X85 Y162.5 E0.2993 F1200
G1 E-1
G1 X125 Y142.5 F7200
G1 E1
G1 X125 Y162.5 E0.2993 F1200
G1 E-1
;
; finish
;
M104 S0 ; turn off hotend
M140 S0 ; turn off bed
G1 Z30 Y200 F7200 ; move away from the print
M84 ; disable motors
M502 ; resets parameters from ROM (for those who do not have an EEPROM)
M501 ; resets parameters from EEPROM (preferably)
;

psavva, 2017-12-29:
@Sebastianv650  Maybe you can have a look?

ikarisan, 2017-12-30:
Here are the results of two Thingiverse objects:

https://www.thingiverse.com/thing:2497674
[image]
[image]

https://www.thingiverse.com/thing:2723449
See the corners marked with A and B.
[image]
[image]

Sebastianv650, 2018-01-01:
With 1.1.7 some things were changed inside the planner code. I thought the important ones have been reverted some days later..
Can you test again with 1.1.8 and make sure after flashing it go to Control-> Initialize EEPROM (if you use EEPROM). After that, power cycle the printer. After testing 1.1.7 the first time I got some realy crazy results without that initialization although all settings displayed the right values.

If it doesn't work with 1.1.8, I'm asking you for some patience. My initial plan was to optimize the LIN_ADVANCE code starting today to keep up with some changes that were done in 1.1.7, but at the moment I'm not trusting the planner acceleration / jerk code. If there is something strange in it, LIN_ADVANCE can't work so I want to be absolutely sure Marlin calculates smooth movements before going on. Therefore I have to create some debuging code today to read all the relevant motion speeds first.
In doubt, stay with 1.1.6 until then.

ikarisan, 2018-01-01:
@Sebastianv650 Please give me some days. Yesterday my heatbed connector melted (I "like" the Anet engineers) and the spare parts will not arrive before Thursday. :(

MasterPIC, 2018-01-01:
I think Linear Advance test pattern is a good test.
I switched to bugfix-2.0.x and I found the same blobs. If bugfix-2.0.x is "aligned" to 1.1.8 (on Anet A8) I can check within the next few hours if it can be of help.

MasterPIC, 2018-01-01:
OK, I just checked.
Initializing the EEPROM didn't solve the issue: I found the same blobs on the LIN_ADVANCE pattern test...
But by disabling retraction (0mm as retraction in the form available online to test LIN_ADVANCE) the issue has gone.

So maybe the test pattern has been changed: I may be wrong but I think in the past alternate pattern was not available and so retraction was required to properly test LIN_ADVANCE. So probably alternate pattern has been added but retraction that should not occur on the new alternate pattern is still effective...

In a few words I think it's just a bug affecting the pattern generator: in case of alternate pattern no retractions should occur on the testing pattern (except for the optional lines printed to prime the nozzle before printing the effective test)...

Sineos, 2018-01-01:
Sorry for letting this slip through. Fixed here:
https://github.com/MarlinFirmware/MarlinDocumentation/pull/124

ikarisan, 2018-01-02:
@MasterPIC But this doesn't explain the blobs on all my objects, printed with Marlin 1.1.x-bugfix (now on 1.1.8) as you can see in [my comment](https://github.com/MarlinFirmware/Marlin/issues/8935#issuecomment-354544034)

MasterPIC, 2018-01-02:
@ikarisan I can't exclude LIN_ADVANCE bugs. Anyway could you check with LIN_ADVANCE disabled (K=0) what changes?

ikarisan, 2018-01-02:
@MasterPIC Yes, but not until Friday or Saturday. :(

ikarisan, 2018-01-06:
@MasterPIC I switched over from Anet A8 board to an "MKS Gen 1.4" and four A4988 driver. Then I disabled LIN_ADVANCE in the configuration_adv.h file and reflashed Marlin bugfix-1.1.x (ba9879f8fa271a5a5f02dc1ee60768256d030254). All of my printed object are still printed with these blobs. :(

Sebastianv650, 2018-01-06:
OK so it's not related to LIN_ADVANCE. Do you have also tested 1.1.6 on your new board (all without LIN_ADVANCE first)? Maybe do a second side-by-side comparison to the config files?

ikarisan, 2018-01-06:
No, I haven't tested 1.1.6 on the new board.

MasterPIC, 2018-01-06:
The issue could be related to jerk and acceleration values.

chinoox, 2018-01-06:
i do not any issues with k-factor in 40 or 50 in direct extruder. Maybe is for firmware retraction, i don't know.
tomorrow I will send photos.

ikarisan, 2018-01-07:
@MasterPIC I tried jerk values between 5 and 30, accelerations between 200 and 800 and printing speeds between 50mm/s and 90mm/s. The results at the corners are always the same. :( 

But when printing the XYZ cube it is interesting to note that the disturbances occur only OUTSIDE the area of the letters. Above and below the X and Y letter the corners have blobs. O.o 

BTW I did not change the standard values for acceleration and jerk. Since Marlin 1.1.5 I am using these values for my Anet A8:

```
#define DEFAULT_MAX_FEEDRATE          { 400, 400, 8, 50 }
#define DEFAULT_MAX_ACCELERATION      { 2000, 2000, 100, 10000 }
#define DEFAULT_ACCELERATION          400     // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  1000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   1000    // X, Y, Z acceleration for travel (non printing) moves
#define DEFAULT_XJERK                 10.0
#define DEFAULT_YJERK                 10.0
#define DEFAULT_ZJERK                  0.3
#define DEFAULT_EJERK                  5.0
```

ikarisan, 2018-01-08:
In addition I tried different values for the retraction distance. But it doesn't matter if I use 0,6mm or 1,8mm. The result is the same. Every few layers there are these frayed corners. While between the corners the surface looks fine. 

Can I reflash Marlin 1.1.6, reset all EEPROM values to 1.1.6th default and still use my TMC2130 drivers?

ikarisan, 2018-01-10:
I downgraded to Marlin 1.1.6 and this is what I got:

[image]
[image]
[image]
[image]

ikarisan, 2018-01-10:
K-Factor pattern was printed with this file
[kfactor_test_1.1.6.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/1620137/kfactor_test_1.1.6.gcode.txt)

closed 2018-02-11 by thinkyhead