MarlinFirmware/Marlin #8937: bugfix 2.0.x: Weird Pins for Stepper CS in pins_RAMPS_FD_V1.h

dembach, 2017-12-28:
```
// Steppers
//

#define X_CS_PIN           68

#define Y_CS_PIN           60

#define Z_CS_PIN           58

#define E0_CS_PIN          67

#define E1_CS_PIN          61

#define E2_CS_PIN          59
```
cannot find these PINs in any Document regarding RAMPS_FD by bobc, can you tell me where to find them ? if these are just placeholders, we should assign them to something more realistic, maybe as counterpart to the Servos.

Bob-the-Kuhn, 2017-12-29:
Looks like some of the analog pins are being re-defined for use as digital I/O.

I have no idea what aa X_CS_PIN is used for.  My guess is that someone had a specific application and we've inherited the pins file from it.

dembach, 2017-12-29:
@Bob-the-Kuhn :  CS ->  ChipSelect 

It is beeing used for TMC2130/TMC2660 Chipset to select the Stepper-Driver we want to communicate with over SPI.

so it is a digital pin .  On RAMPS-FD there are almost all Digital Pins consumed... so you have to choose any other Pin that is already beeing assigned but not used in your configuration.

most times you do not have got a Servo, so it might be possible to choose these.

closed 2018-02-04 by dembach

tig33r, 2018-06-04:
Issue closed so I assume this pins are dream one of dev and does not available on ramps-fd board neither v1 and v2? Have to use servos or smth else?