MarlinFirmware/Marlin #8938: http://marlinfw.org/tools/lin_advance/k-factor.html

plunderfest, 2017-12-28:
### Bug Report
- Description: --- http://marlinfw.org/tools/lin_advance/k-factor.html
- Expected behaviour: --- compile K factor code  
- Actual behaviour: ---  Does nothing - will not compel code 
- Steps to reproduce: 

  http://marlinfw.org/tools/lin_advance/k-factor.html
Page will allow you to change values, but does nothing after pressing button to compile G-CODE

fiveangle, 2017-12-28:
Dupe of #8909

-=dave

chinoox, 2017-12-29:
the gcode view is very small :(

fiveangle, 2017-12-29:
close ?

ikarisan, 2017-12-30:
Maybe this Google Chrome issue should be fixed, too:
[image]

Sineos, 2017-12-30:
https://github.com/MarlinFirmware/MarlinDocumentation/pull/124
Just needs merging.

closed 2018-02-11 by thinkyhead