MarlinFirmware/Marlin #8939: bugfix 2.0.x Due not starting up if #define HAVE_TMC2130

dembach, 2017-12-28:
if #define HAVE_TMC2130 is active, i can compile, but the Arduino Due is not starting up.

i only disable this line and it is working.... same on MK4Due (marlin4kimbra).

it seems there is an issue with 32-bit and the TMC2130 library.

teemuatlut, 2017-12-28:
I was just testing a printer with Archim2 which has the same MCU as DUE and has TMC2130 drivers, so this would seem very odd. I'll try this out tomorrow with a regular Arduino DUE board.

dembach, 2017-12-29:
@teemuatlut : where can i get an Archim2 ? Ultimachine has none yet.... 
your Library is faulty , changes to get it working can be found here:

[https://github.com/Beherith/TMC2130Stepper](url)

thanks goes to @Beherith of MK4Duo Team ....

teemuatlut, 2017-12-29:
The Archim2 is a sample board provided by UltiMachine so that I can bring support for it both in the TMC2130 library and in Marlin.

ImplementOfWar, 2018-01-14:
@teemuatlut How do you have an Archim2 board?  How do I get one for testing?  Ultimachine told me it wouldn't be available until early next year.

teemuatlut, 2018-01-14:
Ultimachine came asking for official upstream support in Marlin 2.0 and I was interested but my first question was if they were going to provide a sample board to help with the development. I think it also weighed some that I was responsible for the TMC Arduino libraries and the implementation in Marlin, as well as having worked with Watterott and Trinamic before. So it's always about what you can do for them.

ImplementOfWar, 2018-01-14:
@teemuatlut I hate you, :)  I offer them nothing other then the general user feedback I could give.  I just want to print stuff..  I'm using the MKS SBASE right now on most of my printers;  I want to upgrade to a nicer 32 bit board.  I didn't want to go REARM or AZTEEG.  I saw in the commits that Ultimachine was coming out with Archim 2.0 so I decided thats the board I was going to wait for.   I am waiting while YOU get to play with it (and make sure the stepper drivers are supported).  This is where my HATE comes from.   Its more envy then HATE;  I'm just not good with words and feelings. :D

teemuatlut, 2018-01-14:
You could look into something like the ReprapWorld Ultratronics Pro.
It has the same Atmel chip as Archim2 and DUE and it has 7 removable stepper sockets. Even an integrated ethernet jack when we decide to do something in that area.

dembach, 2018-02-04:
seems to work fine with tmc lib 2.1.4

closed 2018-02-04 by dembach

thinkyhead, 2018-02-05:
Yes, the latest TMC libs are looking good!