MarlinFirmware/Marlin #8942: BED PID Tune not working

mesnik, 2017-12-28:
------------------------------------------------------------
### Bug Report

- Description: PID for Bed autotune failing
- Expected behaviour: system to run the autotune
- Actual behaviour: error message: "Heating failed. PRINTER HALTED Please reset"

- Steps to reproduce:
  - in Simplify3D, Communication's tab, run commad M303 E-1 C5 S80
[Marlin_27Dec17_v1.zip](https://github.com/MarlinFirmware/Marlin/files/1592233/Marlin_27Dec17_v1.zip)

First time writing in here! Looks very nicely organized!! I have been using Skynet3D for the last year and decided to change to Marlin.  Everything seems to be normal, prints are great and all. The only issue I am facing is to run the Bed PID autotune.  I reloaded Skynet and ran the Bed PID autotune from there fine. Reflashed Marlin and still getting error. So I must be missing a configuration somewhere... I did a search and couldn't find the answer...

Thanks for the help!!!!

revilor, 2017-12-29:
What exactly happens to the bed temperature when you run PID autotune? Is the bed heating but very slowly, or not heating at all? And are bed PID parameters in your Configuration.h (DEFAULT_bedKp etc.) the ones you got from Skynet3D autotune?

mesnik, 2017-12-29:
Thanks for the reply!! 

So nothing happens. The bed doesn’t heat up.

And yes, the info I loaded into Configuration.h was what I was using on Skynet3D autotuned.

What is weird to me is that  the bed runs normal when I print something...

revilor, 2017-12-29:
Well, the job of the PID controller is to hold the bed temperature as close to the target as possible. From ambient to target temperature it's 100% power anyway. So you could try to manually heat the bed up to 75° or so and start PID autotune from there.

PID tuning was a little bit tricky for my heated bed too, because the required settings to avoid huge overshoots when heating up also mean that it takes some time in the beginning before some effect on bed temperature is visible at all. One could try to set WATCH_BED_TEMP_PERIOD to a very high value in that case, but preheating the bed for PID tuning did the job for me.

mesnik, 2017-12-29:
Thanks for the suggestion.  Unfortunately it did not work...

As the message says, it halted and the both temps (extruder and bed) just shut off.

: (

patrovite, 2017-12-30:
Hi,

I had the same problem on my Anet A8 with the firmware 1.1.8.

mesnik, 2017-12-30:
Ah! Good! I mean, bad for you, but good that I am not the only odd ball out here (as far as this goes...hahahaha!)

xkilowatts, 2017-12-30:
Is it just your bed? Or hotend as well? I can't PID tune either, as soon as I send the M303 command via Terminal on Octoprint it kills the printer, doesn't even give it a chance to start heating.

mesnik, 2017-12-31:
Funny enough, it is the bed only!!??

galluslongus, 2017-12-31:
Try to disable
#define THERMAL_PROTECTION_BED
For PID Tuning, after Tuning Switch it on again.

mesnik, 2017-12-31:
You are a GENIUS!!! It worked perfectly!!!!  Thanks so much for posting!!!

HAPPY NEW YEAR!!

houseofbugs, 2018-01-01:
This is a bug and you should not have to turn that off to PID tune the bed. I have submitted a report as well. Just tested on 2 different machines running 1.1.8 release.

mesnik, 2018-01-02:
Thanks for checking into it!!!

thinkyhead, 2018-01-03:
@RowanMeara — I've been looking over the `M303` thermal protection code and can't locate the source of the issue. Apparently it only applies to the bed thermal protection, and apparently it manifests quickly. I will do some more testing later today, but maybe you can spot the cause…?

galluslongus, 2018-01-03:
Yes, it only happens on Bed Pid Tuning. I copied my working PID setting to 1.1.8 config to be sure it works properly. In normal printing there is no issues. But when you try to Tune your PID for the Bed again, the thermal protection comes in quickly. My workaround is to disable thermal protection for the bed during the pid tuning and enable  after it.

revilor, 2018-01-04:
PR  #9037 fixes the issue, or setting WATCH_BED_TEMP_PERIOD in Configuration_adv.h to no more than 127 seconds

closed 2018-01-05 by thinkyhead

zag-o-mat, 2018-01-11:
The example configuration_adv.h for an Anet A8 the value for WATCH_BED_TEMP_PERIOD is set to 180. If the code will not be changed, at least that value hs to be changed in the repository.

I would suggest to change the datatype in the temperature.cpp from int8_t to uint8_t. Then you can use a time up to 255 seconds.

Am I allowed to just make the change and do a pull request, or how is the "official" process?

xeddog, 2018-02-21:
I have just completed a new install of a RAMPS 1.4 board and Marlin 1.1.8.  I just wanted to add that I have this problem with the bed, AND the nozzle.  I'm not sure if this matters, but it happens when using Octoprint terminal and entering the command, or when using the control box menu.  Hopefully a fix will resolve both.

Wayne

thinkyhead, 2018-02-23:
@xeddog — Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

xeddog, 2018-02-24:
I hate to be the bearer of bad news, but the bugfix-1.1.x I downloaded today produced this:

Recv: ok[...]Send: M303 E0 S220 C8
Recv: PID Autotune start[...]
Recv: Error:Heating failed, system stopped! Heater_ID: 0
Changing monitoring state from 'Operational' to 'Error: Heating failed, system stopped! Heater_ID: 0'
Recv: Error:Printer halted. kill() called!

and:

Send: M303 E-1 S60 C8
Recv: PID Autotune start[...]
Recv: Error:Heating failed, system stopped! Heater_ID: bed
Changing monitoring state from 'Operational' to 'Error: Heating failed, system stopped! Heater_ID: bed'
Recv: Error:Printer halted. kill() called!

Environment:

Computer is Ubuntu Linux 16.04 Desktop running Firefox 58.0.2 (64-bit), and Octoprint Version 1.3.6.

Printer is a Creality CR-10 running a RAMPS 1.4 shield on a Arduino Mega.

Wayne Jolly

xeddog, 2018-02-25:
Side question if I may.  I have only read a little, but it seems like Marlin 2.0 is mostly restructuring the code and adding 32-bit processor support.  Are any new features being added for the 8-bit users amongst us?

Thanks,

Wayne

xeddog, 2018-02-25:
Please disregard my previous email where I stated that I was the bearer of bad news.  I made an error in the configuration regarding the filament runout sensor on my CR-10 with RAMPS controller, and the sensor was always triggered.   When I fixed that, and reloaded the firmware, the PID autotunes ARE working.  I must have screwed up something in the earlier tries.

Wayne

thinkyhead, 2018-02-25:
@xeddog — Marlin 2 for AVR has perfect parity with Marlin 1.1 at this time. However, 1.1.9 will be the last major Marlin 1.x.x release, so all users will be required to update to Marlin 2.0 for new features after that.