MarlinFirmware/Marlin #8949: Compile withArduino IDE Windows file name TOO LONG?

bruce356, 2017-12-29:
Thank you for submitting your feedback to the Marlin project.

### Compile Error
When I compile for Due/RADDS Boards with Dual X axis Marlin v2 - 29/12/2017
Using Arduino IDE v1.8.5 and v1.8.6
Paste_the_error_text_here
```
Archiving built core (caching) in: C:\Users\ADMINI~1\AppData\Local\Temp\arduino_cache_877073\core\core_arduino_sam_arduino_due_x_dbg_1341d1f4ed93183182d3c469edc5f9f9.a
fork/exec C:\Users\Administrator\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-gcc.exe: 
The filename or extension is too long.
Error compiling for board Arduino Due (Programming Port).
------------------------------------------------------------

AnHardt, 2017-12-29:
https://github.com/MarlinFirmware/Marlin/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aopen+The+filename+or+extension+is+too+long

bruce356, 2017-12-29:
Thanks but this does not work for me, added "build.path=C:\marlin" as recommended, but now get errors as follows:-
Archiving built core (caching) in: C:\Users\ADMINI~1\AppData\Local\Temp\arduino_cache_482232\core\core_arduino_sam_arduino_due_x_1341d1f4ed93183182d3c469edc5f9f9.a
sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_test_unit_ready':

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:24: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_read_capacity':

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:30: undefined reference to `Sd2Card::cardSize()'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:32: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_removal':

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:49: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_usb_read_10':

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:76: undefined reference to `Sd2Card::readStart(unsigned long)'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:83: undefined reference to `Sd2Card::readData(unsigned char*)'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:87: undefined reference to `Sd2Card::readStop()'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:93: undefined reference to `Sd2Card::readStop()'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:77: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:77: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_usb_write_10':

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:110: undefined reference to `Sd2Card::writeStart(unsigned long, unsigned long)'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:118: undefined reference to `Sd2Card::writeStop()'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:123: undefined reference to `Sd2Card::writeData(unsigned char const*)'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:127: undefined reference to `Sd2Card::writeStop()'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:111: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:111: undefined reference to `card'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Due (Native USB Port).

bruce356, 2017-12-29:
would anyone know where I can download Arduino v1.9 beta for windows install file
thanks

petrzjunior, 2017-12-29:
@bruce356 Yes, in Arduino 1.9 beta, the bug is fixed. You can grab it right on the Arduino Download page ([https://www.arduino.cc/en/Main/Software](https://www.arduino.cc/en/Main/Software)). There are regular, beta and hourly builds always available.

teemuatlut, 2017-12-29:
Compiling DUE is bugged atm. You can pass those errors by enabling `SDSUPPORT`, even if you don't actually use SD cards.

bruce356, 2017-12-29:
@petrzjunior  thanks, I did download this nightly build version earlier, but what confused me is that its version number is 1.8.6 not 1.9 beta so I thought is was not the right version.
@teemuatlut  thank you very much for that information. That worked.
Regards - Bruce

boelle, 2019-02-19:
@bruce356 so your problem is fixed?

maybe close this one?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek