MarlinFirmware/Marlin #8950: [FR] allow LCD to display long file names

junwoo091400, 2017-12-29:
Hello, would it be possible to show full-name of the .gcode File by 'flowing' animation?

if it can be implemented without too much memory consumption, it can be useful to people who name their files very long.

Or is this feature implemented already?

Thank you for reading my issue.

cjsoong, 2017-12-29:
Configuration_adv.h
// Enable this option to scroll long filenames in the SD card menu
  //#define SCROLL_LONG_FILENAMES

junwoo091400, 2017-12-30:
@cjsoong  OOPS, thank you so much. My bad not reading through Configuration_adv.h thoroughly.. :)

closed 2017-12-30 by junwoo091400

v1ru3, 2017-12-30:
only problem is, it doesn't compile in 1.1.8:

cardreader.cpp:726: error: incompatible types in assignment of 'char*' to 'char [66]'

               sortnames[i] = strdup(LONGEST_FILENAME);