MarlinFirmware/Marlin #8952: D9 on Ramps not working for part cooling fan

R3play210, 2017-12-29:
I'm very sorry if this is the wrong section to post but I'm completely lost.
Googled everything but can't get it to work.

Basically I have a Ramps 1.4, connected my part cooling fan to D9 but nothing happends when I tell it to turn on.
I'm reading 0.67v on D9.. that stays when the fan is turned on or off..

Using Marlin 1.1.8
Defined board as: 
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_14_EFB

Anyone knows what I'm doing wrong?

R3play210, 2017-12-29:
No, atleast I dont see anyhing

revilor, 2017-12-29:
As the D9 LED is not on or blinking when you turn on the fan it looks like the D9 MOSFET is dead (could also be a broken LED, though). So is this a new setup or did you use the RAMPS board before? With D9 working? 

To rule out that there is a short in the fan wiring I would unplug the fan from D9 and measure the resistance between the fan wires. If the resistance is close to zero you have a short somewhere in the fan wires or the fan itself.

If you measure some resistance there (could be up to the MOhms) then you could try

#define MOTHERBOARD BOARD_RAMPS_14_EEF

and connect the fan to D8. Just to see if it works there.

R3play210, 2017-12-30:
Got myself a new board.
Flashed firmware on it and only connected the LCD
Turned on the fan to max. 
And guess what, 12.2v on D9

So yeah, guessing it's a faulty mosfet on the other one.
Going to ask for a replacement though from the seller :)

Thanks for your help 👍

closed 2017-12-30 by R3play210