MarlinFirmware/Marlin #8953: Bugfix 2.0 (Dec 29) compile error with //Z_MIN_PROBE_REPEATABILITY_TEST enabled, Ramps.

KevMags, 2017-12-29:
### Compile Error
When I compile with `#define Z_MIN_PROBE_REPEATABILITY_TEST (Ramps, Bugfix 2.0 (Dec 29).  No errors with 1.1.8.
` I get an error:
```
Arduino: 1.8.2 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/Kmags/Documents/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10802 -build-path /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940 -warnings=none -build-cache /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_cache_837878 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_modified_sketch_957018/Marlin2.0Dec29GoGoGo.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/Kmags/Documents/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10802 -build-path /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940 -warnings=none -build-cache /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_cache_837878 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_modified_sketch_957018/Marlin2.0Dec29GoGoGo.ino
Using board 'mega' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Using core 'arduino' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Detecting libraries used...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/mega" "/var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/Marlin2.0Dec29GoGoGo.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/mega" "-I/Applications/Arduino.app/Contents/Java/libraries/LiquidCrystal/src" "/var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/Marlin2.0Dec29GoGoGo.ino.cpp" -o "/dev/null"
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_AVR/HAL_AVR.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_AVR/HAL_spi_AVR.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_AVR/MarlinSerial.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_AVR/persistent_store_impl.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_AVR/servo_AVR.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_AVR/watchdog_AVR.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/EepromEmulation_Due.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/HAL_Due.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/HAL_spi_Due.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/HAL_timers_Due.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/InterruptVectors_Due.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/MarlinSerialUSB_Due.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/MarlinSerial_Due.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/Servo_Due.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/persistent_store_impl.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/u8g_com_HAL_DUE_st7920_sw_spi.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/usb/ctrl_access.c
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/usb/sd_mmc_spi_mem.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/usb/sysclk.c
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/usb/udc.c
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/usb/udi_cdc.c
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/usb/udi_composite_desc.c
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/usb/udi_msc.c
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/usb/uotghs_device_due.c
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/usb/usb_task.c
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_DUE/watchdog_Due.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_LPC1768/HAL.cpp
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_LPC1768/HAL_LCD_I2C_routines.c
Using cached library dependencies for file: /var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T/arduino_build_270940/sketch/src/HAL/HAL_LPC1768/HAL_LCD_pin_routines.c
[... cut ...]
Error compiling for board Arduino/Genuino Mega or Mega 2560.

------------------------------------------------------------
### Bug Report
- Description: ---Compile error when Z_MIN_PROBE_REPEATABILITY_TEST enabled
- Expected behaviour: ---error free compile
- Actual behaviour: ---error on compile
- Steps to reproduce:
  - Do this:  uncomment //Z_MIN_PROBE_REPEATABILITY_TEST (Line #730, Configuration.h)

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/1593298/Configs.zip)

  

------------------------------------------------------------

Ordenkeim93499, 2018-01-02:
Try to download a never version of the Arduino IDE 
https://www.arduino.cc/en/Main/Software
youre running 1.8.2 on a MAC right?

KevMags, 2018-01-02:
Similar error using 1.8.5.

Arduino: 1.8.5 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

/var/folders/xr/ppq2tfc53yq_hlmsdjy4_hmm0000gp/T//ccKazjub.ltrans12.ltrans.o: In function `GcodeSuite::M48()':
ccKazjub.ltrans12.o:(.text+0x544): undefined reference to `out_of_range_error(char const*)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

closed 2018-02-11 by thinkyhead