MarlinFirmware/Marlin #8958: UBL bed leveling mk42 9point grid

tonysmartantzoulatos, 2017-12-29:
Hello and thank you in advance for your assistance. I am new here and I did not know another way besides issue for my question.
I have a steel frame i3 with a prusa mk42 bed. 
I have managed to set bilinear perfect to the 9copper points but the first layer does not seem proper.
I have tried all day to set ubl with no success. Following the marlin page instruction I can seem to have p1 set directly over the copper point with the nozzle resulting hitting the bed. 

Is there any way I can use my abl 3x3 offsets and have ubl use these in order to work ?
If not is there a way abl cal actually produce a good first layer?

Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

fiveangle, 2017-12-29:
https://github.com/MarlinFirmware/Marlin/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aopen+mk42+bed

Roxy-3D, 2017-12-30:
> I have managed to set bilinear perfect to the 9copper points but the first layer does not seem proper.
I have tried all day to set ubl with no success. Following the marlin page instruction I can seem to have p1 set directly over the copper point with the nozzle resulting hitting the bed.

I would probably set the mesh size to 10 x 10.    I would then do is G29 P0 to zero out the entire mesh. 
And at this point jump into a G26 and G29 P4 R cycle to edit the mesh points to reflect reality.

Be sure you can print a small object at the center of the bed before doing this.   If you can do that...   You can start the G26 at the center and watch to make sure the nozzle isn't getting too close to the bed.   If it is, you can stop it...  And edit those mesh points to have more clearance and then restart the G26.

It will take a little more time to do this compared to using G29 P1 auto probing...   But UBL will work just fine if you go this route.

fiveangle, 2017-12-30:
> I would probably set the mesh size to 10 x 10

No, you can't do this because there are only 9 valid points on the MK42 bed where the PINDA sensor will trigger so the head will crash on first try.

From  my https://github.com/MarlinFirmware/Marlin/issues/6601#issuecomment-338246147 above:
> Easiest is probably to enable BILINEAR, set the probe limits to match the outside edges of the 9 grid points on the bed, set number of grid points x and y to 3, mesh inset to 0, and set custom z homing position to ensure it hits the center point. Done and done.

-=dave

fiveangle, 2017-12-30:
btw- I would probably also experiment with EXTRAPOLATE_BEYOND_GRID 

-=dave

Roxy-3D, 2017-12-30:
> No, you can't do this because there are only 9 valid points on the MK42 bed where the PINDA sensor will trigger so the head will crash on first try.

I was suggesting to not use the PINDA sensor to probe the bed.   Instead...  start with a null mesh and edit to reflect the bed's distortion (with regard to the Z-Axis).

fiveangle, 2017-12-30:
Ahhhhhh !  Yeah, I don't think I'd recommend manual probing, given how much the thin PCB MK42 bed seems to like to shapeshift on temperature changes.  I tried doing MBL (while trying 1.1 RC6 with Sebastian's LIN_ADV patches on my MK2 way back when because I wanted to print faster) and got very inconsistent results resulting in first layer headaches that made it too much to deal, so I went back to stock Prusa Research FW on my MK2.

-=dave

tonysmartantzoulatos, 2017-12-30:
Is there a way to use the 3X3 abl points and have only a 9 point ubl grid?

Roxy-3D, 2017-12-30:
Yes.    You would declare the mesh to be 3 x 3.   But you would have to get the bed size and the inset exactly right to get probes aligned to those magic points.   Personally...  I don't think it would be worth the effort.   

If the bed is pretty flat without a lot of weird warping...    3 x 3 might produce good results.   

My gut hunch is you are better off abandoning the 9 probe points and just edit a high resolution mesh to reflect reality.    I don't have one of those printers so my opinion isn't worth much.   But that is my best guess.

tonysmartantzoulatos, 2017-12-30:
Thank you Roxy-3D for your reply. I understand the fact that more probe points equals better results, but for now this is the bed I have and any other point would give a lot different induction results. These 9 points are the only constant in this bed. My question in order to make it work is:
When you say bed size, do you mean the general configuration bed size? How does the inset setting works? The higher the number, the wider the points between the ? This is what I have not found any documentation on. And last, do I make all these changes on configuration.h or also make changes on configuration_adv.h?

Roxy-3D, 2017-12-30:
All of the changes are made to Configuration.h unless you want a non-symmetrical inset.   In that case, the extra detail is in Configuration_adv.h.      Bed size is the size of the bed in the X & Y direction and is assumed to start at 0.   If this is not the case, once again you can make corrections in Configuration_adv.h.

If I was trying to bring up UBL on your printer...  I would home using the center point on the bed.    And I would ignore the  other points.   If the mesh accurately defines the bed, you won't need them.    It may make sense to define two corner points and the mid point on the opposite side of the bed as your three point leveling coordinates.    Using that information you could do a 3-Point tilt (leveling) of your mesh should the need arise.

closed 2018-01-02 by tonysmartantzoulatos