MarlinFirmware/Marlin #8960: Bug Report:steppers not released after force stopping print

Witgang, 2017-12-29:
### Bug Report

- Description: steppers are not released after stopping print trough the lcd menu.
- Expected behaviour: steppers should be released as it was before i enabled the advanced pause feature.
- Actual behaviour: stopping the print locks the nozzle to it's last possition for a min or so, the only way to move it is to either battle against the steppers or hard reset the printer.
- Steps to reproduce: enable advanced pause feature & the nozzle park feature start a print and stop it.
    using ramps 1.4 EFB with RepRapDiscount Full Graphic Smart Controller 

 i fixed it by adding "enqueue_and_echo_commands_P(PSTR("G27 P2"));" at line 799 inside ultralcd.cpp
probably a total hack job but it does the trick:D

Witgang, 2017-12-30:
another thing i found is during the stop command from the lcd the board looses the position of the print head and the park nozzle comand just slams the x carriage into the endstop and then skips a few stepps or sometimes it just raises a bit and moves but not enough to reach the endstop.
   if i put a few park nozzle comands in a test.gcode file it works fine, it raises, goes to the x_min hits the endstop and stops.

thinkyhead, 2018-02-11:
Duplicate of #8875

closed 2018-02-11 by thinkyhead