MarlinFirmware/Marlin #8961: Advanced Pause + Extruder Runout

CCS86, 2017-12-29:
Hi guys, I tried to enable Advanced Pause today and got a compile error that it is not compatible with Extruder Runout prevent. I wasn't able to find info on why this is the case. 

I would love to use both. Is this possible?

Thanks!

Ordenkeim93499, 2018-01-02:
Well to clear things up why this is the case : Advanced pause means the printer moves the head to a parking position and waits there until you've changed the filament or just resume with printing. 
In case that you want to change the filament the Extruder Runout prevention works against it cause it wants to extrude some filament and pulls it in the advanced pause wants to pull it out.
Hope i helped you with this part :)

CCS86, 2018-01-06:
Hi @Ordenkeim93499, I see what you are saying.

In my experience, extruder runout prevent only functions when the printer is idle. For example, if I chose to print a file from the SD card while the extruder is already hot, but the bed still needs to warm up, runout prevent stops functioning.

Would this not be the case when using advanced pause? Is the machine effectively at idle when in that paused state? If so, would it be possible to write in a check in the runout prevent function to ensure the machine is not in a paused state?

thinkyhead, 2018-02-11:
Marlin doesn't actually have a "paused" state. But perhaps it can set a flag when `EXTRUDER_RUNOUT_PREVENT` should be turned off.

boelle, 2019-02-19:
@CCS86 Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

CCS86, 2019-02-20:
This is a feature request.

I don't think there is an inherent incompatibility between these two features, so enabling them to work together is probably best described as a feature request.

boelle, 2019-02-20:
you posted it as a question in the first post

boelle, 2019-02-20:
i just read what i see without thinking so yes it could be seen as a request

boelle, 2019-06-22:
wrong label

InsanityAutomation, 2019-06-22:
Duplicate with another FR, but can add the 2.1 milestone here as well.