MarlinFirmware/Marlin #8967: TMC2130 skipping steps in stealthchop and hybrid mode but works fine in spreadcycle

kkmalu, 2017-12-30:
- Description: I am using a CoreXY D-bot printer with Ramps 1.4 and TMC2130 on X and Y-axis (other steppers are still running with DRV8825).

- Expected behaviour: StealthChop and hybrid mode should work fine (?) given that my machine when running in SpreadCycle only, set to 650mA current runs cool and without any problems but is obviously a little loud.

- Actual behaviour: 
1) When set to StealthChop mode only, the machine would home just fine then it would start stuttering when only one motor is used to bring the extruder to center of bed, moving diagonally (remember it a coreXY setup). If I increase the current, sometimes it would move to center just fine (most times it wont) but then in the prints it will start skipping steps and layers will start move to right.

2) Thinking StealthChop is not working as well as I had thought, activated Hybrid mode this time (threshold set to 100). With current still set to 650mA, it would come to center of the bed just fine but still would skip steps during the print (when speeds at 60-80mm/s), not as often as StealthChop mode only but it still does it enough to ruin even small prints. I tried bumping the current and tested a range all the way up to 900mA (increments of 25mA starting from 650mA), the motors would start heating up slowly but the TMC2130 driver feels really hot (cant touch them for more than a second) with only two minutes into the print. Strange thing is why would the drivers heat up in the hybrid mode even when the current is set to low 650mA, when same current settings in SpreadCycle mode only runs nice and cool? (If I drop it down to 600, the motors dont move as well even in SpreadCycle mode). When I had installed the DRV8825 drivers, my current was set to 1.1A-1.28A.

3) The only solution seems to work in these modes is when I drop the acceleration parameter down to 100 but at a penalty of snail speeds.

- Steps to reproduce:
Have done this for multiple test runs with playing around only with the Amp settings with little luck. Slow speeds like 40mm/s works just fine in StealthChop, 60mm/s speeds would work ok for a little bit and would skip every now and then. 80mm/s+ speeds would stutter the motor.

Not sure what I am doing wrong or how can I fix it? Only thing I can think of is to loosen the tension on the belt a little bit mechanically but then again, in SpreadCycle and with DRV8825 setup, it works fine but is loud.

Any suggestions? Please see my config files attached.

Thanks!

[CoreXY_Config.zip](https://github.com/MarlinFirmware/Marlin/files/1594721/CoreXY_Config.zip)

s0ftcorn, 2017-12-30:
You could try to compile marlin with TMC_DEBUG and then look for reported stuff with this gcode: http://marlinfw.org/docs/gcode/M122.html

franji, 2017-12-30:
I have exactly the same problem.

I am using an similar system CoreXY D-Bot with TMC2130 on X and Y-axis, 8825 in Z and E.
I was using marlin 1.1.6 and everything worked perfectly but two days ago update to 1.1.8 and the same thing happens to me that you say.
The engines are losing steps and the layers are moving.
I have tried to modify the intensity but this does not solve the problem.
I have also tested marlin 1.1.7 the same thing happens.I have seen that there are several modifications for both CoreXY and TMC2130 and at the moment I was looking a bit there.
Meanwhile I continue with 1.1.6.

kkmalu, 2017-12-31:
@s0ftcorn : I have tried doing that but not sure how would that information help? Is there anything specific that I should be looking for? Like amps or something else?

kkmalu, 2017-12-31:
@franji : Thanks for sharing your experience as well. So are you able to run StealthChop alone using 1.1.6 version of marlin? What is your amps for X and Y set to?

thinkyhead, 2017-12-31:
@teemuatlut Can you comment?

Also, what do you think of this…? https://github.com/Beherith/TMC2130Stepper

franji, 2017-12-31:
I am using 750 mA in marlin 1.1.6 and StealthChop works correctly.

teemuatlut, 2017-12-31:
You can see [here](https://github.com/MarlinFirmware/Marlin/commit/0ac0324af2bdf65455e2ec282fc6a9b46ce511d5#diff-95647db495322b5544f2ad2ae372646b) for the relevant changes to the initialization of the drivers that were made for 1.1.7. Basically the hysteris values were changed and the blank time was changed. You can revert these by using the `TMC_ADV` macro.

650mA sounds a bit low, especially for stealthChop. I'm personally running X at 1150mA and Y at 1100mA, but I've added quite a lot of cooling.

It's possible the hybrid threshold values are thrown off by the coreXY system.

The driver thermals depend on other things as well as the current setting. In stealthChop the PWM frequency is also a factor.

teemuatlut, 2017-12-31:
@thinkyhead Please see their [pull request](https://github.com/teemuatlut/TMC2130Stepper/pull/15) and my reply.

Also if you've missed, I'd like your input on [this](https://github.com/teemuatlut/Marlin/issues/15).

dembach, 2017-12-31:
The changes of Berith are for Arduino Due SAM boards necessary, because they wont boot up but compile correctly.

Changes
-  Removed begin() from constructor
-  compiler directive to properly detect SAM boards

kkmalu, 2017-12-31:
@teemuatlut : I am not sure if I understood your recommendation completely (sorry, I am still a noob). However, per my understanding, I changed blank time back to 36 (from 24), hysterisis start to 0 (from 3) and hysterisis end to 1 (from 2). Is this what I was supposed to change? If yes, then it still did not work out. The motors were stuttering just like before. Upped the Amps to 1100 and still no good and the driver heats up very quickly!

This might be a stupid question but I had my DRV8825 drivers set to 1/32 microstepping. I have left this setting to 1/32 when I switched to TMC2130. Would this have an effect given than we do loose torque as we increase the microstepping? I know the interpolation goes to 256 but not sure if setting the microsteps to 1/32 would affect it in anyway?

gavatron3000, 2018-01-01:
I can confirm the hybrid thresholds are thrown off by the core systems. I have a corexz and spread cycle is activated far lower than the 100mm it's set at

kkmalu, 2018-01-01:
@franji : I tried flashing my board with 1.1.6 version and I have similar issues. It would home fine but with Z_Safe_homing, when it comes to the center of the bed, it would start skipping steps. I also notice differences in homing speeds when it goes to hit the endstop (slow speed) and when it comes to center of the bed (fast speed). Its the fast speeds where it skips. Do you mind sharing your files with me so that I can try and figure out how different are our settings?

franji, 2018-01-02:
If I send you the changes over 1.1.6
At 750 mA in stealthChop mode it prints me perfectly I've been with them for almost a week, I'm printing a lot to test them.
[Files_Marlin_1.1.6.zip](https://github.com/MarlinFirmware/Marlin/files/1597358/Files_Marlin_1.1.6.zip)

kkmalu, 2018-01-02:
Thanks! I'll give this a go.

marcio-ao, 2018-01-03:
@teemuatlut : 

> 650mA sounds a bit low, especially for stealthChop. I'm personally running X at 1150mA and Y at 1100mA, but I've added quite a lot of cooling.

We are struggling with this issue right now. Seems like the TMC drivers overheat at anything above about 960mA, especially when printing in an enclosure.

At present, we are using SpreadCycle on X and Y, and StealthChop on Z. We were able to get good prints using StealthChop on XY by setting the currents to 960mA, but this would cause frequent overheat warnings so we switched to SpreadCycle with CoolChop on XY.

@kkmalu : If overheating drivers are not an issue for you, I recommend at least 960 mA for SteathChop on XY. This is the minimum current at which we were able to get good prints without missed steps.

kkmalu, 2018-01-03:
Well for me, at 900mA+, the drivers heat up very fast and could be taken care of by using a really big cooling fan. However in doing so I noticed that the stepper motors also heated up (in Hybrid mode) enough to bend the PLA piece holding the motors on a longer print.

Over heating issues may be circumvented by significant cooling for decent print times but for most long prints (3h+) the motors heat up a lot and is not a good option. I am using spreadcycle myself at 650mA and everything runs cool.

I can also confirm that when I move just X or Y or in combinations with G0 commands, they run just fine and very quietly in hybrid modes (they seem to run as stealthchop) at speeds less than 100mm/sec (this is the threahold set for hybrid mode). However, when you print a part, even at about 60-70mm/sec, they somehow switch to spreadcycle (based on the noise they produce). The threaholds seem to be thrown off by coreXY is consitent to as previously reported by others.

gavatron3000, 2018-01-03:
If I run mine at 900ma they fault after a few minutes in stealth mode, at 800ma they will run for roughly 3-4 hours then cut out. Mind you that's with no fan on the drivers

xabdullahx, 2018-01-05:
@gavatron3000 did you have step skipping issues at 800ma? I haven't tried running mine for more than half an hour, but Y skips steps after every 10-20 layers so my prints are completely shifted.
Maybe i should add heatsinks and increase the current.

gavatron3000, 2018-01-05:
Yeah at 800ma after maybe 2 hours or so I'd have a slight layer shift or if the driver monitor was enabled it'd just shut the printer off. I've gone back with the a4988 drivers as they just work. But noisy. I also read a post from DC42 guy (reprapfirmware author) about the output resistance of the driver being high and causing the heating problems that everyone seems to have, and that being one of the reasons they went with the TMC2660. There were others of course

Lyr3x, 2018-01-21:
I'm facing the same issue. Using 1.1.8 in hybrid mode (XY threshold set to 100).

When printing with 80-100 mm/s the printer is loosing steps (only X) while printing with speeds <= 60mm/s everything is working fine. 

When printing 60mm/s its enough to set the X Current to 650mA. For 80mm/s 800mA isnt enough, but more than 800 will result in overheating drivers. Maybe i should try 1.1.6.

ikarisan, 2018-01-22:
Same problem here with bugfix-1.1.x (e668e053c2b22135c1dcd19e2b484533d3b019ec) and MKS Gen 1.4 board on an Anet AM8. While printing small objects is okay, all objects where the axis are moving at higher speed, I get massive layer shifts. 

My currents are set to 700mA on X and Y and my motors are getting very hot.

[image]
[image]

gavatron3000, 2018-01-22:
I don't it's firmware related all these problems. Its has to be the drivers as the silly things just get far to hot and cut out at relatively low currents. That's my take anyways

kkmalu, 2018-01-22:
I use the original Trinamic drivers and I have same issues so not sure if it has to do with hardware since they run fine with high (and as low as 650mA) currents on spreadcycle mode. Additionally, the threshold values in hybrid mode are somehow thrown off as even at 60mm/s speeds the spreadcycle turns on when threshold set to 100.

Lyr3x, 2018-01-23:
I found out that i had some mechincal issues at my X Axis. The bearing from the pulley was broken so it couldnt run smoothly. While printing with speeds above 60 mm/s is always lost steps.

After fixing i tried a large print with 75mm/s printing, 90mm/s infill and 130 mm/s travel speed. Worked flawless for me. Also while printing the spreadcycle mode did not kicked in (threshold 100 for X and Y).

NeelD87, 2018-02-08:
I am having similar issues even on small prints. 60 mm/s is always lost steps. Lower speeds work ok like 45 mm/s. Using Prusa clone.

thinkyhead, 2018-02-17:
@teemuatlut — We're finding that TMC drivers are prone to heat issues, but have been able to resolve them for the most part with bigger sinks and active cooling. What in your experience are reasonable ranges for current (and torque / acceleration / speed) when using StealthChop versus SpreadCycle modes?

thinkyhead, 2018-02-17:
Let's move this discussion to #9368 so there's one less issue for maintainers to follow.

closed 2018-02-17 by thinkyhead