MarlinFirmware/Marlin #8969: SD Card not working

R3play210, 2017-12-30:
Have Marlin 1.1.8
Using Ramps 1.4 with full graphic smart controller (SD card slot built in LCD)
When I insert a SD card the LCD says that a card has been inserted, however when you go into the menu it says that there is no SD card.
Simplify3D also says after connecting to the printer SD init fail.

Any idea what's happening here?

R3play210, 2017-12-30:
SDSUPPORT is enabled btw

jonealex, 2017-12-30:
I solved it some weeks ago, try formating the SD card with https://www.sdcard.org/downloads/formatter_4/index.html

Try it.

R3play210, 2017-12-30:
No results.
Found a 2gb card tried the same but still nothing :(

R3play210, 2017-12-30:
Upon inserting the SDCard during a print the printer stops for a brief moment and then resumes.
Thougt of mentioning this aswell

GMagician, 2017-12-30:
I have the same issue

jonealex, 2017-12-30:
Changing SD card and formating with that software works fine for mi, still SD init error on the log, but working.

GMagician, 2017-12-30:
Can't change SD, I have just one, reformatted but nothing changes

R3play210, 2017-12-30:
I just replaced the EXP2 ribbon cable with another one I had laying around and it works!
Fitted the old ribbon cable back in and it doesn't work.

So I'm guessing a faulty ribbon cable :)

GMagician, 2017-12-31:
I don't have other cables so I swapped the ones I have and issue is still present.

GMagician, 2017-12-31:
I made some test and it fails due to 'SD_CHECK_AND_RETRY' in this part:
```cpp
if (cardCommand( CMD59, 1 ) != R1_IDLE_STATE) {
    error(SD_CARD_ERROR_CMD59);
```

houseofbugs, 2018-01-04:
Just chiming in here. SD is broken in 1.1.8. Cannot get it working on any machines with it.

GMagician, 2018-01-09:
I made other test. CMD59 return 5 that is invalid command but I'm almost sure that before I saw it working!
CRC is not supported by all SD?

GMagician, 2018-01-13:
Please note that #9163 doesn't solve @R3play210 issue.
My reported issue is only present in marlin 2.0

shinigamiburakku, 2018-10-29:
Hola a todos use te traslate Google please,, la solución al problema de la sd está en ultralcd.cpp en la línea 1181 y 1188,, dice así #if !PIN_EXISTE (SD_DETECT) en las dos parte pero en la definísimo de pines se nombra como SD_DETECT_PIN solo cambiar lo que está entre paréntesis y solucionado en la pantalla lcd aparecerá el menú para cambiar la SD está es la solución que encontre marlin 1.1.9 y 2.0 bugfix testeado en ramps 1.4

closed 2018-11-02 by thinkyhead