MarlinFirmware/Marlin #8970: BLTouch not working, bug?

jonealex, 2017-12-30:
I recently bought a BLTouch clone. When i put it on my RAMPS 1.4 when i get `G28` it's not working, deploys and stows too fast without waiting to be triggered... I don't know what i can do...

Configuration files are attached. Video of the problem attached too.

[Marlin_BLTouch_Problem.zip](https://github.com/MarlinFirmware/Marlin/files/1594799/Marlin_BLTouch_Problem.zip)

Thanks !

R3play210, 2017-12-31:
I see that you have BLTOUCH_DELAY enabled.
Did you also try it when disabling this? That's how I have it and I don't have any problems.

After just a quick glance in `configuration.h` you also need to specify your X and Y MAX_POS

> // Travel limits (mm) after homing, corresponding to endstop positions.
> #define X_MIN_POS -33
> #define Y_MIN_POS -10
> #define Z_MIN_POS 0
> #define X_MAX_POS X_BED_SIZE
> #define Y_MAX_POS Y_BED_SIZE
> #define Z_MAX_POS 240
>

jonealex, 2017-12-31:
I tried with no BLTOUCH_DELAY and the same thing...

More info:
- Triangelabs BLTouch

:(

GMagician, 2017-12-31:
I have a 3d-touch clone and using M119 it reports always "open" when stowed or deployed, but it works perfectly.
Have you tried with M119?

studiodyne, 2018-01-01:
Amen !

[](https://github.com/MarlinFirmware/Marlin/issues/5705)

AnHardt, 2018-01-01:
Du ti the way BLtouch works, you have to execute the M119 0 to 5 ms after the probe triggers. Else you will miss the only 5ms long signal. Good luck.
During the probes this is handled in time now.

jonealex, 2018-01-01:
Well...

- `M119` works fine, TRIGGERED when STOWED, OPEN when DEPLOYED. (or inverse if logic is inverted, i tried it).
- `G28` and `G29` still not working, as the video shows.

Still trying to find the problem. Same configuration as attached. Same problem as attached video shows.

GMagician, 2018-01-01:
@AnHardt that's why I asked to test it with M119 since he gets 'triggered' when stowed, this is the real problem.
@jonealex you can try to **disconnect probe input cable** and test board input with M119. First test input connected to gnd and second test input connected to 5V **always turn off power** before changing, just to be sure not to damage equipment.

jonealex, 2018-01-01:
@GMagician I don't know why but now, TRIGGERED when DEPLOYED, OPEN when STOWED. Changing `Z_MIN_ENDSTOP_INVERTING` not working. I test with other BLTouch and the same thing.

GMagician, 2018-01-01:
Usually you don't have to change endstop inverting with bltouch, it's already overriden with default correct value.
Since you have 'triggered' it means that something is not correct with sensor. Trigger signal is 5ms long, as stated by @AnHardt, you will never see triggered.
Since you get open and triggered it means that your input is working and bltouch is not.
When you turn on the printer bltouch should do a self test. What is led status after test is finished?

Roxy-3D, 2018-01-01:
> Due to the way BLtouch works, you have to execute the M119 0 to 5 ms after the probe triggers. Else you will miss the only 5ms long signal. Good luck.

A small update for people:  The current BL-Touch probes now have a 10 ms pulse.

> I don't know why but now, TRIGGERED when DEPLOYED, OPEN when STOWED. Changing Z_MIN_ENDSTOP_INVERTING not working. I test with other BLTouch and the same thing.

My experience with BL-Touch probes is the opposite.   And when in an error condition, they report triggered also.   That is probably a good thing because if the probe can't fully deploy, you don't want the nozzle to be driven down into the bed.

GMagician, 2018-01-01:
@jonealex  It's normal that if you change Z_MIN_ENDSTOP_INVERTING you get reversed result. In your situation bltouch is giving you a signal when stowed and other signal when deployed but it shouldn't. it should always give you 'triggered' or 'open' (and 'open' means that Z_MIN_ENDSTOP_INVERTING has correct value).
Other test, have you tried to change regulation screw position?

jonealex, 2018-01-01:
@GMagician When I turn on the printer, after self test LED is ON and stays ON without blinking.

About **regulation screw position** i tried some positions, but in some of them BLTouch don't STOW/DEPLOY.

Some info:
-If I deploy it manually it fast go to stowed position. If I deploy manually for a long time it stays deployed with LED blinking.

GMagician, 2018-01-01:
When you deploy it should stay down with led off. Since it blinks it seems that you have a H/W issue on such bltouch...very strange you have same issue in more than one device

Roxy-3D, 2018-01-01:
Yeah...  But he said 'Manually deploy'.     I hope he doesn't mean he is manually pulling the pin into a deployed state.   That should result in an error state and the probe will blink...

jonealex, 2018-01-01:
I mean that doing it with GCODE works with no blinking, but with my hand LED is blinking. Now trying again with the other BLTouch... of Geetech.

GMagician, 2018-01-01:
To deploy manually have you used m280? with what value in S?
If not used m280 then try M280 P0 S10/90

Z_MIN_ENDSTOP_INVERTING_ and Z_SERVO_ANGLES are not need when bltouch is enabled. Proper values are overridden automatically by firmware.

jonealex, 2018-01-01:
@GMagician `M280 P0 S10/90` works fine. I'm trying with the other BLTouch and stows but not deploys. 

More info:
- I tried changing the PIN of signal and not working.

GMagician, 2018-01-01:
Ops just see now @Roxy-3D and your answer...deleted my last message.
@Roxy-3D with manually I meant not really manually but I though by m280 and not by lcd. Bad assumption

Roxy-3D, 2018-01-01:
@Bob-the-Kuhn    Bob, I seem to remember some of the BL-Touch clones actually had inverted signals in some of the states.    Is that true?     Is it possible we are seeing that here?

GMagician, 2018-01-01:
That sound strange and give solution to this issue. I still use geetech and they works as expected. Bad news that @jonealex one doesn't deploy.
I burned one of mine because of missing flyback diodes and in such situation it doesn't deploy (you have to replace internal diode to fix it)

jonealex, 2018-01-01:
More info that maybe helps @Roxy-3D @GMagician. The BLTouch that works fine with `M280` is ANTC LABS BLTouch, that I have read that @Bob-the-Kuhn has one of this BLTouch.

The problem gets me angry. 👎

GMagician, 2018-01-01:
I have geetech bltouch and it works too...I just ordered a spare part from Triangelabs and I really hope it's not so problematic

Roxy-3D, 2018-01-01:
> The problem gets me angry. :-1:

Yes.   But this isn't so much a problem with the firmware as it is a problem with the people cloning BL-Touch probes and not doing a good job.   And it kind of sounds like the Triangelabs people didn't even test it with Marlin.    How does that make any sense?

jonealex, 2018-01-01:
@GMagician @Roxy-3D I hope that we (you) can solve this problem or I want to get my printer out by my window.

Roxy-3D, 2018-01-01:
@jonealex   In conditionals_lcd.h try changing this block to look like this:
```
    #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
      #undef Z_MIN_ENDSTOP_INVERTING
      #define Z_MIN_ENDSTOP_INVERTING true   // this is usually false for BL-Touch probes
      #define TEST_BLTOUCH() _TEST_BLTOUCH(Z_MIN)
    #else
      #define TEST_BLTOUCH() _TEST_BLTOUCH(Z_MIN_PROBE)
    #endif
  #endif
```
If that works for you...   I'm not sure how we address it in the main code base.   Maybe we need to allow the user to specify the sub-type of the BL-Touch probe.   ie.  Antclabs, Geetech, Trianglelabs, etc.

jonealex, 2018-01-01:
@Roxy-3D I have this error while compiling:

`Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING.`

Roxy-3D, 2018-01-01:
OK...  We need to also defeat the sanity check to try this hack:   in SanityCheck.h change this block:
```
  #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
    #if ENABLED(Z_MIN_PROBE_ENDSTOP)
      #error "Enable only one option: Z_MIN_PROBE_ENDSTOP or Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN."
    #elif DISABLED(USE_ZMIN_PLUG)
      #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires USE_ZMIN_PLUG to be enabled."
    #elif !HAS_Z_MIN
      #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
//    #elif ENABLED(Z_MIN_PROBE_ENDSTOP_INVERTING) != ENABLED(Z_MIN_ENDSTOP_INVERTING)
//      #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
    #endif
```

jonealex, 2018-01-01:
@Roxy-3D I tried. 
- `M280`: All right. 
- `G28/G29`: Deployed and stowed, without waiting deployed to touch the bed. Z axis only goes up.

Roxy-3D, 2018-01-01:
OK...    I think what this is telling us is you don't need the two hacks we just did in the previous posts.   
Probably this means the 'invert' setting is close to what a real BL-Touch does.   But the fact that it is seeing the probe as triggered immediately means something else is different.   It maybe we need a delay before the probe returns the 'correct' value after being deployed.

I think you need to build the firmware with DEBUG_LEVELING_FEATURE turned on.   And then do a M111 S255 and post a log of the output.   They might give us a clue where things are different.

(And revert those previous changes we made!!!)

jonealex, 2018-01-01:
@Roxy-3D Done !
```
>>> G28
SENDING:G28
echo:G28
>>> gcode_G28
Machine Type: Cartesian
Probe: BLTOUCH
Probe Offset X:-26 Y:-40 Z:-2.40 (Left-Front & Below Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
  current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
Raise Z (before homing) to 5.00
>>> do_blocking_move_to(0.00, 0.00, 5.00)
<<< do_blocking_move_to
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, -379.50, 0.00)
  current_position=(0.00, 0.00, 5.00) : sync_plan_position
<<< do_homing_move(X)
Move Away:
>>> do_homing_move(X, 5.00, 0.00)
  current_position=(0.00, 0.00, 5.00) : sync_plan_position
<<< do_homing_move(X)
Home 2 Slow:
>>> do_homing_move(X, -10.00, 25.00)
  current_position=(0.00, 0.00, 5.00) : sync_plan_position
<<< do_homing_move(X)
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = -33.00
 soft_endstop_max = 220.00
> home_offset[X] = 0.00
  current_position=(-33.00, 0.00, 5.00) :
<<< set_axis_is_at_home(X)
  current_position=(-33.00, 0.00, 5.00) : sync_plan_position
  current_position=(-33.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
<<< homeaxis(X)
  current_position=(-33.00, 0.00, 5.00) : > homeX
>>> homeaxis(Y)
Home 1 Fast:
>>> do_homing_move(Y, -345.00, 0.00)
  current_position=(-33.00, 0.00, 5.00) : sync_plan_position
<<< do_homing_move(Y)
Move Away:
>>> do_homing_move(Y, 5.00, 0.00)
  current_position=(-33.00, 0.00, 5.00) : sync_plan_position
<<< do_homing_move(Y)
Home 2 Slow:
>>> do_homing_move(Y, -10.00, 25.00)
  current_position=(-33.00, 0.00, 5.00) : sync_plan_position
echo:busy: processing
<<< do_homing_move(Y)
>>> set_axis_is_at_home(Y)
For Y axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = -10.00
 soft_endstop_max = 220.00
> home_offset[Y] = 0.00
  current_position=(-33.00, -10.00, 5.00) :
<<< set_axis_is_at_home(Y)
  current_position=(-33.00, -10.00, 5.00) : sync_plan_position
  current_position=(-33.00, -10.00, 5.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
  current_position=(-33.00, -10.00, 5.00) : > homeY
>>> homeaxis(Z)
  current_position=(-33.00, -10.00, 5.00) : set_probe_deployed
deploy: 1
do_probe_raise(10.00)
>>> do_blocking_move_to(-33.00, -10.00, 12.40)
echo:busy: processing
<<< do_blocking_move_to
>>> do_blocking_move_to(-33.00, -10.00, 12.40)
<<< do_blocking_move_to
Home 1 Fast:
>>> do_homing_move(Z, -360.00, 0.00)
set_bltouch_deployed(1)
  current_position=(-33.00, -10.00, 0.00) : sync_plan_position
set_bltouch_deployed(0)
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, 5.00, 0.00)
  current_position=(-33.00, -10.00, 0.00) : sync_plan_position
echo:busy: processing
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, -10.00, 1.00)
set_bltouch_deployed(1)
  current_position=(-33.00, -10.00, 0.00) : sync_plan_position
set_bltouch_deployed(0)
<<< do_homing_move(Z)
>>> set_axis_is_at_home(Z)
For Z axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 240.00
*** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
> zprobe_zoffset = -2.40
> home_offset[Z] = 0.00
  current_position=(-33.00, -10.00, 2.40) :
<<< set_axis_is_at_home(Z)
  current_position=(-33.00, -10.00, 2.40) : sync_plan_position
  current_position=(-33.00, -10.00, 2.40) : > AFTER set_axis_is_at_home
  current_position=(-33.00, -10.00, 2.40) : set_probe_deployed
deploy: 0
do_probe_raise(10.00)
>>> do_blocking_move_to(-33.00, -10.00, 12.40)
echo:busy: processing
echo:busy: processing
<<< do_blocking_move_to
>>> do_blocking_move_to(-33.00, -10.00, 12.40)
<<< do_blocking_move_to
<<< homeaxis(Z)
  current_position=(-33.00, -10.00, 12.40) : > (home_all || homeZ) > final
  current_position=(-33.00, -10.00, 12.40) : sync_plan_position
  current_position=(-33.00, -10.00, 12.40) : clean_up_after_endstop_or_probe_move
<<< gcode_G28
echo:M105
```

Roxy-3D, 2018-01-01:
At this part of the process....   Does it back away and re-probe?   Or does it just back away and keep backing away?
```
do_probe_raise(10.00)
>>> do_blocking_move_to(-33.00, -10.00, 12.40)
echo:busy: processing
<<< do_blocking_move_to
>>> do_blocking_move_to(-33.00, -10.00, 12.40)
<<< do_blocking_move_to
Home 1 Fast:
>>> do_homing_move(Z, -360.00, 0.00)
set_bltouch_deployed(1)
  current_position=(-33.00, -10.00, 0.00) : sync_plan_position
set_bltouch_deployed(0)
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, 5.00, 0.00)
  current_position=(-33.00, -10.00, 0.00) : sync_plan_position
echo:busy: processing
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, -10.00, 1.00)
set_bltouch_deployed(1)
  current_position=(-33.00, -10.00, 0.00) : sync_plan_position
set_bltouch_deployed(0)
<<< do_homing_move(Z)
>>> set_axis_is_at_home(Z)
For Z axis:
```
If it keeps backing away and never probes here...   I think we will need to add some more debug code to the DEBUG_LEVELING stuff to figure out what is going on.

jonealex, 2018-01-01:
@Roxy-3D Z axis always go up, one time for the (wrong) probe and other time to go +10 on Z axis as always to keep away the nozzle from the bed. Between `set_bltouch_deployed(1)` and `set_bltouch_deployed(0)` looks like there's nothing, it does it without pause to wait the bed touching the sensor.

Roxy-3D, 2018-01-01:
In Marlin_main.cpp please replace these code blocks.   Rebuild.   And do the M111 S255 and G28 again.   
```
/**
 * Home an individual linear axis
 */
static void do_homing_move(const AxisEnum axis, const float distance, const float fr_mm_s=0.0) {

  #if ENABLED(DEBUG_LEVELING_FEATURE)
    if (DEBUGGING(LEVELING)) {
      SERIAL_ECHOPAIR(">>> do_homing_move(", axis_codes[axis]);
      SERIAL_ECHOPAIR(", ", distance);
      SERIAL_ECHOPAIR(", ", fr_mm_s);
      SERIAL_CHAR(')');
      SERIAL_EOL();
    }
    SERIAL_ECHOLNPGM("at entry to do_homing_move():");
    endstops.report_state();
  #endif

  #if HOMING_Z_WITH_PROBE && ENABLED(BLTOUCH)
    const bool deploy_bltouch = (axis == Z_AXIS && distance < 0);
    if (deploy_bltouch) set_bltouch_deployed(true);

    SERIAL_ECHOLNPGM("  after BL-Touch deploy:");
    endstops.report_state();
    delay(50);
    endstops.report_state();
    delay(50);
    endstops.report_state();
    delay(50);
    endstops.report_state();
  #endif
  #if QUIET_PROBING
    if (axis == Z_AXIS) probing_pause(true);
  #endif

  // Tell the planner the axis is at 0
  current_position[axis] = 0;

  #if IS_SCARA
    SYNC_PLAN_POSITION_KINEMATIC();
    current_position[axis] = distance;
    inverse_kinematics(current_position);
    planner.buffer_line(delta[A_AXIS], delta[B_AXIS], delta[C_AXIS], current_position[E_AXIS], fr_mm_s ? fr_mm_s : homing_feedrate(axis), active_extruder);
  #else
    sync_plan_position();
    current_position[axis] = distance;
    planner.buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], fr_mm_s ? fr_mm_s : homing_feedrate(axis), active_extruder);
  #endif

  stepper.synchronize();

  #if QUIET_PROBING
    if (axis == Z_AXIS) probing_pause(false);
  #endif

  #if HOMING_Z_WITH_PROBE && ENABLED(BLTOUCH)
    if (deploy_bltouch) set_bltouch_deployed(false);
    SERIAL_ECHOLNPGM("  at exit:");
    endstops.report_state();
  #endif

  endstops.hit_on_purpose();

  #if ENABLED(DEBUG_LEVELING_FEATURE)
    if (DEBUGGING(LEVELING)) {
      SERIAL_ECHOPAIR("<<< do_homing_move(", axis_codes[axis]);
      SERIAL_CHAR(')');
      SERIAL_EOL();
    }
[... 167 lines cut ...]
  // Re-enable stealthChop if used. Disable diag1 pin on driver.
  #if ENABLED(SENSORLESS_HOMING)
    #if ENABLED(X_IS_TMC2130)
      if (axis == X_AXIS) tmc_sensorless_homing(stepperX, false);
    #endif
    #if ENABLED(Y_IS_TMC2130)
      if (axis == Y_AXIS) tmc_sensorless_homing(stepperY, false);
    #endif
  #endif

  // Put away the Z probe
  #if HOMING_Z_WITH_PROBE
    if (axis == Z_AXIS && STOW_PROBE()) return;
  #endif

  #if ENABLED(DEBUG_LEVELING_FEATURE)
    if (DEBUGGING(LEVELING)) {
      SERIAL_ECHOPAIR("<<< homeaxis(", axis_codes[axis]);
      SERIAL_CHAR(')');
      SERIAL_EOL();
    }
  #endif
} // homeaxis()
```
This should let us see exactly what the probe status is at deploy and retract.   (It kind of feels to me like that is inverted...  or at least delayed in being correct.)

jonealex, 2018-01-01:
@Roxy-3D Done again !
```
>>> G28
SENDING:G28
echo:G28
>>> gcode_G28
Machine Type: Cartesian
Probe: BLTOUCH
Probe Offset X:-26 Y:-40 Z:-2.40 (Left-Front & Below Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
  current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
Raise Z (before homing) to 5.00
>>> do_blocking_move_to(0.00, 0.00, 5.00)
<<< do_blocking_move_to
>>> homeaxis(X)
status before Home 1 Fast:
Home 1 Fast:
>>> do_homing_move(X, -379.50, 0.00)
at entry to do_homing_move():
  after BL-Touch deploy:
  current_position=(0.00, 0.00, 5.00) : sync_plan_position
  at exit:
echo:endstops hit:  X:-0.00
<<< do_homing_move(X)
status after do_homeing_move() Fast
Move Away:
>>> do_homing_move(X, 5.00, 0.00)
at entry to do_homing_move():
  after BL-Touch deploy:
  current_position=(0.00, 0.00, 5.00) : sync_plan_position
  at exit:
<<< do_homing_move(X)
status after Move Away:
Home 2 Slow:
>>> do_homing_move(X, -10.00, 25.00)
at entry to do_homing_move():
  after BL-Touch deploy:
  current_position=(0.00, 0.00, 5.00) : sync_plan_position
echo:busy: processing
  at exit:
echo:endstops hit:  X:-4.80
<<< do_homing_move(X)
status after slow do_homing_move():
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = -33.00
 soft_endstop_max = 220.00
> home_offset[X] = 0.00
  current_position=(-33.00, 0.00, 5.00) :
<<< set_axis_is_at_home(X)
  current_position=(-33.00, 0.00, 5.00) : sync_plan_position
  current_position=(-33.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
<<< homeaxis(X)
  current_position=(-33.00, 0.00, 5.00) : > homeX
>>> homeaxis(Y)
status before Home 1 Fast:
Home 1 Fast:
>>> do_homing_move(Y, -345.00, 0.00)
at entry to do_homing_move():
  after BL-Touch deploy:
  current_position=(-33.00, 0.00, 5.00) : sync_plan_position
  at exit:
echo:endstops hit:  Y:-0.00
<<< do_homing_move(Y)
status after do_homeing_move() Fast
Move Away:
>>> do_homing_move(Y, 5.00, 0.00)
at entry to do_homing_move():
  after BL-Touch deploy:
  current_position=(-33.00, 0.00, 5.00) : sync_plan_position
  at exit:
<<< do_homing_move(Y)
status after Move Away:
Home 2 Slow:
>>> do_homing_move(Y, -10.00, 25.00)
at entry to do_homing_move():
  after BL-Touch deploy:
  current_position=(-33.00, 0.00, 5.00) : sync_plan_position
  at exit:
echo:endstops hit:  Y:-5.32
<<< do_homing_move(Y)
status after slow do_homing_move():
>>> set_axis_is_at_home(Y)
For Y axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = -10.00
 soft_endstop_max = 220.00
> home_offset[Y] = 0.00
  current_position=(-33.00, -10.00, 5.00) :
<<< set_axis_is_at_home(Y)
  current_position=(-33.00, -10.00, 5.00) : sync_plan_position
  current_position=(-33.00, -10.00, 5.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
  current_position=(-33.00, -10.00, 5.00) : > homeY
>>> homeaxis(Z)
  current_position=(-33.00, -10.00, 5.00) : set_probe_deployed
deploy: 1
do_probe_raise(10.00)
>>> do_blocking_move_to(-33.00, -10.00, 12.40)
echo:busy: processing
<<< do_blocking_move_to
>>> do_blocking_move_to(-33.00, -10.00, 12.40)
<<< do_blocking_move_to
status before Home 1 Fast:
Home 1 Fast:
>>> do_homing_move(Z, -360.00, 0.00)
at entry to do_homing_move():
set_bltouch_deployed(1)
  after BL-Touch deploy:
  current_position=(-33.00, -10.00, 0.00) : sync_plan_position
echo:busy: processing
set_bltouch_deployed(0)
  at exit:
echo:endstops hit:  Z:-0.00
<<< do_homing_move(Z)
status after do_homeing_move() Fast
Move Away:
>>> do_homing_move(Z, 5.00, 0.00)
at entry to do_homing_move():
  after BL-Touch deploy:
  current_position=(-33.00, -10.00, 0.00) : sync_plan_position
echo:busy: processing
  at exit:
<<< do_homing_move(Z)
status after Move Away:
Home 2 Slow:
>>> do_homing_move(Z, -10.00, 1.00)
at entry to do_homing_move():
set_bltouch_deployed(1)
  after BL-Touch deploy:
  current_position=(-33.00, -10.00, 0.00) : sync_plan_position
set_bltouch_deployed(0)
  at exit:
echo:endstops hit:  Z:0.00
<<< do_homing_move(Z)
status after slow do_homing_move():
>>> set_axis_is_at_home(Z)
For Z axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 240.00
*** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
> zprobe_zoffset = -2.40
> home_offset[Z] = 0.00
  current_position=(-33.00, -10.00, 2.40) :
<<< set_axis_is_at_home(Z)
  current_position=(-33.00, -10.00, 2.40) : sync_plan_position
  current_position=(-33.00, -10.00, 2.40) : > AFTER set_axis_is_at_home
  current_position=(-33.00, -10.00, 2.40) : set_probe_deployed
deploy: 0
do_probe_raise(10.00)
>>> do_blocking_move_to(-33.00, -10.00, 12.40)
echo:busy: processing
echo:busy: processing
<<< do_blocking_move_to
>>> do_blocking_move_to(-33.00, -10.00, 12.40)
<<< do_blocking_move_to
<<< homeaxis(Z)
  current_position=(-33.00, -10.00, 12.40) : > (home_all || homeZ) > final
  current_position=(-33.00, -10.00, 12.40) : sync_plan_position
  current_position=(-33.00, -10.00, 12.40) : clean_up_after_endstop_or_probe_move
<<< gcode_G28
echo:M105
```

Roxy-3D, 2018-01-01:
It doesn't look like the firmware is seeing a bad value for the probe status.   

Is it possible something got corrupted in your file set?   I'm wondering if you should download a fresh copy of the code base and drop your Configuration.h (and Configuration_adv.h) files into it.

What ever is wrong...   It isn't much.    But this is very confusing....

jonealex, 2018-01-01:
@Roxy-3D I recently downloaded 1.1.8 version clean, and put my configuration files... Same result...

Roxy-3D, 2018-01-01:
OK...   We might need @Bob-the-Kuhn  to weigh in.   He knows a lot more about how the BL-Touch (and its clones) behave.   He might know the answer to what is happening off of the top of his head.

jonealex, 2018-01-01:
@Roxy-3D All right, I will wait for @Bob-the-Kuhn opinion. Thanks a lot for your help guys 👍 , I have my printer stopped since one week and I want a lot start printing...

Roxy-3D, 2018-01-01:
Let's wait and see what @Bob-the-Kuhn says...   But one idea is I have extra BL-Touch probes.   If you send your Triangle Labs probe to Bob that would make things easier for him to debug and figure out a fix.   And I could send you a legitimate (Antclabs) BL-Touch to replace it.

jonealex, 2018-01-01:
I put here all my Marlin files to make problem solving easier.

Thanks a lot.
[Marlin-1.1.8_BLTouch_Problem.zip](https://github.com/MarlinFirmware/Marlin/files/1596489/Marlin-1.1.8_BLTouch_Problem.zip)

jonealex, 2018-01-01:
@Roxy-3D @GMagician @Bob-the-Kuhn **PROBLEM SOLVED !**

I read this from @Bob-the-Kuhn : 

To run a clone you need the following settings:

```
//#define BLTOUCH 
#define Z_ENDSTOP_SERVO_NR 0
#define Z_SERVO_ANGLES {10,90} // Z Servo Deploy and Stow angles
#define NUM_SERVOS 1
#define SERVO_DELAY 750  // milliseconds 500 is marginal
```

You also need to set `#define Z_MIN_ENDSTOP_INVERTING true` or `Z_MIN_PROBE_ENDSTOP_INVERTING true`, depending what pin the probe is tied to.

When all is configured properly then `M119` will report `OPEN` when the clone is deployed and `TRIGGERED` when it is stowed.

**NOW WORKING BLTOUCH !**

jonealex, 2018-01-01:
Some info:

- Problem with offsets, but working BLTouch.

jonealex, 2018-01-02:
**UPDATED**

As the BLTouch works, at start don't do any self test, not working leveling (LED blinking).

I think that is like pulse it's not corresponding with readings of the signal wire.

jonealex, 2018-01-02:
Now, I have this code enabled on `Conditionals_LCD` and BLTouchs goes perfect:

 ```
 #if true
    #ifndef Z_ENDSTOP_SERVO_NR
      #define Z_ENDSTOP_SERVO_NR 0
    #endif
    #ifndef NUM_SERVOS
      #define NUM_SERVOS (Z_ENDSTOP_SERVO_NR + 1)
    #endif
    #undef DEACTIVATE_SERVOS_AFTER_MOVE
    #if NUM_SERVOS == 1
      #undef SERVO_DELAY
      #define SERVO_DELAY { 750 }
    #endif
    #ifndef BLTOUCH_DELAY
      #define BLTOUCH_DELAY 750
    #endif
    #undef Z_SERVO_ANGLES
    #define Z_SERVO_ANGLES { BLTOUCH_DEPLOY, BLTOUCH_STOW }

    #define BLTOUCH_DEPLOY    10
    #define BLTOUCH_STOW      90
    #define BLTOUCH_SELFTEST 120
    #define BLTOUCH_RESET    160
    #define _TEST_BLTOUCH(P) (READ(P##_PIN) != P##_ENDSTOP_INVERTING)

    // Always disable probe pin inverting for BLTouch
    #undef Z_MIN_PROBE_ENDSTOP_INVERTING
    #define Z_MIN_PROBE_ENDSTOP_INVERTING false

    #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
      #undef Z_MIN_ENDSTOP_INVERTING
      #define Z_MIN_ENDSTOP_INVERTING true
      #define TEST_BLTOUCH() _TEST_BLTOUCH(Z_MIN)
    #else
      #define TEST_BLTOUCH() _TEST_BLTOUCH(Z_MIN_PROBE)
    #endif
  #endif
```

Bob-the-Kuhn, 2018-01-02:
There's a utility that can identify the type of BLTouch (pulse vs. clone) and does a rough measurement of the pulse width.

Enable **PINS_DEBUGGING** in **configuration_adv.h** and then issue a **M43 S** command.

jonealex, 2018-01-18:
@Bob-the-Kuhn I get with `M43 S` this output:

`>>> M43 S
SENDING:M43 S
Servo probe test
.  using index:  0
.  deploy angle: 10
.  stow angle:   90
. probe uses Z_MIN pin: 18
. uses Z_MIN_ENDSTOP_INVERTING (ignores Z_MIN_PROBE_ENDSTOP_INVERTING)
. Z_MIN_ENDSTOP_INVERTING: true
. deploy & stow 4 times
WARNING - INVERTING setting probably backwards
please trigger probe`

jonealex, 2018-01-18:
More info:

@Bob-the-Kuhn With `M119` deployed and stowed always on `TRIGGERED`.

GMagician, 2018-01-18:
Theoretically with your configuration you don't need to set Z_SERVO_ANGLES nor Z_ENDSTOP_SERVO_NR these values are already set/overwritten when you enable BLtouch.
Finally as I know M119 should always return open because TRIGGERED is present only for 10ms (or 5ms on newer Bltouch).

Some strange things about trianglelabs bltouch. They state that it's TRIGGERED by default, maybe what @Roxy-3D said about "some" reversed clone is real.

When you say

> When extruder FAN conected, BLTouch do some strange things.

what do you mean? Connected or activated?
Because of PWM noise I had some strange behaviour too (finally burned).
I added some flyback diodes on FAN and BED (mine are PWM controlled)

Roxy-3D, 2018-01-18:
> because TRIGGERED is present only for 10ms (or 5ms on newer Bltouch).

Actually...  It is the opposite of that.  The original BL-Touch probes put out a 5ms pulse.  We asked Antclabs to make the pulse longer and they did.

GMagician, 2018-01-18:
@Roxy-3D ops...

Roxy-3D, 2018-01-18:
I wish they would have made the pulse 50 ms long....    10 ms is still pretty short.

GMagician, 2018-01-18:
For interrupt 5ms are enough but for polling maybe 10ms are not the best

jonealex, 2018-01-19:
@Roxy-3D @GMagician @Bob-the-Kuhn I have installed a Geetech BLTouch, that i think it's most common? But the result it's the same thing as described on my last message...

:(

jonealex, 2018-01-22:
@GMagician @Roxy-3D @Bob-the-Kuhn 

All right, now BLTouch of Geetech working (only `G28`). When I do `G29`, first it goes home fine but, when it goes to the first level point doesn't deploy. My current settings:

**Configuration.h**
```
#define BLTOUCH
#if ENABLED(BLTOUCH)
  //#define BLTOUCH_DELAY 750   // (tested with this activated too)
#endif
```
**Conditionales_LCD.h**
```
  #if ENABLED(BLTOUCH)
    #ifndef Z_ENDSTOP_SERVO_NR
      #define Z_ENDSTOP_SERVO_NR 0
    #endif
    #ifndef NUM_SERVOS
      #define NUM_SERVOS (Z_ENDSTOP_SERVO_NR + 1)
    #endif
    #undef DEACTIVATE_SERVOS_AFTER_MOVE
    #if NUM_SERVOS == 1
      #undef SERVO_DELAY
      #define SERVO_DELAY { 100 } //50
    #endif
    #ifndef BLTOUCH_DELAY
      #define BLTOUCH_DELAY 375
    #endif
    #undef Z_SERVO_ANGLES
    #define Z_SERVO_ANGLES { BLTOUCH_DEPLOY, BLTOUCH_STOW }

    #define BLTOUCH_DEPLOY    10
    #define BLTOUCH_STOW      90
    #define BLTOUCH_SELFTEST 120
    #define BLTOUCH_RESET    160
    #define _TEST_BLTOUCH(P) (READ(P##_PIN) != P##_ENDSTOP_INVERTING)

    // Always disable probe pin inverting for BLTouch
    #undef Z_MIN_PROBE_ENDSTOP_INVERTING
    #define Z_MIN_PROBE_ENDSTOP_INVERTING false

    #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
      #undef Z_MIN_ENDSTOP_INVERTING
      #define Z_MIN_ENDSTOP_INVERTING false
      #define TEST_BLTOUCH() _TEST_BLTOUCH(Z_MIN)
    #else
      #define TEST_BLTOUCH() _TEST_BLTOUCH(Z_MIN_PROBE)
    #endif
  #endif
```

closed 2018-02-14 by thinkyhead

jonealex, 2018-07-13:
Here I ham, my BLTouch has broken and I put again Antclabs BLTouch and... Testing works, but doing G28/G29 not working... Deploys-Stows without waiting to touch anything...

adrianmihalko, 2018-08-06:
So Trianglelabs TL-Touch is supported at this moment? Many guides are using this probe using this config:

```
#define ENDER3

// If you are using our EZOut filament sensor kit please follow the install guide
// and then uncomment the #define EZOUT_ENABLE line below.
//#define EZOUT_ENABLE

#define AUTO_BED_LEVELING_BILINEAR 
#define BLTOUCH
#define SERVO0_PIN 27
#define Z_SAFE_HOMING

// EZABL Probe Mounts (Ender 3 uses the same mounts as CR-10)
//#define CR10_VOLCANO
//#define CR10_V6HEAVYDUTY
//#define CR10_OEM
//#define CR10_FANG
//#define TM3DAERO
//#define TM3DAERO_EXTENDED
//#define PETSFANG
#define CUSTOM_PROBE
```

thinkyhead, 2018-08-07:
That's the TH3D config by @houseofbugs. AFAIK the "**TL**Touch" should be compatible with the BLTouch. It should be trying to be, anyway…